micro-ROS/micro_ros_arduino #7: Example nodes: no communication between micro-ROS agent and ROS2

anaelle-sw, 2020-10-28:
Hello!

I am fresh new to microROS, and I am testing the publisher example on a Teensy board. I am not able to see the topic advertised by the micro-ROS publisher via ROS2 on my computer.

## Setup: 
- Computer: ROS2 Foxy / Ubuntu 20.04
- Board: Teensy 3.1 / Arduino IDE 1.8.13 / Teensyduino 1.53
Board is connected to computer via USB cable.

## Steps to reproduce issue:
- After adding micro_ros_arduino as a .zip library into Arduino IDE, I upload the publisher code on the board (via Arduino IDE). 
- On the computer, I run a micro-ROS agent via command line: `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6` (as mentioned in ROS Discourse [here](https://discourse.ros.org/t/micro-ros-teensy-support/16852/10?u=anaelle-sw)), which returns:
```
[1603891560.502795] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

[1603891560.502964] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

[1603891561.374699] info     | Root.cpp           | create_client            | create                 | client_key: 0x50E95322, session_id:0x81

[1603891561.374753] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1357468450, address: 1

[1603891561.374864] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 19, data: 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

[1603891561.376922] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 112, data: 0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00

[1603891561.380689] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x50E95322, participant_id: 0x000(1)

[1603891561.380755] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 14, data: 0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00

[1603891561.380771] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80

[1603891561.382248] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80

[1603891561.382283] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80

[1603891561.382292] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80

[1603891561.382369] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80

[1603891561.382388] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80

[1603891561.388881] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 148, data: 0000: 81 80 01 00 01 05 89 00 00 0B 00 02 02 02 00 00 7B 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F
0040: 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73
0060: 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E
0080: 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00 00

[1603891561.388938] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80

[1603891561.389098] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x50E95322, topic_id: 0x000(2), participant_id: 0x000(1)

[1603891561.389200] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 14, data: 0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00

[1603891561.389223] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80

[1603891561.389234] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80

[1603891561.390459] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80

[1603891561.390492] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80

[1603891561.390508] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80

[1603891561.390571] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80

[1603891561.390596] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80

[... 36 lines cut ...]

[1603891561.405667] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50E95322, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 02 00 02 00 80
```
So it seems the serial communication between the node and the board is established. Messages seem to be sent by the board and received by the agent

- I try to get the messages on ROS2 via rqt, or CLI: `ros2 topic echo /[tab tab]`. The only topics available are '/parameter_events' and '/rosout'. 

## Questions
- Instead of launching the micro-ROS agent via the previous command line, I tried:
` docker run -it --rm -d /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6 ` (as mentioned is this repo's readme) wich returns:`docker: invalid reference format.`
Or also the command line:
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00` (which is from the micro-ros tutorials) which returns:
```
[1603892595.656215] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

[1603892596.410382] info     | Root.cpp           | create_client            | create                 | client_key: 0x50E95322, session_id: 0x81

[1603892596.410424] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1357468450, address: 1
```
The communication between the board and the agent seems to not be established.
Could you please confirm that the CLI `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6` is the right way to run agent for micro_ros_arduino?

- According to the micro-ROS tutorials, once the board is flashed and the micro-ROS agent is running, it is possible to get the messages via `ros2 topic echo`. Is it also the case for micro_ros_arduino, or is there something I am missing?

Thank you for helping!

pablogs9, 2020-10-28:
Hello @anaelle-sw, thanks for trying this repo!

It seems that you micro-ROS client connects properly with the agent in the first bash capture you show but it is not sending anything to the DDS world, when it sends data to the ROS2 DDS dataspace you see something like:

```bash
[1603894397.859135] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x526C89A2, len: 36, data: 
0000: 81 80 64 01 07 01 1C 00 01 71 00 15 00 00 00 60 C4 AB 40 40 00 00 00 80 A7 FC 28 40 00 00 00 A0
0020: 0D D9 0B 40
[1603894397.859241] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 24, data: 
```

In order to ensure that everything is working properly, lets use the dockerized version of the micro-ROS Agent by now, because depending on the FastDDS versions you have installed in your computer maybe there is some incompatibility, specially if you have multiple versions of ROS 2 installed in the same computer.

So in order to launch the Agent: (Notice that I have modified the command, there was a typo in the `-d`, it should be `-v`)

```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6
```

and now, in another terminal, open a ROS 2 foxy docker and test the echo:

```bash
docker run -it --rm --net=host ros:foxy
# Inside the docker:
ros2 topic list
ros2 topic echo [your topic name]
```

anaelle-sw, 2020-10-28:
Thank you for your quick answer!
Everything works perfectly well with dockers (for the publisher example, and the subscriber too!). Thank you very much! 
Just to specify my case, only the Foxy distribution for ROS2 (sourced), and ROS1 noetic (not sourced for the tests),  are installed on my computer.

You say:
> lets use the dockerized version of the micro-ROS Agent _by now_ 

Do you know when/if a library not depending on the FastDDS version is planned to be released?

pablogs9, 2020-10-28:
Well, FastDDS is the default middleware implementation for ROS 2 so it is recommended to use it. Also, the micro-ROS agent uses FastDDS to interact with the rest of ROS 2 nodes.

I would be happy if you get the `ros2 ...` version of the agent working. Now that we know that everything works as expected, can you try again:

```bash
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6
```

and paste here the complete output?

Remember that if you see these `[** <<DDS>> **]` traces, the `ros2 topic echo` should works.

Give me feedback.

anaelle-sw, 2020-10-28:
When running agent with:
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_6014540-if00 -v6`
the output is:
```
[1603899006.028592] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1603899006.028746] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1603899006.028833] info     | Root.cpp           | create_client            | create                 | client_key: 0x50C7822F, session_id: 0x81
[1603899006.028857] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1355252271, address: 1
[1603899006.028883] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1355252271, address: 1
[1603899006.028891] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1355252271, address: 1
[1603899006.028894] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1603899006.028896] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1355252271, address: 1
[1603899006.028909] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1603899006.028917] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1603899006.028933] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1603899006.031817] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 112, data: 0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1603899006.034468] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x50C7822F, participant_id: 0x000(1)
[1603899006.034514] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 14, data: 0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1603899006.034525] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1603899006.035597] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1603899006.035612] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1603899006.035619] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1603899006.035658] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1603899006.035668] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1603899006.046018] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 148, data: 0000: 81 80 01 00 01 05 89 00 00 0B 00 02 02 02 00 00 7B 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F
0040: 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73
0060: 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E
0080: 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00 00
[1603899006.046047] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1603899006.046163] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x50C7822F, topic_id: 0x000(2), participant_id: 0x000(1)
[1603899006.046238] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 14, data: 0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1603899006.046246] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1603899006.046251] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[... cut ...]
[1603899009.106913] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 16, data: 
```
And so on...
As you can see the `[** <<DDS>> **]` traces appears. But I cannot check out the topic with `ros2 topic echo` (output: `WARNING: topic [/micro_ros_arduino_node_publisher] does not appear to be published yet`), while in the docker the command line: `ros2 topic echo /micro_ros_arduino_node_publisher` works (the received data are dispalyed in the shell, no message is missed).

pablogs9, 2020-10-28:
Have you updated your ROS2 foxy installation recently?

They have released some minor updates for the Foxy version, check if a `apt update` and `apt upgrade` install new versions of the ROS 2 Foxy packets.

Meanwhile, I'm going to ask if there are any know incompatibilities between agent and Foxy versions of the middleware...

anaelle-sw, 2020-10-28:
Yes, my Foxy installation is up to date. 

> this is not a micro-ROS + Arduino issue

--> Good to know, thanks.

I will let you know if I can get it working.

anaelle-sw, 2020-10-30:
Apparently, the problem was coming from the environment variable `ROS_DOMAIN_ID`, which was set in `.bashrc`, (for usage of Turtlebot3 as recommended [here](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/#save-bash-command-for-setup)):
`ROS_DOMAIN_ID=30`

When this environment variable is not set up in`.bahsrc`, I can get the messages with `ros2 topic` command, without using the Foxy docker.

pablogs9, 2020-10-30:
Nice to know!

I'm going to close this since it is solved.

Do not hesitate to open any other issue if you have more questions. Also, we are willing to hear about user feedback. If you want to share your use case we are starting a micro-ROS community here: https://discourse.ros.org/c/embedded/9.

closed 2020-10-30 by pablogs9

Alex-Beh, 2023-07-04:
> When this environment variable is not set up in`.bashrc`, I can get the messages with `ros2 topic` command, without using the Foxy docker.

It solved my issue but may I know the reason behind it? So I cannot set the DOMAIN_ID in .bashrc when I want to use micro-ros?