micro-ROS/micro_ros_arduino #9: Simple service server: no response is sent

anaelle-sw, 2020-11-02:
Hi! I am trying to create an "add_two_ints" service server, based on the example from micro-ROS-demos repo [here](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/addtwoints_server/main.c), but modified to get the " `void_setup()` / `void_loop()` format ". The server receives the request, but no response is sent to the service client.

## Setup:
- Computer: ROS2 Foxy / Ubuntu 20.04
- Board: Teensy 3.1 / Arduino IDE 1.8.13 / Teensyduino 1.53
Board is connected to computer via USB cable.

## Steps to reproduce:
The .ino source code uploaded on the Teensy is:
 ```
#include <micro_ros_arduino.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

// ros:
rcl_init_options_t options;
rcl_node_options_t node_ops;
rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

// service:
rcl_service_options_t service_options;
rcl_service_t serv;
const rosidl_service_type_support_t * service_type_support;
rcl_wait_set_t wait_set;

// **** SETUP ****
void setup() {
  delay(5000); 
  allocator = rcl_get_default_allocator();

  options = rcl_get_zero_initialized_init_options();
  rcl_init_options_init(&options, rcl_get_default_allocator());

  rclc_support_init(&support, 0, NULL, &allocator);

  node_ops = rcl_node_get_default_options();
  node = rcl_get_zero_initialized_node();
  rclc_node_init_default(&node, "micro_ros_service_server_node", "", &support);

  // init server
  const char * service_name = "/add_two_ints";
  service_options = rcl_service_get_default_options();
  serv = rcl_get_zero_initialized_service();
  service_type_support = ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts);
  rcl_service_init(&serv, &node, service_type_support, service_name, &service_options);
  wait_set = rcl_get_zero_initialized_wait_set();
  rcl_wait_set_init(&wait_set, 0, 0, 0, 1, 0, 0, &support.context, rcl_get_default_allocator());

  executor = rclc_executor_get_zero_initialized_executor();
  rclc_executor_init(&executor, &support.context, 1, &allocator);
}

//**** LOOP ****
void loop() {
  delay(100);
  rcl_wait_set_clear(&wait_set);

  size_t index;
  rcl_wait_set_add_service(&wait_set, &serv, &index);

  rcl_wait(&wait_set, RCL_MS_TO_NS(1));
  for (size_t i = 0; i < wait_set.size_of_services; i++) {
    if (wait_set.services[i]) {
[... 115 lines cut ...]
`ros2 service call add_two_ints example_interfaces/srv/AddTwoInts "{a: 1, b: 0}"`
, which output is:
```
waiting for service to become available...
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=1, b=0) 
```

The micro-ROS agent response is:
```
[1604311838.727954] debug    | Replier.cpp        | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 40, data: 
0000: 01 0F 8E E6 C1 71 01 00 01 00 00 00 00 00 12 03 00 00 00 00 01 00 00 00 01 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00
[1604311838.728068] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x01A06D5E, len: 52, data: 
0000: 81 80 02 00 09 01 2C 00 00 0C 00 08 01 0F 8E E6 C1 71 01 00 01 00 00 00 00 00 12 03 00 00 00 00
0020: 01 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1604311838.745935] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x01A06D5E, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
```
So, the service is recognized as available, and the server receives the request, but the response is not sent. 

I also tried to launch the micro-ROS agent via a docker: 
` sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/teensy -v6`
The outputs are the same.

I guess this issue is due from my .ino source code. Do you have any idea what causes this? Thanks for helping!

pablogs9, 2020-11-03:
Working on it, I'll tell you something asap.

pablogs9, 2020-11-03:
Hello @anaelle-sw I found something:

Please recheck [this line](https://github.com/micro-ROS/micro-ROS-demos/blob/e4cc345159f9b0dcb578a39020311922af0ec150/rclc/addtwoints_server/main.c#L31).

You have written: 
```c
 rcl_wait_set_init(&wait_set, 0, 0, 0, 1, 0, 0, &support.context, rcl_get_default_allocator());
```

and in the demo it says:

```c
rcl_wait_set_init(&wait_set, 0, 0, 0, 0, 1, 0, &context, rcl_get_default_allocator())
```

You are not creating a wait set with one service "slot". Check the function [here](https://github.com/micro-ROS/rcl/blob/20e2496ddc846cdda38d25c1c284b23e24735230/rcl/include/rcl/wait.h#L132).

Close if you are ok with this answer, I'm going to add to this repo a service example!

----

Other considerations, services, and clients are not fully supported in the RCLC (the micro-ROS user API), and you have to use the RCL. This is going to be fixed as soon as we can, but by now please take care when mixing both APIs. I suggest that you can remove the ` rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100))` since you are actually doing a `rcl_wait()` previously.

Of course, you can mix the call to both, but please be aware of this.

pablogs9, 2020-11-03:
You have my working example [here](https://github.com/micro-ROS/micro_ros_arduino/pull/12)

anaelle-sw, 2020-11-03:
Hi! Thanks for these explanations, especially about the RCLC / RCL libraries. I was not really being careful with using one or the other.

I actually tried to use the function as used in the demo:
`rcl_wait_set_init(&wait_set, 0, 0, 0, 0, 1, 0, &context, rcl_get_default_allocator())`
This way, the service client did detect the service as available, but the micro-ROS agent wasn't not notifying that any request was received. So I thought I was closer to a solution using `&support.context`, since the request was received.
Anyway, your example server is working nicely for me, thanks!

Just a quick question before closing this issue please: in your example service server code, you initialize an executor in the `void setup()` function [here](https://github.com/micro-ROS/micro_ros_arduino/blob/1b00ac36393e849fb4c7f6c658c76d433527208f/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino#L48), and then in the `void_loop()` function you use a `rcl_wait()` [here](https://github.com/micro-ROS/micro_ros_arduino/blob/1b00ac36393e849fb4c7f6c658c76d433527208f/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino#L60). As you explained, it is not necessary to use `rclc_executor_spin_some()`. So why you actually need to initialize an executor anyway? Sorry if this question does not belong here, it may be quite related to "general" micro-ROS

pablogs9, 2020-11-03:
Hello @anaelle-sw,

You are right, if in the loop we are using a `rcl_waitset` directly, the executor is not needed. In fact, as you can see [here](https://github.com/micro-ROS/rclc/tree/master/rclc), the RCLC executor is just a sort of intelligent wrapper for the `waitset`, internally the executor handles its own waitset and allows the user setting callbacks, managing the execution order and so on...

So basically, we are using here the rcl_waitset because we don't have yet support for services in the executor. As soon as we merge [this](https://github.com/micro-ROS/rclc/pull/44), you can forget the waitsets and work with the RCLC API and its callbacks.

I hope that this is not a big inconvenient.

anaelle-sw, 2020-11-03:
Ok, I understand better now.
Actually, I am as new to RCL than to RCLC, so it won't be a big deal in my case. 
I will close this issue. Thanks for your time!

closed 2020-11-03 by anaelle-sw

pablogs9, 2020-11-25:
Related: Check [here](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino) the new service/client API in RCLC.

anaelle-sw, 2020-11-25:
Very nice to have service callbacks directly in the API. This way, it is really lighter and more intuitive to use. Thanks for notifying me about it!

anaelle-sw, 2020-11-25:
Just three quick questions about this new RCLC API and your example code:
_(if you want, I can open a new issue instead of posting here, let me know)_

1) In your example code, you declare a `rcl_wait_set_t` but don't initialize it, neither use it (as well as a `rmw_request_id_t` and a `rosidl_service_type_support_t` by the way). In my case, it works well if these are not declared. If I don't misunderstand, a wait set is not required anymore with the RCLC API, as you have an executor spinning (`rclc_executor_spin_some` in the loop), and RCLC directly handles this, is this right?

2) I don't need the `res` and `req` outside the service callback. Can I just don't declare these are global variables, and go with
```
rclc_executor_add_service(
                       &executor,
                       &service, 
                       &example_interfaces__srv__AddTwoInts_Request(), 
                       &example_interfaces__srv__AddTwoInts_Response(),
                       service_callback);
```
when initializing my service? My code builds this way, and everything seems to work normally. But can this lead to known problems? (I am thinking about memory leak for instance)

3) Is there a way to wait for the response to be sent, inside the callback? Actually, I need to copy strings inside `res->message`. As you told me in [this post](https://github.com/micro-ROS/micro_ros_arduino/issues/23#issuecomment-724497240), I cannot directly copy paste strings, but must use memory allocation and memory copy. For instance, when following this instruction, one of my services' callbacks looks like this:
```
void trigger_srv_cb(const void * req_in, void * res_in){
  std_srvs__srv__Trigger_Request * req = (std_srvs__srv__Trigger_Request *) req_in;
  std_srvs__srv__Trigger_Response * res = (std_srvs__srv__Trigger_Response *) res_in;
 
  res->message.data = (char*)malloc(200 * sizeof(char));
  res->message.size = 0;
  res->message.capacity = 200;
  memcpy(res->message.data, "trigger executed", 200);
  res->message.size = strlen(res->message.data);

  free(res->message.data);
}
```
The problem is I need to execute `free` after `malloc` to avoid memory leak. With the previous API, I call the function `rcl_send_response` before doing `free`. But with RCLC API (which is really more convenient, and that I am really willing to use) the response might not be sent before executing `free`. What is the correct way to handle this?

pablogs9, 2020-11-25:
Hello @anaelle-sw just I forgot to merge https://github.com/micro-ROS/micro_ros_arduino/pull/45, so:
1. solved in https://github.com/micro-ROS/micro_ros_arduino/pull/45

2. Wow, this is tricky. Where do you have defined `example_interfaces__srv__AddTwoInts_Response()` I'm not sure if they are pure C99 implementation or  Arduino C++. Usually you have to provide a req and res object to the executor to let it have some memory to save the req and res since the executor is not allowed to alloc memory. In Arduino I usually put all the variables as global since `setup()` stack variables can't be used in `loop()`... Does your solution work?

3. If you follow the example we provide, you have `req` and `res` declared as global variables. So I suggest you to do the 
```c
example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;

void setup(){
...
  res.message.data = (char*)malloc(200 * sizeof(char));
  res.message.size = 0;
  res.message.capacity = 200;
  RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
....
```
in the `setup()`. Then you provide those two object to the executor and you have guaranteed that in the callback the executor will pass you the couple of pointers that are the same that you provide: so the res is already init and allocated with the malloc. So in the callback you just do:
```c
void trigger_srv_cb(const void * req_in, void * res_in){
  std_srvs__srv__Trigger_Request * req = (std_srvs__srv__Trigger_Request *) req_in;
  std_srvs__srv__Trigger_Response * res = (std_srvs__srv__Trigger_Response *) res_in;
 
  memcpy(res->message.data, "trigger executed", 200);   <-- be careful with this 200, your string is not 200!
  res->message.size = strlen(res->message.data);
}
```

**Also be SUPER careful with copying not NULL-ending string because serialization for strings relies in a '\0' at the end... Maybe is safer a `fprintf(res->message.data, "%s", "blablabla")`.**

anaelle-sw, 2020-11-25:
2) Well, I didn't defined `example_interfaces__srv__AddTwoInts_Response()` actually, just included the file `example_interfaces/srv/add_two_ints.h`. Yes the code works this way, but I just tested it with your example code for now, with a few service calls.

3) Ok then, I will follow your advice and just use global variables (I am used to limit the number of global variables when programming ROS nodes for instance, but in the case of Arduino code I guess it is not a problem).

The problem with `fprintf(res->message.data, "%s", "blablabla")` is that you can only see this debug string via the serial monitor provided by Arduino IDE, is this right? But here we would need to get the `res->message` in ROS, or at least in a terminal. 

> ```c
>  be careful with this 200, your string is not 200!
> ```

This could be a quite big problem with our application. We need to regularly copy strings in a message (which is published by our board). The problem is the strings have different length. So having something like this:
 `memcpy(res->message.data, "trigger executed", sizeof("trigger executed"));`
, and later on something like this (on the same `res->message`): 
`memcpy(res->message.data, "foo", sizeof("foo"));`
, results on publishing a message like this: 
`"foogger executed"`.

My solution for this was to set a maximum string length(` #define MAX_LEN_STR 100`), and use it every time I need to copy a string:  
`memcpy(res->message.data, "trigger executed", MAX_LEN_STR);`. 
For now, it works very well with our string-containing message publication, and a ROS node processing the strings. The tests I ran for now lasted around 20min max, no problem related to the published strings occurred. What can happen if the size given in `memcpy` is not the one of the copied string?

pablogs9, 2020-11-26:
Sorry I didn't write the function well. I didn't mean `fprintf` I meant `sprintf`.

You are having the "foogger executed" string because you are not handling the '\0' character for null-ending the string. My solution for copying a string to a buffer is:

```c
sprintf(res->message.data, "%s", "blablabla")
res->message.size = strlen(res->message.data);
```

This writes "blablabla\0" to your buffer, so strlen will work to obtain the size.

> What can happen if the size given in memcpy is not the one of the copied string?

Multiple things:
- You are copying unnecessary bytes to a buffer, and at some point, this is inefficient
- Imagine that you are telling the code to copy `"blabla"` to `buf[100]` with `memcpy(buf, "blabla", 100);`. You know that `buf` has 100 slots to write but does the literal `"blabla"` (which is in the flash .const section) has 100 characters for reading? Imagine your string is stored at the end of the memory, in this case you will do an illegal read. Imagine you are in Linux, probably you will have a segfault if you read from a section of memory that does not belong to your process.
- Your string are only NULL-terminated if the buffer is previously set to 0

anaelle-sw, 2020-11-26:
Ok, I got it now. Thank you very much for all explanations!