micro-ROS/micro_ros_arduino #14: Custom messages/services generation

anaelle-sw, 2020-11-04:
Hello! I would like to generate custom messages (and then soon, custom services), in order to be able to import them as libraries in my source code .ino. I have some methods in mind, but I don't know exactly how to proceed to get it "cleanly" done.

## Setup
- Computer: ROS2 Foxy / Ubuntu 20.04
- Board: Teensy 3.1 / Arduino IDE 1.8.13 / Teensyduino 1.53

## Questions
So, first method would consists in running a script generating the libraries. We used to do so by using this [.py script](https://github.com/ros-drivers/rosserial/blob/noetic-devel/rosserial_arduino/src/rosserial_arduino/make_libraries.py) from rosserial_arduino package (as on ROS1, we use rosserial to communicate with our Teensy board). Would it be possible to use such kind of script to generate our custom messages/services for micro_ros_arduino? 

Second method is the one mentionned in [micro-ROS demos](https://micro-ros.github.io/docs/tutorials/core/create_new_type/), which generates the libraries (and custom libraries) at build time. This method requires a CMakeLists.txt file. But mircro_ros_arduino library doesn't currently use a CMakeLists.txt for building. Plus, it seems the libraries generated for "standard" micro-ROS are not supported by micro_ros_arduino, is this right? Is there any equivalent method for micro_ros_arduino, allowing to generate a custom library at build time?

Actually, I guess library generation is done when building micro_ros_arduino, as mentionned in the repo's readme with command line:
`docker run -it --rm -v $(pwd):/arduino_project micro_ros_arduino_builder`
Using this, where to put the .msg and .srv (or other?) source files to get them processed? We are not really used to dockers, is it possible to run this `micro_ros_arduino_builder`, but not inside your docker, and get our custom libraries generated?

Thanks for you help!

pablogs9, 2020-11-04:
Is an option for you to use Docker? The first idea I have is to prepare a docker file for building just the messages you want by providing them in a folder. 

If you don't like the idea of using Docker for this purpose I have to think of another solution...

In any case, tomorrow I'll write here a solution for including your custom messages in the current Arduino build system. Meanwhile, provide me feedback about which is your preferred option.

anaelle-sw, 2020-11-04:
We can sure use a Dockerfile from you (I just asked because it something we are not used to, but it should actually not be a problem). Thanks for you for giving such fast solutions!

pablogs9, 2020-11-05:
Ok, I have prepared something. Please push the latest version of this repository before continuing.

## Adding a new package to the build system

First, we are going to create a new package, I assume that you have a ROS 2 Foxy installation in your computer (let me know if not):

```bash
# Go to this library folder installation
cd ~/Arduino/libraries/micro_ros_arduino-0.0.1

# Go to the extra_packages folder and create there you new msg package:
pushd extras/library_generation/extra_packages

# Here follow the instructions of this tutorial: https://micro-ros.github.io/docs/tutorials/core/create_new_type/
ros2 pkg create --build-type ament_cmake my_custom_message
[...]
```

At the end of this procedure you should have your custom message inside the folder:

```bash
extras/library_generation/
├── ...
├── extra_packages
│   ├── extra_packages.repos
│   └── my_custom_message
│       ├── CMakeLists.txt
│       ├── include
│       │   └── my_custom_message
│       ├── msg
│       │   └── MyCustomMessage.msg
│       ├── package.xml
│       └── src
└── ...
```

Regarding this new `extra_packages` folder we have two possibilities:
- Adding directly folders that contain ROS 2 packages
- Adding an entry to [this file](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/extra_packages/extra_packages.repos), you can see a sample of this kind of files [here](https://github.com/micro-ROS/micro_ros_setup/blob/foxy/config/agent_uros_packages.repos). This option allows you to automatically download a Git repo every time the library is built.

_If the packages that you are going to add are standard enough, maybe we can add them to this repo [extra_packages.repos](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/extra_packages/extra_packages.repos).
Let me know._ 

## Rebuilding your own version of the micro-ROS library for Arduino

Now we are going to use docker to build our custom version of the library. In this case, this custom version of the library will have the extra msg that you just have included, but for example, is also possible to modify the configuration of the micro-ROS layers by customizing some of the [.meta files](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/colcon.meta).

```bash
# Go to this library folder installation
cd ~/Arduino/libraries/micro_ros_arduino-0.0.1

# Use the docker to build all the necessary stuff:
docker pull microros/micro_ros_arduino_builder:latest
docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:latest
```

I have added some new parameters to the docker script so if you use it like above it will build for all the supported platforms and it will take longer.

If you want to build just for your platform, use:

```bash
docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:latest -p teensy3
```

So, supposing that you have run the last command, building just for your `teensy3` and with your `my_custom_message` added, your library folder should look like:

```bash
pgarrido@pgarrido:~/Arduino/libraries/micro_ros_arduino-0.0.1$ git status
On branch foxy
Your branch is up to date with 'origin/foxy'.

Changes not staged for commit:
        modified:   available_ros2_types
        modified:   built_packages
        deleted:    src/cortex-m7/fpv5-sp-d16-softfp/libmicroros.a
        deleted:    src/imxrt1062/fpv5-d16-hard/libmicroros.a
        modified:   src/mk20dx256/libmicroros.a
        modified:   src/rmw_microxrcedds_c/config.h

Untracked files:
        extras/library_generation/extra_packages/my_custom_message/
        src/my_custom_message/
```

As you can see:
-  `src/imxrt1062/fpv5-d16-hard/libmicroros.a` (the Teensy 4 library) and `src/cortex-m7/fpv5-sp-d16-softfp/libmicroros.a` (the OpenCR library) has been removed due to the `-p teensy3` option.
- `src/mk20dx256/libmicroros.a` (the Teensy3 library) has been modified: now it has all the objects related to you custom package
- now you have all the headers files of your package here: `src/my_custom_message/`
- you should find you custom msg named in [available_ros2_types](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/available_ros2_types)

---

Let me know if this is ok for your purposes. If you have any better idea, we are open to improvements.

anaelle-sw, 2020-11-05:
Nice, it works perfectly well.

I confirm I have a ROS 2 Foxy installation on my computer.

> If the packages that you are going to add are standard enough, maybe we can add them to this repo [extra_packages.repos](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/extra_packages/extra_packages.repos).

--> Thanks, but for now we would rather keep our messages in our own repo.

> If you have any better idea, we are open to improvements.

--> I will discuss about it with my co-workers, and let you know. But it seems really great to me this way!

It's all good, I will close this issue. Thank you!

closed 2020-11-05 by anaelle-sw

grassjelly, 2021-07-21:
Hi @pablogs9 thanks for your detailed tutorial how to rebuild micro-ROS library for Arduino for custom types.

I followed the steps how to add the custom types to the library and ran this command to generate a new static library:

`docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:latest
`

but it seems to be outdated so I checked your workflow script and found [this](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/.github/workflows/library_generation.yml#L32). I tried to run the command below but I'm not sure if this the right direction to go:

`docker run --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:latest -p teensy3`

Thanks!

pablogs9, 2021-07-21:
Yes... this is outdated, would you mind creating a PR modifying the README.md?

Thanks a lot for the feedback!

grassjelly, 2021-07-21:
Sorry my bad. I was looking at the commands you replied in this thread. This https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library is actually updated.

Cheers!