micro-ROS/micro_ros_arduino #21: no ros2 topics visible

nfry321, 2020-11-06:
**Describe the bug**
The micro-ros agent is able to communicate with a Teensy 3.2, but the rest of ROS2 is not able to connect to the agent.
I.e. `ros2 topic list` & `ros2 node list` show nothing when using micro-ros_publisher example.

**To Reproduce**
Following instructions in readme:
1. Download pre-compiled library (.zip)
2. Include it in your project using Sketch -> Include library -> Add .ZIP Library...
3. Patch Teensyduino
4. Upload publisher example to teensy using arduino ide
(The pub and sub example sketches are listed under incompatible in my examples menu and I get `WARNING: library micro_ros_arduino-0.0.1 claims to run on OpenCR, Teensyduino architecture(s) and may be incompatible with your current board which runs on avr architecture(s).` but they upload fine)
5. run micro-ros agent `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`
6. run `ros2 topic list` `ros2 node list`
[image]
Screenshot shows that messages are sent and received.

**Expected behaviour**
For `micro_ros_arduino_node_publisher` to be visible when running `ros2 topic list`
(Although it is entirely possible I have misunderstood what the example is meant to do!)
 
**System information**
 - OS: Ubuntu 20.04
 - ROS 2 Foxy
 - Teensy3.2
 - Arduino IDE 1.8.13
 - Teensyduino 1.53

**Additional context**
(https://github.com/micro-ROS/micro_ros_setup/issues/179) could be relevant, but I am not opening the serial port twice and the teensy only has one available.

nfry321, 2020-11-06:
Sorry this is a duplicate of a closed issue which I missed #7 
I too had my `ROS_DOMAIN_ID` set. Clearing it stopped the issue.

This does raise a related question, if I want to use this as part of a robot, e.g a Pi connects to a teensy, using micro-ros, on a mobile base, and the pi connects to a laptop via wifi, am I unable to set my `ROS_DOMAIN_ID`, in which case I may get issues with other users on my network?

pablogs9, 2020-11-06:
Hello @nfry321, thanks for using micro-ROS.

Nice to hear that you are able to see your topics being published. Regarding the `ros2 node list` it won't work since ROS 2 Foxy changed the approach the ROS 2 graph is managed and we are working on solving it. Really soon we will have a new version of the agent that is able to manage the micro-ROS graph and you will see your nodes in the `ros2 node list`

Regarding the `ROS_DOMAIN_ID` it should be possible to change it by using [custom QoS](https://micro-ros.github.io/docs/tutorials/core/create_dds_entities_by_ref/). If you create a custom QoS participant using [this configuration](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/domainparticipant.html#domainparticipant-configuration) and the configure micro-ROS to use refs you will have ROS 2 nodes with custom domain ID. 

[This comment](https://github.com/micro-ROS/micro_ros_arduino/issues/20#issuecomment-723128120) is related to this.

I know that this is a quite tricky procedure for changing such a basic thing as the node domain ID, so I'm going to keep this open in order to remember to implement somehow this domain id configuration in the default XML entities generation in micro-ROS. 

If you are curious [this is the XML used by the client](https://github.com/micro-ROS/rmw-microxrcedds/blob/b969cca2fa1a7d5e18e87996b9b82bef59837801/rmw_microxrcedds_c/src/utils.c#L30) to create the micro-ROS participant-node. But modifying this for the Arduino port can be tricky... so I will come back here with a better solution.

nfry321, 2020-11-06:
Thanks for the comprehensive reply, and for this release in general. Mirco-ros on a teensy really makes it accessible.
It's good to know that the domain ID can be configured, I'll look through those links.

pablogs9, 2020-11-10:
Hello @nfry321, I have updated this library to [v0.0.2](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v0.0.2) and now you can change the Domain ID of your nodes using something like:

```c
...
rcl_node_options_t node_ops = rcl_node_get_default_options();
node_ops.domain_id = 10;
RCCHECK(rclc_node_init_with_options(&node, "my_node_name", "", &support, &node_ops));
...
```

I'm closing since resolved. Let me know if this works for you and do not hesitate to reopen if you have some other question. 

CC: @anaelle-sw this solves also some of your old issues

closed 2020-11-10 by pablogs9

nfry321, 2020-11-10:
That works great, thanks for the quick update!

OtaviodaCruz, 2022-01-27:
> Olá @nfry321 , obrigado por usar o micro-ROS.
> 
> É bom saber que você pode ver seus tópicos sendo publicados. Em relação ao `ros2 node list`que não vai funcionar desde que o ROS 2 Foxy mudou a abordagem do gráfico do ROS 2 é gerenciado e estamos trabalhando para resolvê-lo. Muito em breve teremos uma nova versão do agente que é capaz de gerenciar o gráfico micro-ROS e você verá seus nós no`ros2 node list`
> 
> Em relação ao `ROS_DOMAIN_ID`, deve ser possível alterá-lo usando [QoS personalizado](https://micro-ros.github.io/docs/tutorials/core/create_dds_entities_by_ref/) . Se você criar um participante de QoS personalizado usando [esta configuração](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/domainparticipant.html#domainparticipant-configuration) e configurar o micro-ROS para usar refs, você terá nós ROS 2 com ID de domínio personalizado.
> 
> [Este comentário](https://github.com/micro-ROS/micro_ros_arduino/issues/20#issuecomment-723128120) está relacionado a isso.
> 
> Eu sei que este é um procedimento bastante complicado para alterar uma coisa tão básica como o ID de domínio do nó, então vou manter isso aberto para lembrar de implementar de alguma forma essa configuração de ID de domínio na geração de entidades XML padrão em micro- ROS.
> 
> Se você estiver curioso, [este é o XML usado pelo cliente](https://github.com/micro-ROS/rmw-microxrcedds/blob/b969cca2fa1a7d5e18e87996b9b82bef59837801/rmw_microxrcedds_c/src/utils.c#L30) para criar o nó participante do micro-ROS. Mas modificar isso para a porta do Arduino pode ser complicado... então voltarei aqui com uma solução melhor.

Hello everybody. By chance, is it already possible to visualize the nodes? I'm developing a ros2 foxy application where my nodes don't appear, but the topics work correctly, I was kind of lost.

I just want to know to understand if the error is on my side.

pablogs9, 2022-01-27:
@OtaviodaCruz can you open a new issue explaining your use case and problems?

OtaviodaCruz, 2022-02-03:
Hi @pablogs9 , sorry for the delay. Yes I can, I will create

josh-001, 2023-12-07:
> Hello @nfry321, I have updated this library to [v0.0.2](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v0.0.2) and now you can change the Domain ID of your nodes using something like: [...]

exit status 1
'rcl_node_options_t' has no member named 'domain_id'

i am getting this error in arduino
ubuntu 22 
ros2 humble version