micro-ROS/micro_ros_arduino #26: Add support for Arduino Portenta

mbanzi, 2020-11-11:
Hi this is great, how about supporting an actual Arduino board. The new Portenta is super powerful (Dual core Cortex M7+M4, wifi,ble and a bunch of other stuff on a small board) and would be a great addition to this project. Would you consider supporting it if we send you some hardware? Let me know :)

pablogs9, 2020-11-12:
Hello @mbanzi, this new Portenta looks super cool. Of course, we will support it and it would be awesome if you send us some boards. Please reach me at pablogarrido@eprosima.com and I can give you our address.

Thanks for the interest and for sure I will come back soon with the news!

pablogs9, 2020-12-01:
Friendly reminder @mbanzi. We are suuuuuuper excited about these new boards.

mbanzi, 2021-02-02:
Sorry for the delay I'll follow up today :)

pablogs9, 2021-02-04:
We just have received it! Let's try the micro-ROS ports. Thanks a lot, @mbanzi and Arduino team!

[image]

ralph-lange, 2021-02-04:
Does it come with a built-in LoRa module? The box says "LoRa Connectivity", but I could not find it in the spec.

pablogs9, 2021-02-04:
Yes @ralph-lange, it seems that the LoRa module is in the shield: https://store.arduino.cc/portenta-vision-shield

pablogs9, 2021-02-04:
Closing due to https://github.com/micro-ROS/micro_ros_arduino/pull/65

closed 2021-02-04 by pablogs9