micro-ROS/micro_ros_arduino #48: Odometry publisher breaks the program

lukicdarkoo, 2020-11-29:
**Describe the bug**
Calling `rcl_publish(&publisherOdom, &msgOdom, NULL)` crashes the program.

**Firmware**
I tried to create a minimal example just by adding odometry publisher to the provided publisher example:
https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_publisher/micro-ros_publisher.ino

<details>
<summary>ROS2 Odometry Publisher</summary>

```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <nav_msgs/msg/odometry.h>
#include <std_msgs/msg/int32.h>

rcl_publisher_t publisherOdom;
rcl_publisher_t publisherInt32;
nav_msgs__msg__Odometry msgOdom;
std_msgs__msg__Int32 msgInt32;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisherInt32, &msgInt32, NULL));
    rcl_publish(&publisherOdom, &msgOdom, NULL);    // <<<<< This line crashes the program!
  }
}

void setup() {
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
[... 310 lines cut ...]
0000: 81 80 0E 00 07 01 08 00 00 18 00 15 00 00 00 00
[1606655029.440099] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 4, data: 
0000: 00 00 00 00
[1606655029.440228] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0F 00 00 00 80
[1606655030.441554] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 16, data: 
0000: 81 80 0F 00 07 01 08 00 00 19 00 15 00 00 00 00
[1606655030.441781] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 4, data: 
0000: 00 00 00 00
[1606655030.441934] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 10 00 00 00 80
[1606655031.443977] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 16, data: 
0000: 81 80 10 00 07 01 08 00 00 1A 00 15 00 00 00 00
[1606655031.444233] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 4, data: 
0000: 00 00 00 00
[1606655031.444397] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 11 00 00 00 80
[1606655032.446012] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 16, data: 
0000: 81 80 11 00 07 01 08 00 00 1B 00 15 00 00 00 00
[1606655032.446224] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 4, data: 
0000: 00 00 00 00
[1606655032.446345] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 12 00 00 00 80
```
</details>

pablogs9, 2020-11-30:
Working on it, I think we have found a bug. I'll come back with a fix in a while.

pablogs9, 2020-12-01:
Update: we found something in the serialization library [check here](https://github.com/eProsima/Micro-CDR/pull/55).

We are merging soon and updating this library.

I have checked your example building the library with this branch and works fine, some other check are required before merging, I'll alert you here when ready.

pablogs9, 2020-12-01:
Merged https://github.com/eProsima/Micro-CDR/pull/55 and https://github.com/micro-ROS/micro_ros_arduino/pull/50. Please check if now your setup works and give us feedback.

Close if resolved please.

pablogs9, 2020-12-14:
Is this still an issue?

lukicdarkoo, 2020-12-14:
I am sorry, I haven't had time to test it. You can close the issue and I will reopen it if the bug persists (once I find time to test it).

pablogs9, 2020-12-15:
Ok @lukicdarkoo, I'm closing. Please reopen if the issue continues when you can give it a try. And thanks a lot for the feedback!

closed 2020-12-15 by pablogs9