micro-ROS/micro_ros_arduino #56: Timer timeout bellow 500ms doesn't work

lukicdarkoo, 2020-12-06:
The timer calls callback at 2 Hz even though the timeout is less than 0.5s.

**To Reproduce**
1. Take the publisher example and reduce the timer timeout to 100ms.
```diff
$ diff orig.ino new.ino 
64c64
<   const unsigned int timer_timeout = 1000;
---
>   const unsigned int timer_timeout = 100;
79d78
<   delay(100);
```
2. Start the MicroROS agent
```
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -b 115200
```
3. Listen for messages
```
$ ros2 topic hz /micro_ros_arduino_node_publisher
average rate: 1.992
	min: 0.002s max: 1.002s std dev: 0.50018s window: 2
average rate: 1.992
	min: 0.002s max: 1.002s std dev: 0.50020s window: 4
average rate: 1.992
	min: 0.002s max: 1.002s std dev: 0.50017s window: 6
average rate: 1.992
	min: 0.002s max: 1.002s std dev: 0.50016s window: 8
```

**System information (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS 2 Foxy
 - Board: Arduino Zero

nfry321, 2020-12-08:
As part of trying different QoS, I was just testing these limits and could not reproduce this issue.
With the default publisher it maxed out at ~83Hz.
Swapping to a best effort publisher (`rclc_publisher_init_best_effort`) I got 904Hz with the timer timeout set to 1ms. 

Note three differences in my setup are I'm using a Teensy 3.2, my agent is not run using a docker, and I have a domain id set.

pablogs9, 2020-12-09:
Hello @lukicdarkoo, I also I'm testing with the Teensy 3.2 and reducing the timer timeout corresponds with a higher rate at the `ros2 topic hz`.

In fact, applying your patch gives a quite accurate rate: 

[image]

I'm wondering if there is something with the transports. Is this working for you with best effort publisher `rclc_publisher_init_best_effort` as @nfry321 is suggesting?

lukicdarkoo, 2020-12-28:
I have just tested on Portenta H7 and it easily reaches 1000Hz even with `rclc_publisher_init_default`. The same code on Arduino Zero publishes at 2Hz. It seems it is specific to the board, I will investigate.

pablogs9, 2021-01-11:
Cool, let us know when you have results.

I'm going to check your Portenta port ASAP.

HazemAHMED, 2021-06-16:
Hey, I am facing the same Problem on the Publisher Example Code on Arduino Due and ROS FOxy. When I changed the rclc_publisher_init_default to rclc_publisher_init_best_effort I got around 500 HZ It worked. But My problem That I am publishing Two topics while following this example https://micro.ros.org/docs/tutorials/core/programming_rcl_rclc/. Whatever I value of Timer I change  I can't get a Frequency over 10 Hz even when changed rclc_publisher_init_default to rclc_publisher_init_best_effort.

negre, 2021-09-21:
Hi,
I have the same problem on the Arduino Nano RP2040 (with ros foxy or Galactic). Using "rclc_publisher_init_best_effort" doesn't solve the problem, the timer cannot have a frequency higher than 2Hz.

pablogs9, 2021-09-22:
@negre can you open an issue in the Raspberry Pi Pico repo and post your code so we can replicate it?

Acuadros95, 2021-09-23:
This should be fixed on the new release: [v2.0.1](https://github.com/micro-ROS/micro_ros_arduino/releases).
Check https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/228 for the fix.

I am closing the issue, if the problem persists please reopen with feedback.

closed 2021-09-23 by Acuadros95