micro-ROS/micro_ros_arduino #111: micro_ros_arduino on Portenta interfaced with Jetson Xavier NX

guilleherreraf, 2021-02-09:
Hello guys! I'm currently working on a project using ROS2 Dashing installed on a Jetson Xavier NX to do some high-level robotics control. For the low-level controller I'm using a Portenta H7 and I will like to interface the Jetson Xavier NX with the Portenta via Ethernet (with the Vision Shield). I want to be able to send ROS2 messages to the Portenta, and then perform other processes with this information.

Will this be possible with your library? As I see that you work with the ROS2 Foxy version, and I have Dashing installed on my Jetson NX. I would appreciate any guidance. 

Thanks in advance!

pablogs9, 2021-02-10:
Hello @guilleherreraf, yes with this package you should be able to build a micro-ROS application for Arduino Portenta H7, the micro-ROS application will run in the M7 core.

This micro-ROS application will be communicated with a micro-ROS Agent in charge of handling the ROS 2 communication, you can read more about this architecture [here](https://micro-ros.github.io//docs/overview/features/).

Regarding the compatibility between Foxy and Dashing, one of the things that can be critical is the versions of the ROS 2 middleware. Have you tried to communicate Foxy with Dashing with some examples?

guilleherreraf, 2021-02-10:
Hello @pablogs9 thanks for your quick response!
I have not yet tried to communicate Foxy with Dashing. On the other hand, in order to program the Portenta I'm using a Windows 10 machine with the Arduino IDE. I have also a virtual machine (with Ubuntu 20.04 and Foxy installed). Maybe I could try to install the micro-ROS package on the Portenta through the Virtual Machine.... However, I will still have the compatibility issue (Foxy and Dashing). Do you think it is something I will be able to do?

pablogs9, 2021-02-10:
You can just install the micro-ROS library for Arduino in your Arduino IDE... Just download this repo as a .zip and use the Add library from .zip...`

guilleherreraf, 2021-02-10:
So if I understand correctly, using the Arduino IDE (with the installed micro_ros_arduino library) to flash the Portenta I do not need to follow all the cross-compilation process you describe on your website (create, configure, build and flash)? I can just open the publisher example, upload it to the Portenta, and then connect the Portenta via Ethernet on a machine running Linux and I will be able to subscribe to the corresponding topic? (Running echo for instance)

pablogs9, 2021-02-10:
Well:
- Using the micro-ROS module for Arduino you don't need to use the micro-ROS build system (create, configure...) because you have precompiled versions for some platforms.
- micro-ROS needs a micro-ROS Agent in order to talk with the ROS 2 world, so in your Linux machine you will need an agent. We have a dockerized version: `docker run -it --rm --net=host microros/micro-ros-agent:foxy [args]`
- micro-ROS module for Arduino by now only supports serial transports, so you will need to connect your Portenta with a USB-C cable to the machine with the Agent.
- For example, opening an agent for the serial transport in the Portenta should be something like: `docker run -it --rm --net=host microros/micro-ros-agent:foxy serial [PORTENTA SERIAL DEVICE] -v6`
- If you want to have UDP transports (for the communication between client and agent) you will need to implement [these functions](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/src/default_transport.cpp) accordingly. If you need help with this we can guide you, and if contribute them to this repo it will be awesome.

You can read more about the micro-ROS architecture here: https://micro-ros.github.io//docs/overview/features/

pablogs9, 2021-02-10:
And yes, once you have a micro-ROS client connected to the micro-ROS agent you will be able to publish, subscribe, create services/clients, having executors, and some other cool features.

guilleherreraf, 2021-02-10:
Ok @pablogs9 I understand. Well first of all I will try to have the serial example running, and from there try to adapt it to my system requirements. My main concern is the Foxy-Dashing compatibility. But well, I'll see how it works.
Thanks a lot again and I'll keep you posted!

pablogs9, 2021-02-10:
I have done a quick test and it seems that at least for basic types and pub/sub it works:
[image]

guilleherreraf, 2021-02-10:
That is great! I will try to do the same today and let you know how it goes!

guilleherreraf, 2021-02-10:
@pablogs9 I tried running the docker container in the NX but it throws the follwing error, due to the used architecture:

> The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested

Is there a parameter to specify a given architecture?

pablogs9, 2021-02-10:
You need to run this docker in a Linux machine

guilleherreraf, 2021-02-10:
I have a version of Ubuntu on my Jetson NX, the only difference is that it has an architecture different from amd64. 
Anyway, For now I'm trying with the Portenta connected to my machine running the VM with Ubuntu 20.04 and ROS Foxy. I uploaded the publisher example with the Arduino IDE and I have tried both running the client with and without Docker, with the following commands respectively:

No Docker: `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`
Docker: `sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

In both cases I get the following in the terminal:
[image]

I do not get any other message and if I try running `ros2 topic list` I only see these topics:
[image]

jamoralp, 2021-02-10:
Hello, Ubuntu is running under a virtual machine? Because in that case, you should first check that you have full duplex communication between your host machine and the virtualized Ubuntu machine. Otherwise, the agent won't be able to receive requests from the agent or answer to them.

guilleherreraf, 2021-02-10:
Hello @jamoralp. I'm not sure, I'll look into that. Thanks for the suggestion!

B00M3RSIX, 2021-05-09:
Hi.
Im running the agent-docker on my NX. But because NX uses Ubuntu 18.04 you can not run the docker of micro-ros by default (as it based on Ubuntu 20.04).
However I could run it when using a other base docker. 
You have to modify the first line in the micro-ros docker-file to:
`FROM arm64v8/ros:foxy`
However I don't know about the Dashing/Foxy combatibility.

samiamlabs, 2021-07-27:
Are you still having issues running this on the NX @guilleherreraf?

Like @B00M3RSIX I have also been able to run the serial agent in containers based on `arm64v8/ros:foxy` and `arm64v8/ros:galactic` (in my case on a AGX Xavier) without any issues.

I also tried to compile the `micro-ros-agent` for foxy on the 18.04 based system from the newest JetPack, but gave up after a while. It might be possible but seemed to be pretty tricky. I haven't tried to install the dashing version of  `micro-ros-agent`, but it will probably work since the foxy version worked on aarch64.

Nvidia is planning to release a 20.04 based OS as a preview for Jetson Q1 2022 (see https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2h-2021-and-2022/177723)
Using the `micro-ros-agent` for foxy/galactic directly on the main OS or in their special docker images that support hardware-accelerated graphics should be pretty straightforward after that release.

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9