micro-ROS/micro_ros_arduino #117: Build errors 'undefined reference' with ROS2 Rolling release

anaelle-sw, 2021-02-11:
Hi! We are currently trying to migrate our project from ROS2 Foxy to Rolling. The problem is I am not able to build any Arduino code once micro-ROS Rolling is installed.

# Setup
- Ubuntu 20.04, ROS2 Rolling
- Arduino 1.8.13

# Steps to reproduce
I wanted to re-installed micro-ROS from scratch to be sure no Foxy code will be remaining and creating some conflict. So I removed the library installation at `~/Arduino/libraries/micro_ros_arduino` and did this:
```c
# patch teensyduino
cd ~/arduino-1.8.13/hardware/teensy/avr/
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/main/extras/patching_boards/platform_teensy.txt > platform.txt

# install micro_ros_arduino
cd ~/Arduino/libraries/micro_ros_arduino
sudo docker pull microros/micro_ros_arduino_builder:rolling
sudo docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:rolling
```
The build went well. Then I open an example from this repo with Arduino IDE and try to build it.

# Issue
The build of the .ino code failed with error `undefined reference` for every rclc function. For instance, when building the [publisher example](https://github.com/micro-ROS/micro_ros_arduino/blob/main/examples/micro-ros_publisher/micro-ros_publisher.ino), I got the following errors: 
```c
/tmp/arduino_build_453459/sketch/micro-ros_publisher.ino.cpp.o: In function `timer_callback(rcl_timer_t*, long long)':
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/tmp/arduino_build_453459/sketch/micro-ros_publisher.ino.cpp.o: In function `set_microros_transports':
/home/anaelle/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino_build_453459/sketch/micro-ros_publisher.ino.cpp.o: In function `setup':
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
/tmp/arduino_build_453459/sketch/micro-ros_publisher.ino.cpp.o: In function `loop':
/tmp/arduino_modified_sketch_797093/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
```
I double checked that I was building the example code from the main branch, and that the library installation was complete.
Did I made some mistakes switching from Foxy to Rolling? Are some includes missing?
Thank you very much for your help!

pablogs9, 2021-02-11:
Hello @anaelle-sw I just have tested if I'm able to build the int32_publisher example with `main` branch. In fact, [it is built in the CI](https://github.com/micro-ROS/micro_ros_arduino/blob/84a2134dd4243137fd658ce4c375c963efa992f1/.github/workflows/ci.yml#L65) and [yesterday morning it has passed](https://github.com/micro-ROS/micro_ros_arduino/pull/113).

I'm going to replicate your steps in a clean environment (using the Arduino CLI, I guess that it will work with all tools) and I will come back in a while.

In any case, Rolling is an unstable release, do you have any good reason for using it?

anaelle-sw, 2021-02-11:
Ok, thank you. I have build micro-ROS yesterday evening by just doing the `docker pull` and `docker build` steps. And I just have it tried again this morning, but after removing the previous installation.

Yes, for now, Rolling is quite interesting for us because of some features that won't be available on Foxy.

pablogs9, 2021-02-11:
This is a clean installation and builds of the publisher demo using `main` for Teensy 3.1/3.2: 

https://asciinema.org/a/QBYuYOEdMoAHguv0DMbBhhgb7

I do a couple of tricks here, let me explain:
- 00:00 to 00:32: install dependencies in docker
- 00:32 to 01:12: install arduino-cli, install a random library for creating the required folders, and then I clone this repo in branch `main`
- 01:12 to 03:58: install teensy support in Arduino CLI in the most unorthodox way (install the teensy support for the Arduino IDE and copying the required files in the Arduino CLI installation)
- 03:58: patching the `platform_teensy.txt`
- 04:41: check the boards are installed ok
- 04:56: first build of the example. fails because it cant open the teensy loader (GUI) inside the docker
- 06:31: fake the `teensy_post_compile` with a true to avoid the teensy loader being used
- **06:38: successful build of the example**

I know that this is a bit tricky and you don't need to do all these steps. I have followed what we do in the CI: https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/.github/workflows/ci.yml

Now, I'm going to replicate the step of regenerating the library like you are doing, with:

```bash
sudo docker pull microros/micro_ros_arduino_builder:rolling
sudo docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:rolling
```

anaelle-sw, 2021-02-11:
Thanks for your answer, I will try to follow these steps and will let you know.
Yes we need custom messages and to change the .meta file. After cloning the repo, this is what I do: 
```c
# custom build
cat ~/project_repo/src/.../colcon_lowmem.meta > ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta
# custom messages
cp -r ~/project_repo/src/custom_msgs_pkg ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/extra_packages
```
And then I run `docker pull` and `docker run` steps

pablogs9, 2021-02-11:
ok, maybe your library building is failing for some reason.

Could you do the following steps?
1. Clone this repo: `git clone -b main https://github.com/micro-ROS/micro_ros_arduino` in your library folder
2. Add your custom meta and messages
3. Paste here the output of an `git status`
4. Run `sudo docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:rolling`
5. Paste here the output of another `git status`

Maybe an error in your meta or in any of your messages is making the build fail. Is possible also to paste here the output of the `docker run` command?

anaelle-sw, 2021-02-11:
Actually, the .meta is the same that I used before for Foxy version! II didn't see that it has some changes for Rolling. I changed this and let you know whether it solves the problem. 

First `git status` returns:
```c
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   extras/library_generation/colcon_lowmem.meta

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	extras/library_generation/extra_packages/wyca_msgs/

no changes added to commit (use "git add" and/or "git commit -a")
```
The library is currently building, it may take a dozen of minutes...

pablogs9, 2021-02-11:
In order to reduce the library building time remember to use `-p` to just build the library for your platform.

```
sudo docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:rolling -p teensy32
```

Regarding the meta, please share it here, we have changed a couple of things this week regarding the transports.

anaelle-sw, 2021-02-11:
The meta I had before:
```c
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=ON",
                "-DUCLIENT_EXTERNAL_SERIAL=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=3",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=2",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom_serial"
            ]
        }
    }
}
```
The new one (based on the current version in main branch):
```c 
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=3",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=2",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

anaelle-sw, 2021-02-11:
So, it is entirely my bad, the .ino code builds properly when micro-ROS is built with the correctly updated meta file. 
Sorry for bothering you because of this mistake. Thanks a lot for helping

closed 2021-02-11 by anaelle-sw

pablogs9, 2021-02-11:
Ooook don't worry! Please give us feedback about the Rolling release, you are the only ones using it, as far as I know.