micro-ROS/micro_ros_arduino #123: No message passing through the agent with ROS2 Rolling

anaelle-sw, 2021-02-12:
Good morning! We are currently trying to migrate our project to ROS2 Rolling, but I have some trouble making messages pass through the agent.

# Setup
- Ubuntu 20.04
- ROS2 Rolling
- Teensy 3.2

# Steps to reproduce
As we are using a Teensy board, I previously installed the micro-ros-arduino lib for Rolling. 

First, I tried to use the agent installed from snap. On Foxy, before migrating to Rolling, the agent was installed from snap release, and everything used to work nicely. I re-install it from snap, having sourced Rolling as `ROS_DISTRO`. Then I run the agent (`micro-ros-agent serial --dev /dev/teensy_elodie -v6` ) and upload on a Teensy board the [publisher example](https://github.com/micro-ROS/micro_ros_arduino/blob/main/examples/micro-ros_publisher/micro-ros_publisher.ino). Here's the output of the agent: 
```c
[1613119649.240504] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1613119649.740824] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1613119649.741063] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1613119650.320015] info     | Root.cpp           | create_client            | create                 | client_key: 0x5083E048, session_id: 0x81
[1613119650.320126] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1350819912, address: 0
[1613119650.320291] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5083E048, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
```
And then it stops.
I run `ros2 topic list` and `ros2 node list` in another shell, but nothing is returned.

As a second attempt, I tried to install the agent from source main branch, following these steps:
```c*
mkdir uros_ws && cd uros_ws
git clone -b main https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
rosdep update && rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh host
ros2 run micro_ros_setup build_firmware.sh
source install/local_setup.bash
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
```
<i> Note: When running `ros2 run micro_ros_setup create_agent_ws.sh`, an output surprised me:
```c
=== ./eProsima/Micro-CDR (git) ===

Already on 'foxy'
Your branch is up to date with 'origin/foxy'.
```
Not sure if this is normal, as I tried to build the agent for Rolling </i>

Then I run the agent (`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/teensy_elodie -v6`), and upload the same publisher example as before. The output is the same.

# Issue description
The agent returns that the session with the board is established, which is great. But only one message seems to be sent. Plus, ROS doesn't not find the topic nor the micro-ROS node. 
No error seems to happen on the board side, as the board's LED is not blinking.

# Questions
- Is the micro-ros-agent snap release adapted for being used with ROS2 Rolling? If not, is a Rolling snap release planned? (we would be very interested in this)
- Is the main branch corresponding to Rolling version? Can you help me fix the communication between the board, the agent, and ROS?

Thank you very much for your help!

pablogs9, 2021-02-12:
I did a mess with the new external transport API, these are the fixes:
- https://github.com/micro-ROS/micro_ros_arduino/pull/122
- https://github.com/micro-ROS/micro_ros_arduino/pull/121

I'm merging as soon as CI pass.

Please checkout this new branch and test it... This is not related to the library but with the Arduino code, so I guess that maybe you can apply the changes and use the same library that you have precompiled.

Please close if this solves the issue.

anaelle-sw, 2021-02-12:
Thank you very much for the quick fix! It works really great, with the agent installed from snap as well as installed from sources!

closed 2021-02-12 by anaelle-sw

pablogs9, 2021-02-12:
Cool! You are using Rolling with the default middleware?

anaelle-sw, 2021-02-12:
Yes for now we are using CycloneDDS implementation. We didn't yet really discuss this, so I have no idea whether it will be definitive.

nfry321, 2021-02-12:
I was just drafting an issue when I saw this, thanks for the quick fix!