micro-ROS/micro_ros_arduino #142: request documentation: how add new boards and custom messages?

gbr1, 2021-02-26:
Hi! Thank you for your project! It is very useful!
I'm testing micro-ros for arduino to use as a  "rosserial" for ROS2.
I would like to add to other arduino compatible boards like Bluepill (stm32f103) and Iot node L475, but I'm not able to find any documentation on how to add. :'(
I need also to add custom topic formats (e.g. custom joint messages) but I don't understand how I can add them.
Sorry for the "noob" issue.
Thank you in advance.

pablogs9, 2021-02-26:
Hello @gbr1,

currently, we don't have tutorials on these topics but:
- You can find how to add new messages [here](https://github.com/micro-ROS/micro_ros_arduino/issues/14#issuecomment-722242175)
- You can find how to add support for a new board [here](https://github.com/micro-ROS/micro_ros_arduino/pull/98/files), it relies on the `generate_lib` tool of [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup) and you can find a tutorial about this [here](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/).

Let me know if you need more support.

gbr1, 2021-02-26:
@pablogs9 thank you so much!

closed 2021-03-12 by pablogs9