micro-ROS/micro_ros_arduino #145: Problem with building firmware for Teensy 4.0 and micro-ros

masoudir, 2021-02-26:
## Issue 

I have issue with building micro-ros-arduino for teensy 4.0.

Errors:

Linking .pio/build/teensy40/firmware.elf
.pio/build/teensy40/src/main.cpp.o: In function `timer_callback(rcl_timer_t*, long long)':
main.cpp:(.text._Z14timer_callbackP11rcl_timer_tx+0xc): undefined reference to `rcl_publish'
.pio/build/teensy40/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0x16): undefined reference to `rmw_uros_set_custom_transport'
main.cpp:(.text.setup+0x36): undefined reference to `rcutils_get_default_allocator'
main.cpp:(.text.setup+0x4c): undefined reference to `rclc_support_init'
main.cpp:(.text.setup+0x5a): undefined reference to `rclc_node_init_default'
main.cpp:(.text.setup+0x60): undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
main.cpp:(.text.setup+0x6c): undefined reference to `rclc_publisher_init_default'
main.cpp:(.text.setup+0x80): undefined reference to `rclc_timer_init_default'
main.cpp:(.text.setup+0x90): undefined reference to `rclc_executor_init'
main.cpp:(.text.setup+0x9a): undefined reference to `rclc_executor_add_timer'
.pio/build/teensy40/src/main.cpp.o: In function `loop':
main.cpp:(.text.loop+0x14): undefined reference to `rclc_executor_spin_some'

#### Steps to reproduce the issue

I have cloned micro-ros-arduino repo and I have used platformio for building it. I have put uros_lib inside "lib" folder of platformio.
I have included <Arduino.h> and then I have copied contents of [this](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_publisher/micro-ros_publisher.ino) in main.cpp.

I know that `teensy_transports.c.in` is for mapping some functions of uros for arduino and also I have tried to cut its contents inside main.cpp, but nothing has changed.

How can I build correctly?

The mentioned functions (rclc_executor_spin_some, rclc_executor_init, ...) are also defined some where inside uros_lib but why they are not detected for compiler?

pablogs9, 2021-02-26:
Platformio is not currently supported. Please use Arduino IDE or Arduino CLI.
And remember to [patch the teensyduino addon](https://github.com/micro-ROS/micro_ros_arduino#patch-teensyduino)

closed 2021-03-01 by pablogs9

DaniGarciaLopez, 2021-11-25:
Any updates? I would like to also use platformIO instead of Arduino IDE or CLI and it throws the same errors mentioned here. I saw some recent issues mentioning platformIO as the used IDE so i bet it is compatible. Any hints on how to configure the project are welcome?

Thanks!

masoudir, 2021-11-25:
Hi @DaniGarciaLopez , you can use my github page [link](https://github.com/masoudir/teensy41_uros_example_publisher_with_static_library) in order to get help. Of course it had been mention in #414 as well.

pablogs9, 2021-11-26:
Hello @DaniGarciaLopez and @masoudir currently we do not have time to implement a solution for Platform.IO but it would be great if you contribute a possible solution so we can integrate it into the building ecosystem of micro-ROS.

@masoudir could you brief a tutorial on how to use the micro-ROS Arduino static libraries for Platform.IO so we can include it in this repo README.md or in micro.ros.org? It can be a section in the current README.md or even a new one .md file that we can reference from it. I would like it to be generic and not only for Teensy, is it possible?

Thanks a lot for the efforts!

DaniGarciaLopez, 2021-11-26:
Thanks for the answers. Adding a build_flag to platform.ini file correctly detects microros. I tried with the precompiled galactic version and with a custom compilation and both worked good.
```
[env]
platform = teensy
framework = arduino
upload_port = /dev/ttyACM0
build_flags = -L $PROJECT_DIR/lib/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard -l libmicroros

[env:teensy40]
board = teensy40
```

masoudir, 2021-11-26:
Thanks @pablogs9 and @DaniGarciaLopez . Yes of course. It would be an honor for me to contribute more into your project. There are some small things needed to be considered during using platformio for microros. I will create a readme for that in future days.