micro-ROS/micro_ros_arduino #155: micro_ros_arduino will stop after 30~40 minutes.

Tatsuya-2, 2021-03-04:
Hi! Thank you for the great project! I am working on a project using ROS2 robot. I am trying to connect with the micro ROS agent from my Teensy board. It was successful. However, teensy will stop after 30~40 minutes.

Does it occur in your environment? 

- Hardware description: 

  Tested board: Teensy 4.0 and 4.1

- Installation type:

  micro-ros-Agent: I was followed micro-ros instruction

  micro-ros-Arduino: I was test each way arduino IDE and platformio.(In platformio used rebuilded library)

- system information (please complete the following information):
  OS: Ubuntu 20.04
  ROS 2 Foxy

- Version

  micro_ros_setup: foxy (#275 Commits on Feb 11, 2021)

  micro_ros_arduino: foxy (#99~#154 every version tested)

  Micro-DDS-XRCE: foxy (#219 Commits on Feb 15, 2021)

#### Steps to reproduce the issue

- Upload the script using Arduino IDE or Platformio.(script: micro-ros_recconectioin_example and micro-ros_publisher)

- Connect micro-ros agent

  ```sh
  # Terminal 1
  ros2 run micro_ros_agent micro_ros_agent serial --dev [boardID]] -v6
  ```

- Output messages from a topic

  ```sh
  # Terminal 2
  ros2 topic echo /int32_publisher_rclc
  ```

- publisher will stop

  - micro-ros-agent and topic echo terminal will stop update

  ```sh
  # Terminal 1
  0020: 00 00 00 00
  [1614855668.344434] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x75078075, len: 16, data: 
  0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
  [1614855668.344590] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x75078075, len: 36, data: 
  0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
  0020: 00 00 00 00 #<- this is the last message. 
  
  # Terminal 2
  data: 1
  ~~~~~~
  data: 2106
  ---
  data: 2107
  ---
  data: 2108
  ---       #<- this is the last message
  ```

  

Thanks in advance!

pablogs9, 2021-03-05:
I'm going to do some tests and I'll come back

pablogs9, 2021-03-05:
I have able to reproduce it, I'm trying to find a bug in the timing of the transport, I'll come back with a solution.

Meanwhile, @anaelle-sw have you found something related to this? Are you running your use-cases for more than 40 minutes?

pablogs9, 2021-03-05:
Is possible for you @Tatsuya-2  to test this branch: https://github.com/micro-ROS/micro_ros_arduino/pull/158. Just checkout and run the example, is not necessary to rebuild the micro-ROS precompiled library.

I have had it running for more than an hour without crashing:

[image]

One of my guesses are that `micros()` functions rolls over every ~70 minutes, your publisherr have counted up to 2108 seconds which is 35 minutes (70/2), maybe there was some kind of problem in [this](https://github.com/micro-ROS/micro_ros_arduino/blob/c8c5ae483d8308533bfc100c3bf00ae05ccfd323/src/teensy_transports.c.in#L44) timeout handling.

In the new branch, I have set the Teensy transport to the default Arduino ones which use Arduino API for timing out the reading operation: check [here](https://github.com/micro-ROS/micro_ros_arduino/blob/c8c5ae483d8308533bfc100c3bf00ae05ccfd323/src/default_transport.cpp#L46).

Let me know if this branch solves this issue and in that case, we will merge it.

anaelle-sw, 2021-03-05:
Hello,
On the robot I am using for development, the Teensy application itself has always run well without stopping. 
Some weeks ago, I remember having let ROS, micro-ROS, and the Teensy run for a quite long time (at least a complete day), and I didn't observe this kind of problem.
But in the last weeks, I did not let ROS (and micro-ROS-agent) run for more than 20 to 30 minutes on this robot (as we have problem with the agent regularly crashing). The Teensy application is still running on long period, but it is not connected to an agent, so I have no recent feedback about this.

Tatsuya-2, 2021-03-05:
Thank you very much for your support! @pablogs9 
I will test it and report the result.

Tatsuya-2, 2021-03-05:
> Is possible for you @Tatsuya-2 to test this branch: #158. Just checkout and run the example, is not necessary to rebuild the micro-ROS precompiled library.
>
> I have had it running for more than an [...]

Tatsuya-2, 2021-03-05:
@pablogs9, It works!
I tested [recconection_example](examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) using feature/fix_teensy_transports branch and still working more than 1 hours.

I really appreciate for your help! :)

pablogs9, 2021-03-05:
I still have errors when it is running for more than 70 minutes. I'm going to merge the patch and I will comment here if I found something else.

closed 2021-03-05 by pablogs9

Tatsuya-2, 2021-03-05:
> I still have errors when it is running for more than 70 minutes. I'm going to merge the patch and I will comment here if I found something else.

I had the same problem too. I think a micros() overflow is the problem as you said.

pablogs9, 2021-03-05:
I'm starting to think about [clock_gettime](https://github.com/micro-ROS/micro_ros_arduino/blob/4fc35f9ff290956d44ff407218a74ff19fa88921/src/default_transport.cpp#L15)... It will roll every 70 minutes and probably it is making the middleware to be stuck in a timeout loop.

I'm going to check if it is possible to make a rollover safe function.

pablogs9, 2021-03-05:
Please check and try this fix: https://github.com/micro-ROS/micro_ros_arduino/pull/159

My testing was done here: https://repl.it/@pablogs9/PunctualFragrantMainframe#main.c

Tatsuya-2, 2021-03-05:
> Please check and try this fix: #159
> 
> My testing was done here: https://repl.it/@pablogs9/PunctualFragrantMainframe#main.c

Ok, I will test it from now until 3/9 and report the result at 3/9.

Tatsuya-2, 2021-03-05:
Interim report for you @pablogs9 : [recconection_example](examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) is still working using [#159](https://github.com/micro-ROS/micro_ros_arduino/pull/159) for 2 hours. It looks fixed this issue. :)
I will continue to test until 3/9 morning.

Tatsuya-2, 2021-03-09:
@pablogs9
Teensy 4.0 recconection_example is still publishing the topic.
It looks completely solved. :)
```sh
data: 305348   # <-- Approximately 85 hours
---
```

Thank you for your cooperation!

Tatsuya-2, 2021-03-09:
@pablogs9 
Teensy4.0 recconection_example is still publishing the topic.
It looks completely solved this issue. :)
Thank you for your support!

masoudir, 2021-04-30:
I do not know how you have solved this issue. I was testing my arduino board nucleo64. I could be able to receive data until 71 minutes. Also I have tested my board two time and both 71 minutes.

How did you solve this issue?

pablogs9, 2021-04-30:
Which board is  arduino board nucleo64`? Is supported?

pablogs9, 2021-04-30:
The main change that solved this issue was: https://github.com/micro-ROS/micro_ros_arduino/pull/159/files

masoudir, 2021-04-30:
Thanks. Yes, I am using static libraries of uros for Arduino using platformio.

Also I have changed the micros() function and write a new function instead of that. micros() gives an uint32 output while we are using it as uint64 !!!

I have tested my new micros() function for uros and it has solved the problem. I know #159 can also solve this issue but I think my solution is better.

Thanks in advanced.

My new micros() function is here:

```
uint64_t getMyCurrentMicros(void)
    {
      /* Ensure COUNTFLAG is reset by reading SysTick control and status register */
      LL_SYSTICK_IsActiveCounterFlag();
      uint64_t m = HAL_GetTick();
      const uint64_t tms = SysTick->LOAD + 1;
      __IO uint64_t u = tms - SysTick->VAL;
      if (LL_SYSTICK_IsActiveCounterFlag()) {
        m = HAL_GetTick();
        u = tms - SysTick->VAL;
      }
      return (m * 1000 + (u * 1000) / tms);
    }
```

ottensmeyer, 2021-07-05:
Hello - I'm using an OpenCR 1.0 and believe that I have the same issue.  

I have the OpenCR publishing a topic at 10Hz using a timer, which runs fine for just under 35.79 min, at which point, the host appears to freeze.  

I know that the OpenCR is continuing to run, as the main loop has a toggle for one of the onboard LEDs, which continues after the freeze (i.e. OpenCR not locked up), however a separate toggle that runs in a 1Hz timer callback stops, so that LED remains in whatever state it was at the time of the hang (and no messages from that timer are displayed on the terminal.

I accidentally left the OpenCR and terminal running while in a meeting, and noticed later that my clock display (on a browser-based UI) said 178.96 min (instead of 35.79), then observed that the terminal began to scroll data from the OpenCR sometime after 215 minutes (based on the OpenCR onboard clock).  Watched it over the next few hours, and found that it runs for 35.79 normally, freezes for that long, picks up again and runs normally for another 35.79 and repeats.  This continued for about 6.5 hrs when I finally shut things down.  

Feels like a rollover or signed/unsigned  mismatch or similar issue, given that it works normally for exactly half the time, then hangs (as if an invalid time value is being handled) for the other half of the time, then repeats.

I naively tried to adapt the Teensy modifications in the opencr_transports and related files as appears in https://github.com/micro-ROS/micro_ros_arduino/pull/159/files, suspect I got something wrong. 

Could I ask for an update to the OpenCR code as well?  

Thanks!
Mark

pablogs9, 2021-07-06:
Hello @ottensmeyer I have created this PR: https://github.com/micro-ROS/micro_ros_arduino/pull/357

Can you check out this PR's branch and test if it works?

What I have done is to force OpenCR to use `default_transport.cpp` that uses the Arduino API, so it should handle correctly the time. Let me know if this solves your issue and **if not please open another issue** mentioning this so we can track this problem

ottensmeyer, 2021-07-07:
Hi Pablo,

confirming your suggestion - it looks like I should delete:
  src/arduino_transports.c
and
  src/opencr_transports.c.in
and replace
  src/default_transport.cpp
with the new version that you put together - is that correct?

Will get back to you re: other issue or not later this week - when
something takes 36 min to verify that something's broken and over an hour
to find out if it's made any difference, testing is taking a while.  Looks
like I also have to work through impact of upgrading to 1.0.0/most recent
updates.  I tried 1.0.0 the other day, which caused other things to
"break", so I'm going back again to get to where I was first.

thanks
Mark

pablogs9, 2021-07-08:
Ok, I'm merging the PR #357