micro-ROS/micro_ros_arduino #164: Micro_ros_arduino on esp32

Joelmaca9, 2021-03-08:
Hello guys! I'm currently working on a project using an ESP32 board with arduino as component. I would like to use this library to implement ROS2 in my project but I'm having some troubles, and I don kow if it is posible to do what I want. Has anybody tried before to include this library as component and make it work? 

Thanks in advance!

pablogs9, 2021-03-08:
Hello @Joelmaca9, if you want to add micro-ROS to ESP32 IDF as a component better use this other repo: https://github.com/micro-ROS/micro_ros_espidf_component

pablogs9, 2021-03-09:
In my opinion, if you are going to work with ESP32 you should be using the IDF component, why don't you open an issue there, close this one and we can discuss in the other how to build your application using IDF and micro-ROS as a component?

Joelmaca9, 2021-03-09:
All right! Thank you very much

closed 2021-03-09 by Joelmaca9