micro-ROS/micro_ros_arduino #211: agent foxy not working, rolling yes!

gbr1, 2021-04-09:
I wrote this issue to ask you if it was a known issue and for other people that have same issue.

I have just installed a fresh ROS2 machine (UPxtreme i7 & Ubuntu 20.04) and putted micro-ros-arduino v1.0.0 on Portenta.

Foxy agent had a strange behavior, similar to issue  #111 

`ros2 topic list` not return the micro-ros published topic (I run the publisher example)
Sometimes the right topic appears but `ros2 topic echo /microros...` not show anything.

This issue was not present on my old installation (release v0.0.4 and still working) and a firmware compiled with v1.0.0 works perfectly with old micro-ros-agent:foxy

I pulled the rolling docker and it works perfectly, probably is it going to be in the next foxy update?
Thank you in advance!

pablogs9, 2021-04-12:
Could you provide steps for replicating this?

gbr1, 2021-04-12:
Hi @pablogs9, I hope to have correctly tracked all the stuffs:
1. installed ROS2 on my Ubuntu 20.04 from the tutorial; I didn't setup the hostname neither colcon_cd (on my old laptop it works microros works correctly without hostname)
2. installed microros package in my ROS2 workspace
3. from arduino ide 2, import zip -> micro_ros_arduino tag release 1.0.0
4. installed docker.io from apt and automatically pulled micro-ros-agent:foxy by copy-paste and putting right port `docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev [YOUR BOARD PORT] -v6`
5. loaded publisher example on portenta
6. running micro-ros-agent:foxy 
7. on `ros2 topic list` sometimes `/micro_ros...` topic appeared, a ratio of 1:10 but the agent detect portenta. In these cases `ros2 topic echo /micro_ros...` didn't show anything
8. I change from fozy agent to `micro-ros-agent:rolling` and everything work 
<br>
<br>
<br>
On my old laptop, firmware compiled with v1.0.0 works perfectly without any issue by using a micro-ros-agent not updated since micro-ros-arduino v0.0.4 
<br>
<br>
If you need the micro-ros-agent:foxy is 1.46 GB with image id `b5ebe5e4218f` and micro-ros-agent:rolling is 1.5 GB with image id `6b2f995341fe` 

**NOTE:** On my setup there is always connected a `/dev/ttyACM0` because there is a STM32 connected by USB-CDC, everytime I doule checked the port of Portenta, could be an issue relative to port addressing?

<br>

pablogs9, 2021-04-12:
Does this happens if you use the serial device in `/dev/serial/by-id/[PORTENTA]`? This one is guaranteed to be constant for the same port.

gbr1, 2021-04-12:
@pablogs9 I have just tested, same issue. Foxy not working, rolling yes :'(

amfern, 2021-04-25:
I can reproduce this problem and a somewhat similar issue. running foxy agent and listing topics from ros foxy doesn't work but listing topics from eloquent does work.

1. run agent and wait for connection
```
docker run --init -it -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -b 230400 -v5             ✘ 254 3s 13:23:39
[1619346222.573689] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1619346222.573998] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 5
[1619346228.024523] info     | Root.cpp           | create_client            | create                 | client_key: 0x10ECBE94, session_id: 0x82
[1619346228.024638] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x283950740, address: 0
[1619346228.024778] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x10ECBE94, len: 19
[1619346229.042343] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 16
[1619346229.042422] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 104
[1619346229.042440] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 13
[1619346229.042469] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 13
[1619346229.042546] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 13
[1619346229.042736] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x10ECBE94, len: 32
[1619346229.043690] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x10ECBE94, len: 13
:
```
2. listing topics from foxy doesn't work
```
docker run -ti --rm --net=host ros:foxy ros2 topic list                                                                                 13:20:13
/parameter_events
/rosout
```
3. listing topics from eloquent does work
```
docker run -ti --rm --net=host ros:eloquent ros2 topic list                                                                             13:21:54
/attitude
/flight_status
/heartbeat
/nearest_obstacles
/odometry
/odometry/filtered
/osd_messages
/parameter_events
/rc/override/raw
/reboot
/rosout
```

pablogs9, 2021-04-26:
@amfern your problem seems to be related to ROS 2 middleware and shared memory inside Dockers that are only available on ROS 2 Foxy. [This](https://github.com/eProsima/Fast-DDS/issues/1698#issuecomment-778039676) is related. Try to run both of your docker using `-v /dev/shm:/dev/shm`.

amfern, 2021-04-28:
@pablogs9 I mounted the whole dev in `--privileged` so `-v /dev/shm:/dev/shm` is already mounted

amfern, 2021-04-28:
@pablogs9 yes it works fine when I add  `-v /dev/shm:/dev/shm`

closed 2022-01-28 by pablogs9