micro-ROS/micro_ros_arduino #216: Teensy 4.0 throws RCL_RET_ERROR/RMW_RET_ERROR when initializing support

MatthewWalstra, 2021-04-15:
- Teensy 4.0 
- Micro_Ros_Arduino
- Example Publisher/Subscriber 
- Foxy v1.0.0

#### Steps to reproduce the issue
installed arduino and teensyduino, updated teensy platform.txt, copied example publisher, built, and deployed firmware

#### Expected behavior
Example publisher Node constructs and publishes message received by the micro_ros_agent.

#### Actual behavior
Throws RCL_RET_ERROR/RMW_RET_ERROR in support init (found using print statements)

#### Additional information
Occurs when:
- Deployed from Windows or Linux
- Arduino IDE or PlatformIO
- With or without Micro_Ros_Agent running

#### Questions
- Does the Micro_Ros_Agent need to be running for micro_ros_arduino to initialize?

pablogs9, 2021-04-15:
Yes, you need to have an Agent listening to the micro-ROS Agent in order to initialize the micro-ROS layers.

Does this works if you launch your agent with `docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev [YOUR BOARD PORT] -v6` ?

MatthewWalstra, 2021-04-15:
Running the docker container at the same time solved the issue; it must have been a leftover container last time. Can you run a serial monitor at the same time as the micro-ROS Agent?

closed 2021-04-15 by MatthewWalstra