micro-ROS/micro_ros_arduino #219: Build from source error

anaelle-sw, 2021-04-15:
Hi! I would like to install micro-ROS from sources but a build error happens.

- System: Ubuntu 20.04 / ROS2 Rolling
- micro-ROS installed from source from [latest commit on main branch of micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup/tree/d78469ac492c16fb08892a08dc004cfd1d83c7db)
- micro-ROS Arduino from [latest commit on main branch](https://github.com/micro-ROS/micro_ros_arduino/tree/063a1822f06ab18293ac5fb9453e92c17116e9a7)

## Steps to reproduce the issue
I cloned micro-ROS Arduino and copied a custom meta file:
```c
cd ~/arduino-1.8.13/hardware/teensy/avr/
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/main/extras/patching_boards/platform_teensy.txt > platform.txt
cd ~/Arduino/libraries
git clone -b main https://github.com/micro-ROS/micro_ros_arduino.git
cd ~
cat ~/colcon_lowmem.meta > ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta
```
Then I cloned the micro_ros_setup repo and followed the tutorials:
```c
mkdir microros_ws && cd microros_ws/
git clone -b main https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
sudo apt update && rosdep update
rosdep install --from-path src --ignore-src -y
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh generate_lib
ros2 run micro_ros_setup build_firmware.sh ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/teensy32_toolchain.cmake ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta
```
##  Issue description
This last command failed:
```c
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /home/anaelle/Arduino/libraries/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta
Starting >>> rcutils 
Starting >>> rosidl_cli
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
Starting >>> rmw_implementation_cmake
Starting >>> tracetools
Starting >>> tracetools_trace
Starting >>> test_interface_files
Starting >>> tracetools_read
Starting >>> libyaml_vendor
--- stderr: tracetools                                                               
gcc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
g++: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
g++: error: unrecognized command line option ‘-mthumb’
gcc: error: unrecognized command line option ‘-mthumb’
make[2]: *** [CMakeFiles/tracetools.dir/build.make:79: CMakeFiles/tracetools.dir/src/utils.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/tracetools.dir/build.make:66: CMakeFiles/tracetools.dir/src/tracetools.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:81: CMakeFiles/tracetools.dir/all] Error 2
make: *** [Makefile:144: all] Error 2
---
Failed   <<< tracetools [0.93s, exited with code 2]
Aborted  <<< rosidl_typesupport_interface [0.96s]
Aborted  <<< rmw_implementation_cmake [0.94s]
Aborted  <<< test_interface_files [0.92s]
Aborted  <<< microcdr [1.04s]                                                                   
Aborted  <<< rosidl_cli [1.10s]                                                                 
Aborted  <<< tracetools_read [1.06s]
Aborted  <<< tracetools_trace [1.08s]
Aborted  <<< rcutils [1.13s]
Aborted  <<< libyaml_vendor [1.09s]

Summary: 0 packages finished [1.53s]
  1 package failed: tracetools
  9 packages aborted: libyaml_vendor microcdr rcutils rmw_implementation_cmake rosidl_cli rosidl_typesupport_interface test_interface_files tracetools_read tracetools_trace
  3 packages had stderr output: microcdr rcutils tracetools
  51 packages not processed
```
##  Additional information
The [Cmakefile](https://github.com/micro-ROS/micro_ros_arduino/blob/063a1822f06ab18293ac5fb9453e92c17116e9a7/extras/library_generation/teensy32_toolchain.cmake) I provided is directly taken from this repo and is unchanged.
The error doesn't seem to be related to my custom meta file, but here it is if needed:
```c
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS=3"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=3",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=2",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

I can precise that I have no problem to build the micro-ROS Arduino library from docker, and that I can also launch the micro-ROS agent from a docker. So maybe there is just a something missing here, I might have forget a step. Sorry if it is the case.

Just for information, the reason why I try to use micro-ROS from sources is that I would like to be able to launch it as a node, with a launch file or ros2 cli.

Thanks a lot for helping!

pablogs9, 2021-04-15:
I guess that you are not using the correct compiler. With the docker approach:

1. Inside the docker you have several versions of `arm-gcc` because micro-ROS should be built with the same compiler that Arduino uses. You can check how they are installed [here](https://github.com/micro-ROS/docker/blob/871406de64b73670ad722eae5d7b1d2f879da207/micro-ROS-static-library-builder/Dockerfile#L15).
2. Then in the entrypoint script, depending on the platform, `TOOLCHAIN_PREFIX` [environment variable is set](https://github.com/micro-ROS/micro_ros_arduino/blob/1d7ccf6f43e0295fe88794bebd84bf22b38780a7/extras/library_generation/library_generation.sh#L102)
3. The CMake toolchain uses this `TOOLCHAIN_PREFIX` variable to configure which cross-compiler use. [Here](https://github.com/micro-ROS/micro_ros_arduino/blob/1d7ccf6f43e0295fe88794bebd84bf22b38780a7/extras/library_generation/teensy32_toolchain.cmake#L5). Probably if you are not setting `TOOLCHAIN_PREFIX` the `set(CMAKE_C_COMPILER $ENV{TOOLCHAIN_PREFIX}gcc)` is resolving to `set(CMAKE_C_COMPILER gcc)` so you are trying to crosscompile with the Linux default compiler. And probably this is why `-mthumb` is not being recognized, because it is an [ARM flag](https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html)

anaelle-sw, 2021-04-15:
Ok thanks @pablogs9! Sorry about it, I didn't know that it was required to install compilers.

So I did this:
```c
cd ~/microros_ws
# Installing the compiler as you mentioned
wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 gcc-arm-none-eabi-5_4-2016q2/share/doc \
    && wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 gcc-arm-none-eabi-5_4-2016q3/share/doc \
    && wget https://launchpad.net/gcc-arm-embedded/4.8/4.8-2014-q1-update/+download/gcc-arm-none-eabi-4_8-2014q1-20140314-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-4_8-2014q1-20140314-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-4_8-2014q1-20140314-linux.tar.bz2 gcc-arm-none-eabi-4_8-2014q1/share/doc \
    && wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/7-2017q4/gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2 gcc-arm-none-eabi-7-2017-q4-major/share/doc \
    && wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2 gcc-arm-none-eabi-9-2020-q2-update/share/doc

# Setting TOOLCHAIN_PREFIX to the compiler location
export TOOLCHAIN_PREFIX=~/microros_ws/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-

# Pursuing the tutorial steps
ros2 run micro_ros_setup build_firmware.sh ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/teensy32_toolchain.cmake ~/Arduino/libraries/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta
source install/local_setup.bash
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
```

But the last command failed:
```cStarting >>> microcdr
Starting >>> microxrcedds_agent
Finished <<< microcdr [1.31s]                                                                       
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [1.32s]                                                                       
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [5.83s]                                                                       
--- stderr: microxrcedds_agent                                
Cloning into 'spdlog'...
Cloning into 'fastcdr'...
Cloning into 'sanitizers'...
Already on 'master'
Note: switching to 'v1.0.13'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 174f6ff Bump version to 1.0.13 (#70)
Note: switching to 'v1.4.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
[... 73 lines cut ...]
      |           ~~~~~~~~^~~~~~~~~~
      |           m_liveliness
/opt/ros/rolling/include/rmw_fastrtps_shared_cpp/qos.hpp:107:64: error: ‘const class eprosima::fastdds::dds::ReaderQos’ has no member named ‘liveliness’; did you mean ‘m_liveliness’?
  107 |   qos->liveliness_lease_duration = dds_duration_to_rmw(dds_qos.liveliness().lease_duration);
      |                                                        ~~~~~~~~^~~~~~~~~~
      |                                                        m_liveliness
/opt/ros/rolling/include/rmw_fastrtps_shared_cpp/qos.hpp:109:19: error: ‘const class eprosima::fastdds::dds::ReaderQos’ has no member named ‘history’
  109 |   switch (dds_qos.history().kind) {
      |           ~~~~~~~~^~~~~~~
/opt/ros/rolling/include/rmw_fastrtps_shared_cpp/qos.hpp:120:44: error: ‘const class eprosima::fastdds::dds::ReaderQos’ has no member named ‘history’
  120 |   qos->depth = static_cast<size_t>(dds_qos.history().depth);
      |                                    ~~~~~~~~^~~~~~~
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:89: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [4.43s, exited with code 2]

Summary: 4 packages finished [35.3s]
  1 package failed: micro_ros_agent
  2 packages had stderr output: micro_ros_agent microxrcedds_agent
```

Any idea why? Thanks!

pablogs9, 2021-04-15:
Yes, this a know issue since two days ago, in fact, our [CI is failing](https://github.com/micro-ROS/micro_ros_setup/runs/2349472029?check_suite_focus=true) for the micro-ROS Agent in rolling because they have changed some APIs and we need to update some implementations.

Tomorrow or next week we are going to fix this. Meanwhile can you use the dockerized version?

CC: @jamoralp @Acuadros95

pablogs9, 2021-04-15:
BTW, you don't need to download all the compilers, for your platform just this one:

``` 
wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2 gcc-arm-none-eabi-5_4-2016q3/share/doc \
```

anaelle-sw, 2021-04-15:
Ok, thank you very much for the quick answers!

> Tomorrow or next week we are going to fix this. Meanwhile can you use the dockerized version?

Yes sure, we are using micro-ROS agent inside a docker for the moment. But in the end, it would be better for us to be able to launch it from a launch file (that's why I was trying to install it from sources)

pablogs9, 2021-04-15:
Cool, I will alert you here when we fix it.

anaelle-sw, 2021-04-23:
Hello @pablogs9 ! Any news about this?

pablogs9, 2021-04-23:
Working on it: https://github.com/micro-ROS/micro-ROS-Agent/pull/61/checks?check_run_id=2418442971 and https://github.com/micro-ROS/micro_ros_setup/pull/309

Once these are merged you should be able to build the Rolling agent.

pablogs9, 2021-04-26:
Hello @anaelle-sw, this is solved. I'm closing, please reopen if you find some issues with the Rolling agent build.

closed 2021-04-26 by pablogs9

echogun, 2024-11-07:
Hello, I am a beginner with ROS 2 and micro-ROS. Currently, I have a Raspberry Pi 4 with Ubuntu Mate installed. I needed to install Arduino 1.8.19, which required me to install Teensyduino separately. After setting everything up, I was able to upload code to the Teensy 4.1 without any issues. However, after I started installing micro-ROS Humble, I followed these steps:

cd ~/arduino-1.8.13/hardware/teensy/avr/
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/main/extras/patching_boards/platform_teensy.txt > platform.txt
I also added the library as a zip file to this location. Then, I tried uploading an example sketch, but I got the error:

fork/exec /arm/bin/arm-none-eabi-g++: no such file or directory

It resulted in an error compiling for board Teensy 4.1, and I haven't been able to upload anything since. I also tried uploading a simple example sketch, but I still get the same error. I'm not sure if I installed everything correctly.