micro-ROS/micro_ros_arduino #220: Flashing the firmware on ardiuno due issue

Kakcalu13, 2021-04-15:
I'm hoping this [documents](https://micro.ros.org//docs/tutorials/core/first_application_rtos/zephyr/) is for this one. 
I was able to following until the "Building the firmware" section. I'm using ardiuno DUE and I don't see any JTAG on ardiuno DUE. Do you know if there's any tutorial or something for ardiuno DUE? 

When I decide to test it by run ros2 run micro_ros_setup build_firmware.sh. Everything seems normal. After that, I run ros2 run micro_ros_setup flash_firmware.sh and it just display an error;
```
bwuk@robots:~/microros_ws$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for zephyr platform olimex-stm32-e407
Error: Unsuported OpenOCD USB programmer
bwuk@robots:~/microros_ws$ 
```
Do you know how to do it on ardiuno DUE?

pablogs9, 2021-04-15:
Those instructions are not intended for this micro-ROS support. Please follow [this](https://github.com/micro-ROS/micro_ros_arduino#how-to-use-the-precompiled-library) instructions.

Kakcalu13, 2021-04-15:
Thank you for the quick reply! 

Sorry, I'm a little new to this one. So, do I just export ardiuno to ardiuno path or library path? 

This is what I did so far:
`bwuk@robots:~$ export ARDUINO_PATH=~/Arduino/

bwuk@robots:~$ cd $ARDUINO_PATH/hardware/sam/1.6.12/

bash: cd: /home/bwuk/Arduino//hardware/sam/1.6.12/: No such file or directory

bwuk@robots:~$ cd Arduino/
`

Also I'm not using docker. Am I supposed to use docker to make it work?

pablogs9, 2021-04-15:
Not related, you need to read the section called `How to use the precompiled library`. And [here](https://www.arduino.cc/en/guide/libraries) you can learn how to add a .zip library. Please read the section `Importing a .zip Library`.

Regarding the docker, the micro-ROS agent can be installed using different approaches but I guess that docker is the easiest way.

Kakcalu13, 2021-04-15:
Yeah, I did that yesterday.
It isn't working for me. Here is the error code:

[image]

I will try with the docker

pablogs9, 2021-04-15:
Ok, thats beacause you have to patch you installation: https://github.com/micro-ROS/micro_ros_arduino#patch-samd

Kakcalu13, 2021-04-16:
Sorry if we are still on the same story but..

I followed the instructions:

> cd $ARDUINO_PATH/hardware/sam/1.6.12/

I did this and I get no sam
[image]

So, I decided to use the locate to find the folder. This is what I found (Cut the screenshot to small since its just full of path locations.)
[image]

So, i went back and export this path: /home/bwuk/.arduino15/packages/arduino/hardware/samd/1.8.11/

then ran this line inside the /1.6.11

> curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/foxy/extras/patching_boards/platform_arduinocore_sam.txt > platform.txt
 
 So, I tested it by open ardiuno using the example 
 micro-ros publisher.ino
  and click verify before I could upload it.
  
  The issue seems persists, I afriad. 
[image]

 
 I understand this is still in BETA. I'd like to contribute and get more involves with this while learn as much as I can from this contribution. 
 
 So, I am wondering if Ardiuno updated to change from /ardiuno to /.ardiuno15 or at least, was it something I missed or skipped the step?

Kakcalu13, 2021-04-16:
Crap, I'm literally embarrassed! I was in wrong path :P I was in SAMD, not SAM. 

It should be in the path:
> ~/.arduino15/packages/arduino/hardware/sam/1.6.11

[image]

It's working now! THank you so much! 

I'm going to close this issue since it's already resolved. 

Another thing, I thought you might want to know that it's now .ardiuno (at least to 20.04 Ubutnu Focal)

closed 2021-04-16 by Kakcalu13