micro-ROS/micro_ros_arduino #223: Node doesn't display

Kakcalu13, 2021-04-19:
Topic doesn't visible after added this line;
`rcl_node_options_t node_ops = rcl_node_get_default_options();`
 `node_ops.domain_id = 30;`
` RCCHECK(rclc_node_init_with_options(&node, "micro_ros_arduino_node", "", &support, &node_ops));`
in the ardiuno [code](https://pastebin.com/1SNivEdc)

I ran this line
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Arduino_LLC_Arduino_Due-if00 -v6`
`ROS_DOMAIN_ID=30`
then re-upload to ardiuno and I get no luck. Restart the laptop doesn't help either :P 
[image]

The top sourced in microros_ws along with the bottom left side. The bottom right side sourced in /opt/ros/foxy just to make sure if it can view the node. 

Followed [this](https://github.com/micro-ROS/micro_ros_arduino/issues/21) and I did the same thing, not sure where to take a look now even clear the ros domain in .bashrc. 

Is it a new bug or is it something that I missed? 

- Hardware description: Ardiuno DUE
- RTOS: from the tutorial [list](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/README.md)
- Version or commit hash: Foxy

#### Expected behavior
Display "micro_ros_arduino_node"" node

#### Actual behavior
Doesn't display the node

pablogs9, 2021-04-19:
If you don't change the domain id everything works correctly?

Kakcalu13, 2021-04-19:
Nope, even if I didn't change the domain id. 

I recently commented this line out and comment the ros_domain in .bashrc out as well too then open a new terminal/reupload to ardiuno due. No luck.

pablogs9, 2021-04-19:
Try to not use `Serial` in lines like:

```c
  Serial.begin(9600); // Starts the serial communication
```

This serial port is being used by the micro-ROS client to communicate with the agent. Let me know if deleting all `Serial` from your code the agent receives data.

Kakcalu13, 2021-04-19:
Unfortunately, 

I deleted that line and this too as well;
`Serial.print("Distance: ");`
`Serial.println(distance);`

No node displaying either. Agent nor ros2 were able to receive data. 

Question, was it supposed to be 
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Arduino_LLC_Arduino_Due-if00 -v6`
or should I run at
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6` (As long as the port match, that is)

I tried both and no luck so far

pablogs9, 2021-04-19:
Both agent command should work. Have you tried to use one of the basic examples provided with this library? Does the basic publisher work?

Kakcalu13, 2021-04-19:
Actually since you mentioned that now, it didn't work ever. 

It worked with [this](https://micro.ros.org/docs/tutorials/core/first_application_linux/) but I know this tutorial isn't intend for this library. 

I will start over on micro ros and see if the result is same. I will let you know by tomorrow.

Truly appreciate your time and efforts!

Kakcalu13, 2021-04-20:
Update:

So, I went to [release ](https://github.com/micro-ROS/micro_ros_arduino/releases)list to download the latest. Apparently, I was on .02v instead of v1. 

Before I download the zip, I went ahead and deleted the folders for micro ros ardiuno in ardiuno by going to library and delete them. After that, I added the ZIP to ardiuno and tested the example. Unfortunately, it's not working as I hoped.

So, I started to wonder if it was the agent issue so I tested the agent. It worked
[image]

So, as you can see that the right bottom corner isn't picking anything up from the agent due to source /opt/ros/foxy/setup.bash

That leads me to wonder if I need to source in micro ros ardiuno?

Kakcalu13, 2021-04-20:
I ran colcon build in micro_ros_ardiuno folder then source install/setup.bash inside the folder. 
Still no topic visible
[image]

I'm kinda out of ideas to troubleshoot now. I set ros domain to 10 due to the code:
[image]
[image]
[image]

Still no luck. 

Any new idea or suggestion? @pablogs9

pablogs9, 2021-04-20:
Paste here the ouput of the agent but using the `-v6` flag

Kakcalu13, 2021-04-20:
[image]
Like this?

I'm not familiar nor skilled with agent yet

pablogs9, 2021-04-20:
yes, but with the micro-ROS client being connected

Kakcalu13, 2021-04-20:
hm, I don't think I have that one in any of my steps. I thought it was just agent. Can you tell me the command to run that? 

Apparently, I only have micro_ros_agent as in my package.

pablogs9, 2021-04-20:
the agent but with your board connected an communicating as in https://github.com/micro-ROS/micro_ros_arduino/issues/223#issuecomment-823108560

Kakcalu13, 2021-04-20:
[image]
Like this? (the left top corner isn't being used like other 3. 

How do you connect to the board using udp4 or rather serial? I tried with the serial and it wouldn't run

Such as this
`bwuk@robots:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent udp4 --dev dev/serial/by-id/usb Arduino_LLC_Arduino_Due-if00 -v6`
`Warning: '--port <value>' is required`
`Usage: 'micro_ros_agent <udp4|udp6|tcp4|tpc6|serial|pseudoterminal> <<args>>'`
`For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.`

Kakcalu13, 2021-04-20:
Bingo! Found! Here is the output
[image]

The error part is from CPU reset on ardiuno due. We can see it doesn't connect to ros2

KohlhardtC, 2021-05-01:
I'm new to micro-ros and was having the same issue as OP where `ros2 node list` and `ros2 topic list` showed nothing, and then discovered that rqt_graph _**does**_ show a  **/micro_ros_arduino_node** even when `ros2 node list` shows nothing. I'm not sure why this is the case, but I thought I'd mention this in case it's helpful. 

Agent OS: Ubuntu 20.04
ROS2 distribution: Foxy
Hardware: Teensy 4.1
Teensy client code: [micro-ros_publisher.ino](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_publisher/micro-ros_publisher.ino)

Steps to reproduce:
- Load client code onto Teensy
- unplug USB cable between Teensy and Linux box running the Agent
- plug USB cable back in
- Startup agent using the following command within 2 seconds of plugging in USB cable: `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200  -v 6`
- In another terminal
-- source install/setup.bash
-- rqt_graph
-- select Nodes only and click **_Refresh ROS graph_** button

Also, while writing this report I discovered that if I killed all background ros2 processes, and then ran the agent and rqt_graph, I was able to get `ros2 node list` and `ros2 topic list` to show expected output. While perhaps there is a workaround here, it does not seem like expected behavior....

[image]

[image]

[image]

Kakcalu13, 2021-05-10:
Thank you so much for your time, @KohlhardtC! 

That comment helped me to connect the agent! However it doesn't work like it did to you :( I'm wondering if it's the board issue? I confirmed my agent able to get/send the data but there is no topic nor node for it. 

Agent OS: Ubuntu 20.04
ROS2 distribution: Foxy
Hardware: Ardiuno DUE (Programming port)
client code: micro-ros_publisher.ino

I recorded what I did and it doesn't work as you can see [here](https://youtu.be/dIY7Gp0OX2U)

Another question for you, @pablogs9 @KohlhardtC 

Have you tried this on any ardiuno board using this micro ros ardiuno? If no, I probably need to buy teensy

pablogs9, 2021-05-10:
The officially supported boards (tested by the micro-ROS team) are:

[image]

Kakcalu13, 2021-05-12:
The steps you provide isn't working for me, unfortunately. 

I had to re-install everything on this part. Here is what I've done so far
Between Teensy and Arduino:

1. Plug the micro USB to Teensy 4.1 (the board should have an orange blinking)
2. Press the button on the board.(The blinking should stop)
3. Open terminal and follow this list:
```
wget https://downloads.arduino.cc/arduino-1.8.13-linux64.tar.xz
wget https://www.pjrc.com/teensy/td_153/TeensyduinoInstall.linux64
wget https://www.pjrc.com/teensy/00-teensy.rules
sudo cp 00-teensy.rules /etc/udev/rules.d/
tar -xf arduino-1.8.13-linux64.tar.xz
chmod 755 TeensyduinoInstall.linux64
./TeensyduinoInstall.linux64 --dir=arduino-1.8.13
cd arduino-1.8.13/hardware/teensy/avr/cores/teensy4
make
cd ~/arduino-1.8.13/
sudo ./install
```
Ardiuno and Micro-ROS:

1. Download latest zip from (Micro-ROS)[https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v1.0.0]
2. Open Arduino IDE
3. Click Sketch > Include library > Add Zip > [the zip you downloaded] 
4. cd [Ardiuno PATH]/hardware/teensy/avr/
5. curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/foxy/extras/patching_boards/platform_teensy.txt > platform.txt
6. Open a file called "Micro-ros_publisher.ino" from example in Micro-ros library 
7. Click Download in Arduino IDE 

WIth the agent:
1. Download the code into Teensy
2. Unplug the board then plug and run it in terminal after source install/setup.bash in micro-ros directory
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 9600 -v 6
3. Open another terminal and source install/setup.bash then run ros2 topic list
4. Open another terminal and source install/setup.bash then rqt_graph
5. Click refresh node on it. 

No luck :(

> Also, while writing this report I discovered that if I killed all background ros2 processes, and then ran the agent and rqt_graph, I was able to get `ros2 node list` and `ros2 topic list` to show expected output. While perhaps there is a workaround here, it does not seem like expected behavior....

@KohlhardtC  Question, how do you kill ros processes? 

For me, I used `pkill ros `since aux was able to detects micro_ros running in background however it doesn't display the topic to ros2 visibly. 

Here is the picture
[image]

pablogs9, 2021-05-12:
1. How are you running the micro-ROS Agent?
2. Where have you set the baud rate to 9600? By default it is 115200

Kakcalu13, 2021-05-12:
>     1. How are you running the micro-ROS Agent?
> 
>     2. Where have you set the baud rate to 9600? By default it is 115200

#1 answer:
1) I ran the agent by go in the micro-ros-ardiuno
2) source install/setup.bash
3) once the I plugged the Teensy, I run this
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200 -v 6`

That's it.

Answer #2: 
I actually changed it to 9600 since 115200 didn't work for me. The result is same. I was hoping 9600 will, magically, works :P

KohlhardtC, 2021-05-12:
I just used the kill command. I don't know if the kill was the action that
helped make it possible to see topics/nodes or if something else is going
on. I'm guessing something else is going on since I haven't had any issues
with missing topics/nodes recently.

KohlhardtC, 2021-05-12:
I was thinking about this some more, and I starting to wonder if maybe the problem is ros2 specific and not micro-ros specific? 

For example, I [googled "ros2 node not discovered node list"]( https://www.google.com/search?q=ros2+node+not+discovered+node+list&sxsrf=ALeKk03bEOVioZ7FRvTV9HrVwS7c2xQiww%3A1620847301302&ei=xSqcYNTiEZHr-gTs44joAw&oq=ros2+node+not+discovered+node+list&gs_lcp=Cgdnd3Mtd2l6EAM6BwgjELADECc6BwgAEEcQsANQnydYnydg7ihoAXACeACAAewBiAHUApIBBTAuMS4xmAEAoAEBqgEHZ3dzLXdpesgBB8ABAQ&sclient=gws-wiz&ved=0ahUKEwiU2YPd7sTwAhWRtZ4KHewxAj0Q4dUDCA4&uact=5) and found something like this can happen if there are [network configuration issues](https://answers.ros.org/question/372464/ros2-foxy-cant-discover-nodes-from-other-machines/) or [the ROS_DOMAIN_ID environment variable is set](https://discourse.ros.org/t/how-to-run-example-talker-listener-on-2-machines-with-different-ipaddr/2106).

Kakcalu13, 2021-05-13:
> I was thinking about this some more, and I starting to wonder if maybe the problem is ros2 specific and not micro-ros specific?
> 
> [the ROS_DOMAIN_ID environment variable is set](https://discourse.ros.org/t/how-to-run-example-talker-listener-on-2-machines-with-different-ipaddr/2106).

This made me REALIZED...Since I re-install arduino due to multiple paths. I forgot to add this part since my old file was modified with this part. The new didn't have this! 
```
rcl_node_options_t node_ops = rcl_node_get_default_options();
node_ops.domain_id = 10;
RCCHECK(rclc_node_init_with_options(&node, "my_node_name", "", &support, &node_ops));
```
from #21 

then I inserted the code above and bam! It's working!!!!! 
[image]

Thank you so much for your time! You are a beautiful soul!  @KohlhardtC @pablogs9 

Since this issue is now resolved.

To anyone who is trying to get this working, feel free to @ me in a new issue!

closed 2021-05-13 by Kakcalu13