micro-ROS/micro_ros_arduino #226: micro_ros_arduino with Teensy 4.1

Chubba07, 2021-04-21:
Hello everyone,
I am currently working on a project at a company to investigate the possibility of having micro_ros on a teensy. The main goal was to check if it would be possible to have nodes initiated on the micro controller independently (without a host system with ROS 2) during boot-up.

Going through the examples and other documents related to micro-ros I learned that almost all applications required the presence of micro-ros Agent for initiating the nodes on the micro-controller.

I would like to know if this could be achieved using micro_ros. 

Thanks in advance

pablogs9, 2021-04-22:
Do the nodes in your MCU need to communicate with another MCU or any external device?

As far as I understand from your comment, you need a group of micro-ROS nodes running in the same MCU with no access to the external data space. We are planning a "shared-memory" feature that is going to be released soon. Does this fill your requirements?

This issue is related: https://github.com/micro-ROS/micro-ROS-Agent/issues/37#issuecomment-823969052

himanshuak, 2021-04-22:
Thankyou for the instant response.

This is the scenario  
* The MCU is interfaced with sensors
* We want to have few nodes running the sensors on the MCU and communicate with each other.
* Whenever the MCU is connected to a Linux Machine over Serial communication, then the nodes on linux machine should be able to subscribe to the topics published on the MCU without the presence of an agent.

pablogs9, 2021-04-22:
Ok, I think that micro-ROS can be used in your scenario **except** in the third point. If you want to subscribe from ROS 2 to a micro-ROS topic you need the micro-ROS agent to translate between middleware data spaces.

Why having an intermediate agent is a problem in your use case?

himanshuak, 2021-04-22:
I went through most of the examples related to micro_ros_arduino. Like for example in [micro_ros_reconnection](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) it looks like the micro_ros entities are created and destroyed as per the availability of the micro_ros_agent using the helper functions `create_entities()` and `destroy_entities()`. 
* So this makes me wonder if the agent is only required to translate between middleware data spaces or it is also required to initiate nodes?
* If there is a simple ros timer set up on the MCU  to execute certain functionality (for eg. turn on/off the led) during a certain interval. Does this work without the agent?Just boot up MCU and the led should blink at the frequency of the timer.  Because I tried to implement this and couldn't succeed. 
* And when you said the nodes on the MCU could communicate with each other. Did you mean that they could Publish/Subscribe to a certain topic without the agent?

pablogs9, 2021-04-23:
Well, micro-ROS is based on a middleware called XRCE-DDS. This middleware implements a wire protocol that communicates a client with an agent with the only purpose of interfacing with a DDS network (DDS is the default middleware protocol for ROS 2).
Usually, the clients are MCU where makes no sense to implement a full DDS library, so XRCE-DDS provides a lightweight solution.

So, basically when a micro-ROS node inits, it needs to init the middleware and the middleware needs to connect with an agent. So the answer right now is yes, you need an agent to have a micro-ROS timer. But this makes no sense.

So we are planning two things:
1. To make the micro-ROS middleware implementation kind of independent from the XRCE-DDS in the sense that you can init the micro-ROS layers and create entities with no agent. But of course, in this situation, there will not be micro-ROS to ROS 2 communication.
2. To provide publishers and subscribers that are on the same MCU the ability to communicate between themselves without the need of an agent.

So,

> So this makes me wonder if the agent is only required to translate between middleware data spaces or it is also required to initiate nodes?

The function translates between dataspaces. By implementation limitations right now is required for instantiating nodes (this is going to change).

> If there is a simple ros timer set up on the MCU to execute certain functionality (for eg. turn on/off the led) during a certain interval. Does this work without the agent?Just boot up MCU and the [...]

Now you need the agent for micro-ROS to startup, and for using a timer with an executor.

> And when you said the nodes on the MCU could communicate with each other. Did you mean that they could Publish/Subscribe to a certain topic without the agent?

When the publish and the subscribe are in the same MCU yes, with the new shared memory feature, this communication will happen without agent.

Acuadros95, 2022-07-12:
Closing due to inactivity, reopen if needed.

closed 2022-07-12 by Acuadros95