micro-ROS/micro_ros_arduino #229: Question: Multi-threading with micro_ros_arduino

shecker, 2021-04-22:
## Question on multi-threading with micro_ros_arduino

- Hardware description: Teensy4.1 
- Version or commit hash: foxy

#### Additional information
Thanks a lot for this great library! It's very helpful for our use case.
We are polling from two sensors that are reporting at a high frequency (~200Hz). We want to publish each sensors data on separate publishers with ROS2. Unfortunately we only found examples of single threaded executioners. We have tried using Teensy Threads to solve this problem but are unsure how to proceed properly. Do you have any suggestion on how to implement a multi threaded micro_ros_arduino application? 
With ROS1 we were using rosserial_arduino in combination with FreeRTOS and could implement it. 

Any help would be greatly appreciated!
Thanks a lot

pablogs9, 2021-04-23:
Hello @shecker, micro-ROS middleware is not ready for multithreading (is not thread-safe) but we are working on this patch (https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/216) for enabling it. I guess that this will be merged in the next two weeks.

In the case for example that you are reading each sensor in a thread you can manually lock the publishing operation and this will work, only make sure that the entity creation is not concurrent, something like:

```c
main:
    create micro-ros entities
    launch thread 1
    launch thread 2
    sleep forever

thread 1:
    loop:
        read sensor 1
        lock mutex
        micro-ros publish
        unlock mutex

thread 2:
    loop:
        read sensor 2
        lock mutex
        micro-ros publish
        unlock mutex
```

Basically, the micro-ROS multithreading implementation that we are going to release is based on internal mutexes that handle the concurrent middleware operation.

shecker, 2021-04-23:
Hi @pablogs9 thanks a lot for your reply and explanation.
Originally we were using a dual core esp32 for ROS1 but during the migration to ROS2 we switched to Teensy4.1. Erroneously we thought the Teensy is also dual core, but that turned out to be false, so multi-threading doesn't actually help us much. 
We are thinking of going back to the ESP32, do you have any experience with porting the micro-ros arduino library to a new board such as an ESP32?
Thanks!

pablogs9, 2021-04-23:
We have a [complete port of micro-ROS for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component) and we support ESP-IDF v4.1, v4.2, and v4.3 with ESP32, ESP32-S2, and ESP32-C3.

pablogs9, 2021-04-26:
Closing, reopen if you have more questions.

closed 2021-04-26 by pablogs9