micro-ROS/micro_ros_arduino #232: rclc_executor_spin_some execution takes too much time after running for more than 15 hours

anaelle-sw, 2021-04-23:
Hello Micro-ros team! We are currently trying to test the stability of our micro-ROS application when running for quite long periods of time. Some frequency problems appear between 15 and 17 hours of running. Some clues really make me think it has something to do with the `rclc_executor_spin_some` function.

## Setup
- Teensy 3.2
- micro-ROS Arduino from main branch (Rolling), built and launched via micro-ROS dockers
- Ubuntu 20.04 / ROS2 Rolling / FastDDS middleware implementation

## Steps to reproduce the issue
I uploaded this code to my Teensy board:
```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/bool.h>
#include <std_msgs/msg/int64.h>
#include <std_msgs/msg/empty.h>
#include <std_msgs/msg/char.h>
#include <std_msgs/msg/float64.h>

#define DOMAIN_ID 24
#define LED_PIN 13
#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

// Standard bool publisher
rcl_publisher_t bool_pub;
std_msgs__msg__Bool bool_msg;

// Standard float publisher
rcl_publisher_t float_pub;
std_msgs__msg__Float64 float_msg;

// Standard empty publisher
rcl_publisher_t empty_pub;
std_msgs__msg__Empty empty_msg;

// Standard int subscription
rcl_subscription_t int_sub;
std_msgs__msg__Int64 int_msg;

// Standard char subscription
rcl_subscription_t char_sub;
std_msgs__msg__Char char_msg;

// Micro-Ros
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_init_options_t init_options;
rmw_init_options_t* rmw_options;
unsigned int iter = 0;
unsigned long loop_period_millis = 20;
unsigned long start_loop_time;

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, LOW);
    delay(100);
  }
}

// Subscriptions callbacks
void int_cb(const void * msg_in){  
  const std_msgs__msg__Int64 * msg = (const std_msgs__msg__Int64 *)msg_in;
[... 136 lines cut ...]

Thanks to my the debugging topic `topic_float`, I was able to compute the max and mean time between each loop's step, when running at normal frequency, and when the frequency is divided by 10. Here is the complete spreadsheet I used for computing these values (there is a tab for normal frequency and another for slowed-down frequency): 
[float_timestamp.xls](https://github.com/micro-ROS/micro_ros_arduino/files/6366230/float_timestamp.xls). But briefly, these are the interesting results (values are given as milliseconds):
| Step | Normal frequency - Max duration | Slow frequency - Max duration  | Normal frequency - Mean duration  | Slow frequency - Mean duration  |
| ----- | ----- | ----- | ----- | ----- |
| 1 --> 2 | 4.3 | 205.0  | 0.2 | 162.8 |
| 2 --> 3 | 25.1 | 0.1 | 5.3 | 0.04 |
| 3 --> 4 | 20.0 | 0.1 | 18.9 | 0.04 |
| 4 --> 1 | 0.02 | 0.2 | 0.02 | 0.06 |

First, since that the step 3 (3 --> 4) corresponds to the delay imposed in order to get a 50Hz loop, please note that is normal that this step duration drops, as the loop takes 10 times more time to execute than expected. 
Secondly, the step 1 (1 --> 2) takes almost 1000 times more time that during a loop executing at expected frequency. So, as step 1 only consists in publishing the other topics and in spinning the executor, I really think it has something to do with the `rclc_executor_spin_some` function. It seems that this function execution drastically slows down after approximately 15 - 17 hours.

The funny part is that after approximately 27 - 30 hours of running, the frequency just speeds up and gets back to the expected value (I was able to reproduce this only twice for the moment, and I haven't had the time to let it run more than 30 hours to check if the problem keeps happening chronically)
 
## Additional information
I have just launched 4 other tests to determine more precisely what could cause this latency, with theses following differences from the code I gave above:
- A test without any pointer copy within the subscription's callbacks (the callbacks just do nothing)
- A test without any spin neither any subscription
- A test without subscription, but a 1-handler executor and calling `rclc_executor_spin_some` in the loop (even if there is nothing to do when spinning)
- A test with only one publisher instead of three

I will let you know the results of these next week.

Thanks for your help!

pablogs9, 2021-04-26:
Hello @anaelle-sw, so if I have not understood wrong:
1. The slowdown only appears between 15.5 and 30 hours after execution starts. After and before everything goes well.
2. The slowdown is caused by this code, somehow:
```c
if (iter % 5 == 0) {
    RCSOFTCHECK(rcl_publish(&bool_pub, &bool_msg, NULL));
  }
  if (iter % 20 == 0) {
    RCSOFTCHECK(rcl_publish(&empty_pub, &empty_msg, NULL));
  }
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1)));
```

Ideas:
1. Remove the `rclc_executor_spin_some`, does a basic publication-only node has the problem? Since all pubs are best-effort the only point when the client waits for an agent answer is in the spin time. This test will allow us to know if this client-to-agent sync is what is slowing down your code.
2. Regarding the `(Serial.available() > 0)` could you remember me please where we talked about that? I want to know if this is still necessary.
3. It would be nice if we replicate this in micro-ROS for Linux. The point here is that the timing issue can be related to MCU timing or to the serial port transport. I in micro-ROS for Linux also appears that means that there is some error in the logic of some layer. Can you share your ROS 2 checking node so we also can test?
4. Why are you using `float` as debug pub instead of `uint8`?

anaelle-sw, 2021-04-26:
Good morning @pablogs9,

first here are the results of the tests that has been running this week-end:
1. A test without any pointer copy within the subscription's callbacks (the callbacks just do nothing): the same problem appears (frequency slowed down after 15h30, and speed up to the desired frequency after 45 hours. After this, the frequency has stayed stable)
2. A test without any spin neither any subscription: no problem for almost two days (this should be a positive answer to your idea 1. , right?)
3. A test without subscription, but a 1-handler executor and calling `rclc_executor_spin_some` in the loop (even if there is nothing to do when spinning): after 63 hours, no messages were received by my ROS2 node as the "connection" between the agent and the DDS seems to have been stopped as in the agent output (there were only `[==>> SER <<==]` logs, but no more `[** <<DDS>> **]` logs). After I let `ros2 topic hz` for a few moments, the "connection" was re-established. It might not be related, so I guess the conclusion here is that no problem appeared for this test for 63 hours.
4. A test with only one publisher instead of three: the same problem appears (frequency slowed down after 15h30. This test has only been running for 23 hours, so I have no results for now about the frequency being right again)

To answer your questions and ideas:

>  1. The slowdown only appears between 15.5 and 30 hours after execution starts. After and before everything goes well.

Yes, that's right. For instance, I let a test run for almost 4 days. And after 30 hours, everything was going well.

>  2. The slowdown is caused by this code, somehow:

That's right too. According the results from the test 1 above, it is not because of the pointer copy in the subscriptions callbacks. According to the tests 2 and 4, the `rcl_publish` function is not making the frequency slowing down. And according to the test 3, `rclc_executor_spin_some` for an executor that has no subscription attached is not making the frequency slowing down (at least, no problem appears before 63 hours). So the conclusion here might be that precisely this line
```c
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1)));
```
execution is taking too much time after 15 hours, if there are some subscriptions added to the executor.

> 1. Remove the `rclc_executor_spin_some`

This is the test 1 above. So according to the results, I guess the client-to-agent sync is not the reason of the problem

> 2. Regarding the (Serial.available() > 0) could you remember me please where we talked about that? I want to know if this is still necessary.

It was about the agent freezing after a while, in this [post](https://github.com/micro-ROS/micro_ros_arduino/issues/183#issuecomment-805760375)

> 3. Can you share your ROS 2 checking node so we also can test?

Yes sure, I will post it in few minutes in a new post (so that it would be cleaner)

> 4.  Why are you using `float` as debug pub instead of `uint8`?

At first, I was trying to have something really close to our final application but with standard messages. We should be publishing a custom message that contains an `int` and a `float` at 100Hz. So when I tried to simplified the Arduino code for debugging purposes, I have made the `float` topic being published at 100Hz. Then I just doubled the number of publication for debugging purposes again. We can sure use an `uint8` instead, but the bandwidth used will  just be less "realistic" compared to our final application. But according to the results right above, the publishers are not causing the issue, so it shouldn't be a problem to switch to `uint8`.

anaelle-sw, 2021-04-26:
Here is the ROS2 debugging node that outputs some warning / errors messages if the frequency is not respected on all the topics:
```c
#include <cstdlib>
#include <rclcpp/rclcpp.hpp>
#include <std_msgs/msg/char.hpp>
#include <std_msgs/msg/int64.hpp>
#include <std_msgs/msg/bool.hpp>
#include <std_msgs/msg/float64.hpp>
#include <std_msgs/msg/empty.hpp>

class TestMicroRosConnection: public rclcpp::Node {
public:
  TestMicroRosConnection()
  : Node("test_micro_ros_connection_node",
         rclcpp::NodeOptions()
         .start_parameter_services(false)
         .start_parameter_event_publisher(false)){

    // Init int publisher
    int_pub = this->create_publisher<std_msgs::msg::Int64>(
          "/topic_int",
          rclcpp::QoS(1).best_effort());

    // Init int publisher timer
    int_pub_callback_group = this->create_callback_group(
          rclcpp::CallbackGroupType::MutuallyExclusive,
          false);
    int_pub_callback_group_executor_thread = std::thread([this]() {
      int_pub_callback_group_executor.add_callback_group(
            int_pub_callback_group,
            this->get_node_base_interface());
      int_pub_callback_group_executor.spin();
    });

    int_pub_timer = this->create_wall_timer(
          std::chrono::milliseconds(2000),
          std::bind(
            &TestMicroRosConnection::pub_int,
            this),
          int_pub_callback_group);

    // Init char publisher
    char_pub = this->create_publisher<std_msgs::msg::Char>(
          "/topic_char",
          rclcpp::QoS(1).best_effort());

    // Init char publisher timer
    char_pub_callback_group = this->create_callback_group(
          rclcpp::CallbackGroupType::MutuallyExclusive,
          false);
    char_pub_callback_group_executor_thread = std::thread([this]() {
      char_pub_callback_group_executor.add_callback_group(
            char_pub_callback_group,
            this->get_node_base_interface());
      char_pub_callback_group_executor.spin();
    });

    char_pub_timer = this->create_wall_timer(
          std::chrono::milliseconds(200),
          std::bind(
            &TestMicroRosConnection::pub_char,
            this),
          char_pub_callback_group);

    // Init bool subscription
    bool_sub_callback_group = this->create_callback_group(
          rclcpp::CallbackGroupType::MutuallyExclusive,
          false);
    auto bool_sub_options = rclcpp::SubscriptionOptions();
    bool_sub_options.callback_group = bool_sub_callback_group;
[... 304 lines cut ...]
  std::thread float_sub_callback_group_executor_thread;
  rclcpp::TimerBase::SharedPtr float_timeout_timer;
  rclcpp::CallbackGroup::SharedPtr float_timeout_callback_group;
  rclcpp::executors::SingleThreadedExecutor float_timeout_callback_group_executor;
  std::thread float_timeout_callback_group_executor_thread;
  bool float_timeout_reached = false;
  rclcpp::Time float_last_msg_time;
  int float_last_msg_data;

  // Debug
  rclcpp::Time start_time;
};

//***********
//* Main
//***********
int main(int argc, char **argv){
  rclcpp::init(argc, argv);
  auto node = std::make_shared<TestMicroRosConnection>();
  rclcpp::spin(node);
  rclcpp::shutdown();
  return 0;
}

```

anaelle-sw, 2021-04-26:
And here is the ROS2 debugging node that allows to check the time between to float message, and the loop's step number:
```c
#include <cstdlib>
#include <rclcpp/rclcpp.hpp>
#include <std_msgs/msg/char.hpp>
#include <std_msgs/msg/int64.hpp>
#include <std_msgs/msg/bool.hpp>
#include <std_msgs/msg/float64.hpp>
#include <std_msgs/msg/empty.hpp>

class TestMicroRosConnection: public rclcpp::Node {
public:
  TestMicroRosConnection()
  : Node("log_debug_node",
         rclcpp::NodeOptions()
         .start_parameter_services(false)
         .start_parameter_event_publisher(false)) {

    // Init float subscription
    float_sub_callback_group = this->create_callback_group(
          rclcpp::CallbackGroupType::MutuallyExclusive,
          false);
    auto float_sub_options = rclcpp::SubscriptionOptions();
    float_sub_options.callback_group = float_sub_callback_group;
    float_sub_callback_group_executor_thread = std::thread([this]() {
      float_sub_callback_group_executor.add_callback_group(
            float_sub_callback_group,
            this->get_node_base_interface());
      float_sub_callback_group_executor.spin();
    });

    float_sub = this->create_subscription<std_msgs::msg::Float64>(
          "/topic_float",
          rclcpp::QoS(1).best_effort(),
          std::bind(
            &TestMicroRosConnection::float_callback,
            this,
            std::placeholders::_1),
          float_sub_options);
  }

  ~TestMicroRosConnection(){
    float_sub_callback_group_executor.cancel();
    float_sub_callback_group_executor_thread.join();
  }

private:
  void float_callback(const std_msgs::msg::Float64::SharedPtr msg) {
    std::cout <<
          "Float received : " <<
          msg->data <<
          " after " <<
          this->now().nanoseconds() - float_last_msg_time.nanoseconds() <<
          " ns since last message" <<
          std::endl;
    float_last_msg_time = this->now();
    return;
  }

  //********************
  // Attributes
  //********************
  rclcpp::Subscription<std_msgs::msg::Float64>::SharedPtr float_sub;
  rclcpp::CallbackGroup::SharedPtr float_sub_callback_group;
  rclcpp::executors::SingleThreadedExecutor float_sub_callback_group_executor;
  std::thread float_sub_callback_group_executor_thread;
  rclcpp::Time float_last_msg_time;
};

//***********
//* Main
//***********
int main(int argc, char **argv){
  rclcpp::init(argc, argv);
  auto node = std::make_shared<TestMicroRosConnection>();
  rclcpp::spin(node);
  rclcpp::shutdown();
  return 0;
}

```

pablogs9, 2021-04-26:
Ok, I'm going to try to replicate this.

Meanwhile an idea to test: as explained [here](https://github.com/micro-ROS/micro_ros_arduino/issues/183#issuecomment-805760375) the `while (Serial.available() > 0) {...}` fix was used to handle the hearbeats when all this assumptions are true **at the same time**:
1. `DRMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0`
2. All pubs are best effort
3. There is no subs or the executor is empty

In this case, the executor is not empty, so `rclc_executor_spin_some` will call `rmw_wait` since it has something to wait for. Then `rmw_wait` will use [`uxr_run_session_until_data`](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/include/uxr/client/core/session/session.h#L359), so the communication between client and agent is going to take place and **reading** from the serial port at any other point can cause interference.

So, can you test it **just removing** the  `while (Serial.available() > 0) {...}` part?

anaelle-sw, 2021-04-26:
Thanks for your answer.

> So, can you test it **just removing** the `while (Serial.available() > 0) {...}` part?

Ok, I will launch a test without this

pablogs9, 2021-04-26:
BTW share also your `colcon.meta` please

anaelle-sw, 2021-04-26:
Yes sure, here it is:
```c
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS=3"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=3",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=2",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```
I didn't modified it for the tests, I just used to one we use for our final application

anaelle-sw, 2021-04-26:
Quick update: The frequency is back to the expected value after 29 hours for the test 4 (with only one publisher instead of three). When the frequency was too low, the results about the duration of each step are similar to the ones I gave [when opening this issue](https://github.com/micro-ROS/micro_ros_arduino/issues/232#issue-866232792) It shows that `rcl_publish` are not the problematic call, but that only `RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1))); `is.

pablogs9, 2021-04-26:
No results about the test without `while (Serial.available() > 0) {...}` yet?

anaelle-sw, 2021-04-26:
Not yet, it might takes until 2am tonight, I will come back tomorrow morning with the results

pablogs9, 2021-04-26:
Nice, tomorrow we will see. Huge patience required for this testing :man_facepalming:

anaelle-sw, 2021-04-27:
Hi @pablogs9 !
So about the test without `while (Serial.available() > 0) {...}`, the frequency has dropped down after 16h20. We should be able to know if it goes back to desired value around 5pm today

anaelle-sw, 2021-04-27:
Update: the frequency of the test without `while (Serial.available() > 0) {...}` goes back to expected value after 29h30. And as for all the tests containing subscriptions, the loop's step that takes too much time is calling `rclc_executor_spin_some`

pablogs9, 2021-04-28:
Some other ideas:
- Replicate the case in [micro-ROS for Linux](https://micro.ros.org/docs/tutorials/core/first_application_linux/). Add your code to [this repo](https://github.com/micro-ROS/micro-ROS-demos) and check if this also happens when using UDP transports between two Linux machines using micro-ROS and XRCE.
- Is there any chance of a misbehaving of [this](https://github.com/micro-ROS/micro_ros_arduino/blob/d1373929dc521ca229ad1fe6b6ed24d2fc8f536c/src/default_transport.cpp#L17) functions. Does the Teensy support also provide a `micros()` function that rolls every 4294967295 microseconds?
- I have seen that Teensyduino offers [`ellapsedMicros`](https://www.pjrc.com/teensy/td_timing_elaspedMillis.html) function that handles internally the rollover but is not Arduino standard. Can you try with a custom timing function that uses this?

pablogs9, 2021-04-28:
Another interesting thing for us to debug this is:
- A copy-pasted text of the output of the agent with `-v6` during the normal operation
- A copy-pasted text of the output of the agent with `-v6` during the slow operation

anaelle-sw, 2021-04-28:
> Replicate the case in micro-ROS for Linux

I may be able to try to replicate with micro-ROS for Linux later, but not in the next days. If you have the time to try this before I do, let me know how it turns out. Thanks for your help!

> Is there any chance of a misbehaving of this functions. Does the Teensy support also provide a micros() function that rolls every 4294967295 microseconds?

Yes, the Teensy provides the same [`micros()`](https://www.pjrc.com/teensy/td_timing_millis.html) than an Arduino board. In my understanding of how `micros()` works, I don't think the function you linked can misbehave as it manages properly the rollover.

> I have seen that Teensyduino offers ellapsedMicros function that handles internally the rollover but is not Arduino standard. Can you try with a custom timing function that uses this?

Hm I don't really get what you want me to test here... I mean, the loop function frequency is slowed down around 15 - 17 hours, which is way less than the max duration that can contain `millis()` ( = around 50 days). In our code we only use `millis()` (and never `micros()`), so we shouldn't have any problem before 50 days. Plus, thanks to the "debugging topic", I was able to identify the function `rclc_executor_spin_some` as the reason of the slowdown. So I don't understand what you me to change exactly, and how it should influence the test. Could you give an example please?
That's been said, I agree that using Teensy's `ellapsedMicros` instead of `millis()` could be a good way to manage more cleanly the loop rate and to eliminate the problems related to the rollover. Thank you for pointing that out! We will surely try to implement this at some point.

> A copy-pasted text of the output of the agent with -v6 during the normal operation
> A copy-pasted text of the output of the agent with -v6 during the slow operation

Yes I can provide these soon, but need to wait a bit for the slowdown to happen... Just to let you know, I didn't notice any error or especially unusual outputs from the agent while both normal and slow operation (I ran all the test while having -v6 option)

anaelle-sw, 2021-04-28:
Here is the agent's output during normal operation, right after launching the agent, during approximatively 1 second:
```c
[1619627866.513020] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1619627866.513140] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1619627867.172612] info     | Root.cpp           | create_client            | create                 | client_key: 0x5083E048, session_id: 0x81
[1619627867.172712] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1350819912, address: 0
[1619627867.172886] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5083E048, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1619627868.168032] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 112, data: 
0000: 81 01 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 18 00 00 00
[1619627868.168119] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 124, data: 
0000: 81 01 01 00 01 05 72 00 00 0B 00 02 02 02 00 00 64 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 74 6F 70 69 63 5F 62 6F 6F 6C 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54
0040: 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 42 6F 6F 6C 5F 3C 2F
0060: 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00
[1619627868.168170] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 24, data: 
0000: 81 01 02 00 01 05 0F 00 00 0C 00 03 03 02 00 00 01 00 00 00 00 00 01 73
[1619627868.168568] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 344, data: 
0000: 81 01 03 00 01 05 50 01 00 0D 00 05 05 02 00 00 42 01 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 68 69 73 74 6F 72 79 4D 65 6D 6F 72 79 50 6F 6C 69 63 79 3E 50 52 45 41 4C
0040: 4C 4F 43 41 54 45 44 5F 57 49 54 48 5F 52 45 41 4C 4C 4F 43 3C 2F 68 69 73 74 6F 72 79 4D 65 6D
0060: 6F 72 79 50 6F 6C 69 63 79 3E 3C 71 6F 73 3E 3C 72 65 6C 69 61 62 69 6C 69 74 79 3E 3C 6B 69 6E
0080: 64 3E 42 45 53 54 5F 45 46 46 4F 52 54 3C 2F 6B 69 6E 64 3E 3C 2F 72 65 6C 69 61 62 69 6C 69 74
00A0: 79 3E 3C 2F 71 6F 73 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E
00C0: 64 3E 3C 6E 61 6D 65 3E 72 74 2F 74 6F 70 69 63 5F 62 6F 6F 6C 3C 2F 6E 61 6D 65 3E 3C 64 61 74
00E0: 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 42 6F 6F 6C 5F
0100: 3C 2F 64 61 74 61 54 79 70 65 3E 3C 68 69 73 74 6F 72 79 51 6F 73 3E 3C 6B 69 6E 64 3E 4B 45 45
0120: 50 5F 41 4C 4C 3C 2F 6B 69 6E 64 3E 3C 2F 68 69 73 74 6F 72 79 51 6F 73 3E 3C 2F 74 6F 70 69 63
0140: 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E 3C 2F 64 64 73 3E 00 00 03
[1619627868.168680] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 128, data: 
0000: 81 01 04 00 01 05 76 00 00 0E 00 12 02 02 00 00 68 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 74 6F 70 69 63 5F 66 6C 6F 61 74 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61
0040: 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 46 6C 6F 61 74 36
0060: 34 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 69 6E
[1619627868.168791] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 24, data: 
0000: 81 01 05 00 01 05 0F 00 00 0F 00 13 03 02 00 00 01 00 00 00 00 00 01 73
[1619627868.169234] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 348, data: 
0000: 81 01 06 00 01 05 54 01 00 10 00 15 05 02 00 00 46 01 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 68 69 73 74 6F 72 79 4D 65 6D 6F 72 79 50 6F 6C 69 63 79 3E 50 52 45 41 4C
0040: 4C 4F 43 41 54 45 44 5F 57 49 54 48 5F 52 45 41 4C 4C 4F 43 3C 2F 68 69 73 74 6F 72 79 4D 65 6D
0060: 6F 72 79 50 6F 6C 69 63 79 3E 3C 71 6F 73 3E 3C 72 65 6C 69 61 62 69 6C 69 74 79 3E 3C 6B 69 6E
0080: 64 3E 42 45 53 54 5F 45 46 46 4F 52 54 3C 2F 6B 69 6E 64 3E 3C 2F 72 65 6C 69 61 62 69 6C 69 74
00A0: 79 3E 3C 2F 71 6F 73 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E
00C0: 64 3E 3C 6E 61 6D 65 3E 72 74 2F 74 6F 70 69 63 5F 66 6C 6F 61 74 3C 2F 6E 61 6D 65 3E 3C 64 61
00E0: 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 46 6C 6F 61
0100: 74 36 34 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 68 69 73 74 6F 72 79 51 6F 73 3E 3C 6B 69 6E 64
0120: 3E 4B 45 45 50 5F 41 4C 4C 3C 2F 6B 69 6E 64 3E 3C 2F 68 69 73 74 6F 72 79 51 6F 73 3E 3C 2F 74
0140: 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E 3C 2F 64 64 73 3E 00 00 13
[... cut ...]
[1619627868.961345] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 00 40
[1619627868.961349] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 08 40
```
I will provide the outputs during the slowdown when it will be happening by tomorrow.

Also I changed the way the loop rate is managed by using Teensy's `elapsedMillis`, but I am not sure this is what you wanted me to test

pablogs9, 2021-04-29:
Regarding `elapsedMicros` I was proposing to change [this](https://github.com/micro-ROS/micro_ros_arduino/blob/d1373929dc521ca229ad1fe6b6ed24d2fc8f536c/src/default_transport.cpp#L17) function to use `elapsedMicros`. This is the timing function for the whole micro-ROS stack and as you can see it handles the `micros()` rollover. 

But taking into account that Teensyduino has  `elapsedMicros` it can be something like:

```c
# Not tested

elapsedMicros teensymicros;

 int clock_gettime(clockid_t unused, struct timespec *tp)
  {
    (void)unused;

    uint64_t m = teensymicros;
    tp->tv_sec = m / 1000000;
    tp->tv_nsec = (m % 1000000) * 1000;

    return 0;
  }
```

anaelle-sw, 2021-04-29:
> Regarding `elapsedMicros` I was proposing to change [this](https://github.com/micro-ROS/micro_ros_arduino/blob/d1373929dc521ca229ad1fe6b6ed24d2fc8f536c/src/default_transport.cpp#L17) function to use `elapsedMicros`.

Oh ok, thanks! We can try this, I will let you know how it turns out

Acuadros95, 2021-04-29:
Hello @anaelle-sw, 

I will test this problem on linux and on a Teensy 3.2, could you give us the complete package for the ROS2 debugging nodes?

anaelle-sw, 2021-04-29:
Hello @Acuadros95, here is an archive with the original .ino code (as given in the first post of this issue), and both debugging nodes: 
[test_microros_frequency.zip](https://github.com/micro-ROS/micro_ros_arduino/files/6398410/test_microros_frequency.zip)

I launch the node `test_microros_connection_node` after having flashed the Teensy and launched the agent. And occasionally, I launch the node `log_display_node` to get the amount of time between each step. Because the publisher as a best-effort QoS policy, all the messages are not necessarily received. So I use a spreadsheet (the one I gave in first post) to analyze the results. Let me know if you need a cleaner version of this spreadsheet.

Also, please note that you will need either to export `ROS_DOMAIN_ID=24`, either to modify the .ino code.

Thank you very much for your time!

anaelle-sw, 2021-04-29:
And here are the outputs of the agent while the slowdown is occurring, for approximatively 1 second:
```c
[1619698909.110466] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 4, data: 
0000: 14 00 00 00
[1619698909.110489] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 8, data: 
0000: 0C 00 00 00 00 00 00 00
[1619698909.110509] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5083E048, len: 16, data: 
0000: 81 01 4A F0 09 01 08 00 00 1B 00 16 14 00 00 00
[1619698909.110518] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 4B F0 09 01 0C 00 00 17 00 06 0C 00 00 00 00 00 00 00
[1619698909.111070] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 94 1C 07 01 0C 00 24 66 00 15 00 00 00 00 00 00 00 40
[1619698909.111084] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 95 1C 07 01 0C 00 24 67 00 15 00 00 00 00 00 00 08 40
[1619698909.111089] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 96 1C 07 01 0C 00 24 68 00 15 00 00 00 00 00 00 10 40
[1619698909.111092] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 97 1C 07 01 0C 00 24 69 00 15 00 00 00 00 00 00 F0 3F
[1619698909.111166] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 00 40
[1619698909.111188] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 08 40
[1619698909.111207] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 10 40
[1619698909.111227] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 F0 3F
[1619698909.115683] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 16, data: 
0000: 81 01 98 1C 07 01 05 00 24 6A 00 05 00 00 00 00
[1619698909.115697] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 99 1C 07 01 0C 00 24 6B 00 15 00 00 00 00 00 00 00 40
[1619698909.115702] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9A 1C 07 01 0C 00 24 6C 00 15 00 00 00 00 00 00 08 40
[1619698909.115780] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 1, data: 
0000: 00
[1619698909.115800] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 00 40
[1619698909.115811] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 08 40
[1619698909.135492] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9B 1C 07 01 0C 00 24 6D 00 15 00 00 00 00 00 00 10 40
[1619698909.135508] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9C 1C 07 01 0C 00 24 6E 00 15 00 00 00 00 00 00 F0 3F
[1619698909.135603] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 10 40
[1619698909.135627] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 8, data: 
0000: 00 00 00 00 00 00 F0 3F
[1619698909.310614] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 4, data: 
0000: 14 00 00 00
[1619698909.310660] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5083E048, len: 16, data: 
0000: 81 01 4C F0 09 01 08 00 00 1B 00 16 14 00 00 00
[1619698909.316489] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9D 1C 07 01 0C 00 24 6F 00 15 00 00 00 00 00 00 00 40
[1619698909.316504] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9E 1C 07 01 0C 00 24 70 00 15 00 00 00 00 00 00 08 40
[1619698909.316509] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 9F 1C 07 01 0C 00 24 71 00 15 00 00 00 00 00 00 10 40
[1619698909.316514] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 20, data: 
0000: 81 01 A0 1C 07 01 0C 00 24 72 00 15 00 00 00 00 00 00 F0 3F
[... cut ...]
[1619698910.016490] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5083E048, len: 16, data: 
0000: 81 01 AD 1C 07 01 05 00 24 7F 00 05 00 00 00 00
[1619698910.016604] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 1, data: 
0000: 00
[1619698910.110560] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 4, data: 
0000: 14 00 00 00
```

anaelle-sw, 2021-04-29:
> Regarding `elapsedMicros` I was proposing to change [this](https://github.com/micro-ROS/micro_ros_arduino/blob/d1373929dc521ca229ad1fe6b6ed24d2fc8f536c/src/default_transport.cpp#L17) function to use [...]

I have copied these changes in "Arduino/librairies/micro_ros_arduino/src/default_transport.cpp", rebuild the library with the rolling docker, and launched the test. The changes work for now, the Teensy is receiving and sending correctly the messages. I will let you know if the frequency drops after a certain amount of time

pablogs9, 2021-04-30:
In the meantime, @Acuadros95 is preparing a test set for testing this issue on our side both on Linux and on Teensy. We will start the test next Tuesday.

anaelle-sw, 2021-04-30:
Hello @pablogs9 and @Acuadros95 ,
here are some interesting results: the last test (the one with the function `clock_gettime` modified in order to use `elapsedMicros` instead of `micros()`) has been running for 21 hours now, and no frequency problem occurred. I can let it run for the week-end and confirm the durability

anaelle-sw, 2021-05-03:
> the last test (the one with the function `clock_gettime` modified in order to use `elapsedMicros` instead of `micros()`)

Some updates about this test: It has been running for almost 90 hours now, and only "occasional" frequency problems appeared (only 4 frequency error messages, after around 54h, 62h, 73h, and 83h). So replacing `micros()` by `elapsedMicros` in `clock_gettime` seems to fix the issue. It is just a bit strange to me, as `clock_gettime` seems to properly handle the `micros()` rollover

pablogs9, 2021-05-04:
Ok, so that seems to be solved... if this is a Teensy issue, I'm ok with adding new functionality that only builds for Teensy boards, using the fact that `clock_gettime()` is defined as [weak](https://github.com/micro-ROS/micro_ros_arduino/blob/539af5462fb386d9a14528ee260ff5cc834f56d2/src/default_transport.cpp#L9).

But we are worried because we are not able to see where is the rollover handling inside these `elapsedMicros`, check [here](https://github.com/PaulStoffregen/cores/blob/5bc5f3a62e962182108576b1f95b8f531606dcd7/teensy/elapsedMillis.h#L57). Any idea or something we are missing @anaelle-sw ?

pablogs9, 2021-05-11:
I'm closing this by now. Reopen if you have any other rate problems.