micro-ROS/micro_ros_arduino #258: subscribing to control_msgs/msg/JointJog causes failure

KohlhardtC, 2021-05-13:
## Issue template

- Hardware description: Teensy 4.1
- RTOS: Arduino BareMetal
- Version or commit hash: foxy

#### Steps to reproduce the issue

While testing the NativeEthernet implementation that I'm working on, I tried to subscribe to a topic that uses the JointJog message and discovered that this message type causes the system to hang. 

I've modified the micro-ros_subscriber_twist.ino example to provide a serial example of the problem:

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

//#include <geometry_msgs/msg/twist.h> 
#include <control_msgs/msg/joint_jog.h> 

rcl_subscription_t subscriber;
//geometry_msgs__msg__Twist msg; 
control_msgs__msg__JointJog msg; 
rclc_executor_t executor;
rcl_allocator_t allocator;
rclc_support_t support;
rcl_node_t node;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

//twist message cb
void subscription_callback(const void *msgin) {
  //const geometry_msgs__msg__Twist * msg = (const geometry_msgs__msg__Twist *)msgin;
  const control_msgs__msg__JointJog * msg = (const control_msgs__msg__JointJog *)msgin;
  // if velocity in x direction is 0 turn off LED, if 1 turn on LED
  digitalWrite(LED_PIN, (msg->duration == 0) ? LOW : HIGH);
}

void setup() {
  set_microros_transports();
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

   //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    //ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist), 
	ROSIDL_GET_MSG_TYPE_SUPPORT(control_msgs, msg, JointJog), 
    "micro_ros_arduino_jj_subscriber"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));

}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

Command to send message that should change the LED to LOW:
`ros2 topic pub /micro_ros_arduino_jj_subscriber control_msgs/msg/JointJog "{ header:{stamp:{sec: 10, nanosec: 20 }  , frame_id: 'base_link'  }  ,joint_names:['blah'], displacements:[100], velocities:[1],  duration: 0}"
`  

#### Expected behavior

LED should turn off

#### Actual behavior

LED does not turn off

pablogs9, 2021-05-13:
Hello @KohlhardtC you are not initing the `control_msgs__msg__JointJog` message... We have a lot of issues like this so I have wrote a brief explanation in our webpage: https://github.com/micro-ROS/micro-ROS.github.io/pull/312. It is not merged yet but you can read it [here](https://github.com/micro-ROS/micro-ROS.github.io/blob/58aa5c2de60988e7d4b39501205ede13a61f5088/_docs/tutorials/advanced/handling_type_memory/index.md).

In general you need to initialize sequence memory for the `control_msgs__msg__JointJog msg`:

```c
#define STRING_SIZE 100
#define MAX_ENTITIES 3

// Init header frame id string
msg.header.frame_id.capacity = STRING_SIZE;
msg.header.frame_id.size = 0;
msg.header.frame_id.data = (char*) malloc(STRING_SIZE * sizeof(char));

// Init sequence of strings
msg.joint_names.capacity = MAX_ENTITIES;
msg.joint_names.size = 0;
msg.joint_names.data = (rosidl_runtime_c__String*) malloc(MAX_ENTITIES * sizeof(rosidl_runtime_c__String));

// Init each string of the sequence of strings
for(int i = 0; i < MAX_ENTITIES; i++)
{
    msg.joint_names.data[i].data.capacity = STRING_SIZE;
    msg.joint_names.data[i].data.size = 0;
    msg.joint_names.data[i].data.data = (char*) malloc(STRING_SIZE * sizeof(char));
}

// Init doubles sequences
msg.displacements.capacity = MAX_ENTITIES;
msg.displacements.size = 0;
msg.displacements.data = (double*) malloc(MAX_ENTITIES * sizeof(double));

msg.velocities.capacity = MAX_ENTITIES;
msg.velocities.size = 0;
msg.velocities.data = (double*) malloc(MAX_ENTITIES * sizeof(double));
```

KohlhardtC, 2021-05-14:
Thanks for the tip, I never would have guessed this to be the case. 

Would it be possible to programmatically initialize the memory sequence to help reduce this workload for end users? I am a novice C++ developer and don't have enough grasp on the language to know if there is any chance of making this more convenient.  I wonder how ROS handles this internally? Or maybe we could generate a python script that reads in message files and produces initialization code? It just feels like this is a lot of extra busy work that could be automated in some way.

KohlhardtC, 2021-05-14:
BTW, thank you for sharing the initialization code. I actually couldn't get it to compile because there was an extra reference to 'data' in the for loop. Here's the working code:

```
#define STRING_SIZE 100
#define MAX_ENTITIES 3

// Init header frame id string
msg.header.frame_id.capacity = STRING_SIZE;
msg.header.frame_id.size = 0;
msg.header.frame_id.data = (char*) malloc(STRING_SIZE * sizeof(char));

// Init sequence of strings
msg.joint_names.capacity = MAX_ENTITIES;
msg.joint_names.size = 0;
msg.joint_names.data = (rosidl_runtime_c__String*) malloc(MAX_ENTITIES * sizeof(rosidl_runtime_c__String));

// Init each string of the sequence of strings
for(int i = 0; i < MAX_ENTITIES; i++)
{
    msg.joint_names.data[i].capacity = STRING_SIZE;
    msg.joint_names.data[i].size = 0;
    msg.joint_names.data[i].data = (char*) malloc(STRING_SIZE * sizeof(char));
}

// Init doubles sequences
msg.displacements.capacity = MAX_ENTITIES;
msg.displacements.size = 0;
msg.displacements.data = (double*) malloc(MAX_ENTITIES * sizeof(double));

msg.velocities.capacity = MAX_ENTITIES;
msg.velocities.size = 0;
msg.velocities.data = (double*) malloc(MAX_ENTITIES * sizeof(double));
```

pablogs9, 2021-05-14:
Thanks for the correction! I didn't try it...

The automation of that is required, I agree. But there are some problems regarding the dynamic/static memory selection or on how long are the sequences. The former is typically an application-specific decision, so automating this number can be tricky, but of course, this code can be autogenerated.

I'm going to think about it. Meanwhile, I'm closing this because solved, do not hesitate to reopen if you have other problems.

closed 2021-05-14 by pablogs9