micro-ROS/micro_ros_arduino #269: executor only calling subscription callback when 2 datapoints are available

JTaets, 2021-05-19:
## Issue template

- Hardware description: micro-ros on Teensy 4.1
- RTOS: Linux with PREEMPT kernel

I am sending 100 datapoints from the computer to the teensy at 100Hz at the topic `control` using `rclpy`
Running `ros2 topic hz /control` confirms me that data is indeed sent at 100Hz.

At the Teensy, i am recording when the subscription its callback is called by using `millis()`
Then after all the data is received, i sent the 100 time instances from when the data was received back to the computer on the topic `deltat`. 

Then i take the finite differerence of the timepoints, to get the difference in time between samples being recieved by the Teensy.
Since I send data at 100Hz, I suspect these finite differences to be [10,10,10,10....] (in milliseconds)
But I get [20, 0, 20, 0, 20, 0, 20, 0...]
So i guess my callback on the Teensy only executes when it receives 2 datapoints, although i have `ON_NEW_DATA`
I tried playing a bit with the qos of the subscription at the side of the Teensy, but it didn't do anything.

As a second question, sometimes there are big spikes in the [20, 0, 20, 0, 20, 0, 20, 0...]
Sometimes I get something more like [20, 0,  ....  ,1450, 153, 0, 20, 0, 20, 0...]

My code on the side of the Teensy is:
``` c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/int32_multi_array.h>

rcl_publisher_t publisher;
rcl_subscription_t subscriber;
std_msgs__msg__Int32 msg;
std_msgs__msg__Int32 deltat_msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
int nr_data;
int times[100] = {-1};

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void * msgin)
{  
  const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
  if (nr_data < 99) {
    nr_data+=1; 
  }  
  times[nr_data] = millis();

}

void setup() {
  set_microros_transports();
  
[... 314 lines cut ...]
[INFO] [1621438213.756455045] [pc_control]: recieving: "1645770"
[INFO] [1621438213.757664496] [pc_control]: recieving: "1645790"
[INFO] [1621438213.758946826] [pc_control]: recieving: "1645790"
[INFO] [1621438213.760291947] [pc_control]: recieving: "1645810"
[INFO] [1621438213.761926638] [pc_control]: recieving: "1645810"
[INFO] [1621438213.763778509] [pc_control]: recieving: "1645830"
[INFO] [1621438213.765526870] [pc_control]: recieving: "1645830"
[INFO] [1621438213.766965920] [pc_control]: recieving: "1645850"
[INFO] [1621438213.767580620] [pc_control]: recieving: "1645850"
[INFO] [1621438213.768203561] [pc_control]: recieving: "1645870"
[INFO] [1621438213.768772541] [pc_control]: recieving: "1645870"
[INFO] [1621438213.769418001] [pc_control]: recieving: "1645890"
[INFO] [1621438213.770122822] [pc_control]: recieving: "1645890"
[INFO] [1621438213.770902872] [pc_control]: recieving: "1645910"
[INFO] [1621438213.772618733] [pc_control]: recieving: "1645910"
[INFO] [1621438213.773022403] [pc_control]: recieving: "1645930"
[INFO] [1621438213.773630283] [pc_control]: recieving: "1645930"
[INFO] [1621438213.774017094] [pc_control]: recieving: "1645950"
[INFO] [1621438213.775121784] [pc_control]: recieving: "1645950"
[INFO] [1621438213.775504904] [pc_control]: recieving: "1645970"
[INFO] [1621438213.776193145] [pc_control]: recieving: "1645970"
[INFO] [1621438213.776706555] [pc_control]: recieving: "1645998"
[INFO] [1621438213.777041325] [pc_control]: Reading t: "[1644998, 1645010, 1645010, 1645030, 1645030, 1645050, 1645050, 1645070, 1645070, 1645090, 1645090, 1645110, 1645110, 1645130, 1645130, 1645150, 1645150, 1645170, 1645170, 1645190, 1645190, 1645210, 1645210, 1645230, 1645230, 1645250, 1645250, 1645270, 1645270, 1645290, 1645290, 1645310, 1645310, 1645330, 1645330, 1645350, 1645350, 1645370, 1645370, 1645390, 1645390, 1645410, 1645410, 1645430, 1645430, 1645450, 1645450, 1645470, 1645470, 1645490, 1645490, 1645510, 1645510, 1645530, 1645530, 1645550, 1645550, 1645570, 1645570, 1645590, 1645590, 1645610, 1645610, 1645630, 1645630, 1645650, 1645650, 1645670, 1645670, 1645690, 1645690, 1645710, 1645710, 1645730, 1645730, 1645750, 1645750, 1645770, 1645770, 1645790, 1645790, 1645810, 1645810, 1645830, 1645830, 1645850, 1645850, 1645870, 1645870, 1645890, 1645890, 1645910, 1645910, 1645930, 1645930, 1645950, 1645950, 1645970, 1645970, 1645998]"
[INFO] [1621438213.777332545] [pc_control]: Delta_t: "[12, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 20, 0, 28]"
```

pablogs9, 2021-05-20:
@JanStaschulat any idea on this?

pablogs9, 2021-05-20:
I'm going to make some tests, but could you try meanwhile to spin the executor faster?

Maybe something like:

```c
void loop() {
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(5)));
...
}
```

JanStaschulat, 2021-05-20:
First, 
`
RCCHECK(rclc_executor_spin_some(&executor, RCL_S_TO_NS(50)));
`
is probably a typo. You probably don't want a timeout of 5 seconds when the incoming data rate is 100Hz (e.g. 10ms). Default value would be 1s. It is the duration of the (blocking) wait for new data from DDS-queue.

Your problem seems to be related to the execution time of the subscription callback - or due to some other process which runs on the arduino. When you see the sequence of 0,20,0,20 it means, that two messages are available at DDS queue and are being processed one after the other. But the execution of the callback takes longer than 10ms - or it is preempted by some other tasks that run on arduino. 

You could make an experiment to reduce the sending rate to, say, 10Hz and keep everything as it is. Then, the messages should arrive every 100ms. 

Further more, which options do you have to increase the real-time priority of the process on arduino? I would give it a higher priority than all other tasks that you are running on it. 

That you sometimes see a larger value and then multiple times just 0's in the timestamps also indicates that the CPU is overloaded - for whatever reason. Then the messages are buffered and processed at once. 

I can't imagine, why the executor will wait for every second data, because you specified ON_NEW_DATA. So when new data is available, then the callback is called.

JanStaschulat, 2021-05-20:
What is the resolution of `millis()` on arduino?
https://www.pjrc.com/teensy/td_timing_millis.html
The last value is always zero, which is unexpected. I am not familiar with arduino. Where is the `loop()` function called? I did not see this in the code example.  Is it maybe called periodically every 10ms?

pablogs9, 2021-05-20:
Hello @JanStaschulat, I guess that he/she does not have thread because usually in Teensy platforms you don't have an (RT)OS.

Regarding the `loop` function, you can imagine the Arduino entry point like:

```c
void main(){
   setup();
   for(;;){
      loop();
      handle_internal_events(); \\ Serial, interrupts...
   }
}
```

I agree with decreasing the spin time. The middleware internally, when `rlc_wait` is called returns when receives just one DDS data. Does this explain anything regarding the executor behavior here?

JanStaschulat, 2021-05-20:
Yes, when `rcl_wait` returns, the Executor can only process one data and call the subscription callback once. The timeout is here irrelevant (1s or 5s - whatever). It will just return after this timeout when no data is available. Which will never happen in this use-case. 

I suspect, that from outside the frequency of calling the `loop()` is not continuous - but sampled.
As said before, the last digit of the timestamp (in ms) is always zero. That indicates some base periodicity of the `loop` itself or in the function of `handle_internal_events` -maybe there is a polling of 10ms there?

JTaets, 2021-05-20:
Hi all, thanks for the quick responses.

As suggested I tried this and it worked!

> Maybe something like:
> 
> ```c
> void loop() {
>   RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(5)));
> ...
> }
> ```

I now get clear [10,10,10,10,...].
The 100 Hz wasn't the problem I think. Doing the same at 10 Hz gave me [ 0 200 0 200 ...]

With the new `RCL_MS_TO_NS(5)` added, I also tried 200Hz to see if there is a polling of 10ms as suggested above.
I got as result
``` py
[INFO] [1621509226.089462122] [pc_control]: Delta_t: "[5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 2, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 24290, 8, 0, 10, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5]"
```

As you can see, the occasional high value still remains. Any ideas on how to fix this?

> Further more, which options do you have to increase the real-time priority of the process on arduino? I would give it a higher priority than all other tasks that you are running on it.
>
> That you [...]

The only code that is running on the Teensy is the code that I posted here (unless ofcourse background arduino code I don't have control over).
Is there any resource or sample code to point me in the right direction of how to do this?

The help is much appreciated!
Kind regards.

pablogs9, 2021-05-20:
In order to reduce the "high values" you can try to tune the micro-ROS internal buffers. For example I would try to increase `DRMW_UXRCE_MAX_HISTORY=10` and `RMW_UXRCE_STREAM_HISTORY=8` in [colcon.meta](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/colcon.meta) and then rebuild the micro-ROS library.

Take into account that this is not rebuilding your app but the **micro-ROS library**. Check [here](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) and [here](https://github.com/micro-ROS/micro_ros_arduino/issues/14#issuecomment-722242175).

pablogs9, 2021-05-20:
I'm closing this, please reopen if you still have issues. Thanks for the feedback!

closed 2021-05-20 by pablogs9

JanStaschulat, 2021-05-20:
To optimize performance of the executor, i propose not to check every 5ms for new data, but using a larger timeout, e.g. 100ms:
```
void loop() {
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
...
}
```

JTaets, 2021-05-21:
I got the occasional "high value" fixed. It was a small bug in the code above. `nr_data` wasn't reset properly sometimes.

However, the `RCL_MS_TO_NS(100)` proposed doesn't work.
Then i get the [0 20 0 20 0 20].
I really need the small value.

JanStaschulat, 2021-05-21:
@JTaets That is an interesting observation, that a timeout larger than the frequency does not work. In our examples we always set the timeout to a large value.

JTaets, 2021-05-21:
Apologies for bothering again.
(I can open another issue if you prefer this)

I suspect that ros2 polls the subscription on the Teensy at 1 kHz.
I am wondering if it is possible to increase that polling rate in some way?

The rest of this message is explanation of why I think the polling rate is 1 kHz.
You can skip the rest of the message if you believe that.

The reason I suspect this is that the pure arduino code:
``` c
void setup() {
  Serial.begin(250000); // baud rate
  while (!Serial) {
    ; // Wait for serial port to connect
  }
  Serial.setTimeout(0.1); // sets the maximum milliseconds to wait for serial data, defaults to 1000ms
}
void loop() {
  Serial.println(micros()); 
}
```
Gives (by watching on the Serial Monitor on the Arduino software):
``` c
...
161
162
163
163
164
165
...
```
So the `loop()` in arduino is called as fast as possible and there is no mistake in `micros()`.

My code posted above with `millis()` changed to `micros()` gives absolute timings `(msg_t)` like:
 [14010, 24010, 34010, 35010...]
There are 10000µs inbetween them as expected at 100Hz, But it always ends in ...010µs.
At very few instances, there is 11000µs between them, not something like 10015µs or so.
Leading me to believe ros polls the subscriptions each millisecond (and it ends in 010 because it takes 10µs from the start of the callback interrupt to executing `micros()`.)
Adding `delayMicroseconds(3);` in the callback makes the time always end in 013, so this confirms this.

So coming back to my question, is it possible to increase this polling rate?

JanStaschulat, 2021-05-21:
Could you add
```
Serial.println(micros()); 
```
in the `loop` in your real application? This would just exclude any polling/waiting in the sensor_driver_updates.

JTaets, 2021-05-25:
I might not be completely following with what you mean.
I did:
```
void loop() {
  if (nr_data<99){ // recieve datapoints
    RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1)));
    times[nr_data] = micros(); 
    nr_data+=1;  
  } else { 
    // code for sending times back to the computer
  }   
}
```

The result depends on the rate I send data and the amount of milliseconds given as timeout.
If the send rate is 100Hz and there is 1 ms timeout. I get periodically a difference of (in µs):
... 993 1000 1000 1000 1000 1000 1000 1000 1000 1007  ...
Which i expected, each 10th one is a little longer because then I receive data. The other ones are timeouts.

I confirmed my suspicion of the polling rate of 1kHz more by sending data at 800Hz.
Then i get as time passed between receiving the datapoints periodically (in µs):
... 2000, 1000, 1000, 1000 ...
(I.e. this is the timing in the callback again like in previous posts, not anymore in `loop()`)
Indeed 4 datapoints each 5ms (2000µs + 1000µs +1000µs +1000µs) is 800Hz
Notice again the multiples of 1ms i receive data at.

Absolute timings are (in µs):
3010, 5010, 6010, 7010, 8010

I'm wondering if this polling behaviour is expected of running ros2 on a Teensy. I would guess this would be done with interrupts?
Or could this be my computor which only sends data on the usb line each ms?
Altough `ros2 topic hz /control` gives me 800Hz, i suppose this measures the frequency of writing to the topic, not it sending data over the usb-line?

JTaets, 2021-05-26:
@pablogs9 Is it possible to reopen this issue?
I don't have the permissions to reopen it.

If you prefer, I could reformulate the problem in a new issue because this thread is quite long.

Kind regards