micro-ROS/micro_ros_arduino #278: micro ros agent

KohlhardtC, 2021-05-22:
## Issue template

- Hardware description: Teensy 4.1
- RTOS: arduinio baremetal 
- Version or commit hash: foxy

#### Steps to reproduce the issue

Run the micro ros agent either using Serial or UDP. Both of these generated the error:

`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200 -v 6`

`ros2 run micro_ros_agent micro_ros_agent udp4 -p 8080 -v 6`

I tested this with both the Serial example examples/micro-ros_publisher/micro-ros_publisher.ino and the [NativeEthernet implementation ](https://github.com/KohlhardtC/micro_ros_arduino) but no longer works. 
#### Expected behavior

Agent should run without errors

#### Actual behavior

Agent generates the following output including error:

```

chrisk@brain:~/code/micro-ros/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200 -v 6
[1621704360.515726] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1621704360.515868] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1621704361.519688] info     | Root.cpp           | create_client            | create                 | client_key: 0x49C73FD7, session_id: 0x81
[1621704361.519839] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1237794775, address: 0
[1621704361.520043] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1621704362.519398] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 56, data: 
0000: 81 80 00 00 01 05 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 00 00 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 00 00 00 00 00 00
[1621704362.520401] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621704362.522376] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621704362.526377] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621704362.530316] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49C73FD7, participant_id: 0x000(1)
[1621704362.530655] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1621704362.530825] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621704362.530977] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621704362.531062] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621704362.531033] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621704362.531143] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621704362.531260] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 100, data: 
0000: 81 80 01 00 01 05 5A 00 00 0B 00 02 02 03 00 00 4C 00 00 00 24 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F 70 75 62 6C 69 73 68 65 72 00 00 01 01 00
0040: 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00
0060: 00 01 00 00
[1621704362.531550] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x49C73FD7, topic_id: 0x000(2), participant_id: 0x000(1)
[1621704362.531753] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1621704362.531836] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1621704362.532017] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1621704362.533334] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 24, data: 
0000: 81 80 02 00 01 05 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1621704362.533398] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 02 00 02 00 80
[1621704362.533535] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x49C73FD7, publisher_id: 0x000(3), participant_id: 0x000(1)
[1621704362.533679] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1621704362.533722] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1621704362.533743] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1621704362.533990] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1621704362.535225] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 36, data: 
0000: 81 80 03 00 01 05 19 00 00 0D 00 05 05 03 00 00 0B 00 00 00 00 02 01 00 03 00 00 00 00 00 00 00
0020: 03 00 00 00
2021-05-22 10:26:02.535 [RTPS_PARTICIPANT Error] Couldn't create persistence service for transient/persistent writer -> Function createWriter
2021-05-22 10:26:02.535 [DATA_WRITER Error] Problem creating associated Writer -> Function enable
[*** LOG ERROR #0001 ***] [2021-05-22 10:26:02] [] {argument index out of range}
[1621704362.535435] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 80 00
[1621704362.535443] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1621704362.535592] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
[1621704362.535614] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1621704362.536682] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data:
```

#### Additional information

I assume due to changes that took place when I rebased my branch, therefore I assume something changed in the last few weeks that is causing this issue.

Acuadros95, 2021-05-24:
Hello @KohlhardtC, could you give it a try with the dockerized agent?

`docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

KohlhardtC, 2021-05-24:
Sorry, I forgot to mention that I did try the dockerized agent and got the same error.

Acuadros95, 2021-05-25:
I just tested your repository on a Teensy 4.1 board and the [micro-ros_publisher.ino](https://github.com/KohlhardtC/micro_ros_arduino/blob/foxy/examples/micro-ros_publisher/micro-ros_publisher.ino) example is working fine. Please make sure you are using the latest micro-ROS arduino release: [v1.0.0](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v1.0.0).

Also for the dockerized agent pull the last version in case you are missing any updates: 
`docker pull microros/micro-ros-agent:foxy`

And to start it:
`docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

I also tested with a local agent, following [this](https://micro.ros.org/docs/tutorials/core/first_application_linux/) tutorial to build the agent and executing with: 
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200 -v6`

Please check this cases. if it still doesnt work I will need more info, did you make any modifications on your ROS2 installation or in the micro_ros_arduino library?

KohlhardtC, 2021-05-25:
Thank you for your help Antonio. Getting the updated dockerized agent solved the problem for both the serial and UDP connections.

closed 2021-05-25 by KohlhardtC