micro-ROS/micro_ros_arduino #285: Teensy 4.0 not sending messages / agent not receiving

beauhobba, 2021-05-25:
## Issue template

- Hardware description: Teensy 4.0
- Version or commit hash: Foxy

#### Steps to reproduce the issue
Added Micro Ros to Sketch -> Include library -> Add .ZIP Library...
Patched Teensyduino (not SAM) 
Uploaded micro-ros.publisher.io to the Teensy
docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
Sourced ROS2
/ros2 topic list

#### Expected behavior
The debug command window should receive messages
Rostopic should be updated with the new node

#### Actual behavior
No new node 
No messages being received by the agent

Here is a log of the console:
`beau@beau-VirtualBox:~$ sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1621970780.817450] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1621970781.820991] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1621970782.823024] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1621970783.851576] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 14
[1621970783.852149] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1621970784.609784] info     | Root.cpp           | create_client            | create                 | client_key: 0x49C73FD7, session_id: 0x81
[1621970784.609883] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1237794775, address: 0
[1621970784.610233] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1621970785.609696] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 112, data: 
0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1621970785.610493] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.612410] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
2021-05-25 19:26:25.612 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[1621970785.632203] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.638049] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.643778] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49C73FD7, participant_id: 0x000(1)
[1621970785.660023] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.695054] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1621970785.703735] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.704089] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.706546] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.706956] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.697675] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.725670] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.753196] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.761927] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970785.781254] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1621970785.913994] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
[... 17 lines cut ...]
[1621970787.730532] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970787.930759] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970788.149264] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970788.349716] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970788.550056] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970788.751023] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970788.951144] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970789.151266] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1621970789.362389] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
`

#### Additional information

- I am using a UBUNTU virtual machine with the Teensy added as a USB device through the oracle UI. 
- I have not used any ROS_DOMAIN_ID and I have checked my environmental variables and this has not been set. 
- The same thing happens when I run an agent from source code.

pablogs9, 2021-05-26:
In the copypasted output you are not using `-v /dev/shm:/dev/shm` in the agent. In any case, you are having a problem with the internals of the micro-ROS agent.

Please try to add `-v /dev/shm:/dev/shm` and if this not work I'll provide instructions for having a working micro-ROS Agent.

beauhobba, 2021-05-26:
After using:
`sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`
and
`sudo docker run -it --rm -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`
Still nothing unfortunately (see error log below). However I have noticed that when I do stop the agent  from running the teensy goes into error mode synchronously. 

`beau@beau-VirtualBox:~$ sudo docker run -it --rm -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1622030837.883957] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1622030837.905880] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1622030837.911808] info     | Root.cpp           | create_client            | create                 | client_key: 0x49C73FD7, session_id: 0x81
[1622030837.911847] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1237794775, address: 0
[1622030837.911943] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1237794775, address: 0
[1622030837.911954] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1237794775, address: 0
[1622030837.911959] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1237794775, address: 0
[1622030837.912198] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622030837.912264] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622030837.912319] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622030837.912370] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622030837.916642] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 112, data: 
0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1622030837.930599] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030837.942988] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030837.958358] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030837.972077] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030837.996570] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030837.999801] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030838.018756] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49C73FD7, participant_id: 0x000(1)
[1622030838.039198] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1622030838.039905] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622030838.040361] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622030838.040811] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622030838.041194] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622030838.041670] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622030838.042143] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
[... cut ...]
[1622030851.802811] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030852.004509] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622030852.205264] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C73FD7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
`

pablogs9, 2021-05-26:
Could you please update your library to the latest commit of `foxy` branch in this repo and test again?

beauhobba, 2021-05-26:
Ok so I used the latest clone of https://github.com/micro-ROS/micro_ros_arduino and replaced my current arduino library with it. Now I am sends and receives from the Teensy.
I use 
`sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`
However, I am still not receiving any new ros2 topics in my ros2 topic list. (I have sourced my ROS2)
I should not have a ROS2 Domain set (I checked my environment variables) 

`beau@beau-VirtualBox:~$ sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1622033338.732292] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1622033338.733078] info     | Root.cpp           | create_client            | create                 | client_key: 0x49A56EE4, session_id: 0x81
[1622033338.745017] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1235578596, address: 0
[1622033338.749580] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1235578596, address: 0
[1622033338.750744] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622033338.733044] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1622033338.750819] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1622033338.756183] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 56, data: 
0000: 81 80 00 00 01 05 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 00 00 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 00 00 00 00 00 00
[1622033338.775646] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622033338.775668] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622033338.775674] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622033338.776835] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622033338.777186] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49A56EE4, participant_id: 0x000(1)
[1622033338.793772] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1622033338.797639] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1622033338.799717] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.807635] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.815821] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.819327] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.819888] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.831210] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1622033338.831260] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 100, data: 
0000: 81 80 01 00 01 05 5A 00 00 0B 00 02 02 03 00 00 4C 00 00 00 24 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F 70 75 62 6C 69 73 68 65 72 00 00 01 01 00
0040: 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00
0060: 00 01 00 00
[1622033338.831351] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x49A56EE4, topic_id: 0x000(2), participant_id: 0x000(1)
[1622033338.842450] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[... cut ...]
0000: 81 00 00 00 0B 01 05 00 02 00 02 00 80
[1622033338.916877] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1622033338.916884] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49A56EE4, len: 36, data: 
0000: 81 80 03 00 01 05 19 00 00 0D 00 05 05 03 00 00 0B 00 00 00 00 02 01 00 03 00 00 00 00 00 00 00
0020: 03 00 00 00
`

hobbabeau, 2021-05-26:
I also ran:
`docker pull microros/micro-ros-agent:foxy`
The agent is up to date

alansrobotlab, 2021-05-29:
fwiw I was seeing the same thing on both teensy40 and teensy41 for the past few days that I've been dinking with it.  It looks like the auto-update from 2021-05-28 #291 resolved the issue?

After grabbing the updated micro ros arduino files everythings talking again.  I was using the standard micro ros instructions to create the agent.  Hadn't tried your docker agent until this morning but am seeing the microros topics and messages there as well now.

Thanks!

Alan

hobbabeau, 2021-05-30:
Hello @alansrobotlab  I have updated to the newest patch, and still no ros2 topics from either docker or the standard :( 
Just to clarify I have installed everything correct I performed: 

Added Micro Ros to Sketch -> Include library -> Add .ZIP Library...
Patched Teensyduino (not SAM)
Uploaded micro-ros.publisher.io to the Teensy
sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6

docker run -ti --rm --net=host ros:foxy ros2 topic list

Am I missing anything crucial? Am I doing anything stupid?

pablogs9, 2021-05-31:
Can you share the output of the agent?
Also, try to run your ROS2 instance like: ` docker run -ti --rm -v /dev/shm:/dev/shm --privileged --net=host ros:foxy ros2 topic list`

alexanderbsd, 2021-08-06:
Tested on my Teensy 4.0. Works well out-of-the-box.

closed 2021-08-09 by pablogs9