micro-ROS/micro_ros_arduino #296: How to use array data?

JuFengWu, 2021-05-30:
Hello:

I want to send a array data from my Arduino to my PC.
I tried this code 

`void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &outMsg, NULL));
  }
  int a[10]={0};
  outMsg.data = a;
}`

but it showed up 

`error: no match for 'operator=' (operand types are 'rosidl_runtime_c__int32__Sequence' and 'int [10]')
   outMsg.data = a;`

It seems the type `int a[10]` is not correct, so what kinds of type should I use ?
Thank you!

pablogs9, 2021-05-31:
Could you share the whole code so we can check which is your setup?

JuFengWu, 2021-05-31:
`
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32_multi_array.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32MultiArray outMsg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  int a[10]={0};
  outMsg.data = a;
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &outMsg, NULL));
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "topic"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
[... 41 lines cut ...]
   typedef struct rosidl_runtime_c__ ## STRUCT_NAME ## __Sequence \
                  ^
/home/leowu/Arduino/libraries/micro_ros_arduino-1.0.0/src/rosidl_runtime_c/primitives_sequence.h:43:1: note: in expansion of macro 'ROSIDL_RUNTIME_C__PRIMITIVE_SEQUENCE'
 ROSIDL_RUNTIME_C__PRIMITIVE_SEQUENCE(int32, int32_t)
 ^
/home/leowu/Arduino/libraries/micro_ros_arduino-1.0.0/src/rosidl_runtime_c/primitives_sequence.h:23:18: note:   no known conversion for argument 1 from 'int [10]' to 'rosidl_runtime_c__int32__Sequence&&'
   typedef struct rosidl_runtime_c__ ## STRUCT_NAME ## __Sequence \
                  ^
/home/leowu/Arduino/libraries/micro_ros_arduino-1.0.0/src/rosidl_runtime_c/primitives_sequence.h:43:1: note: in expansion of macro 'ROSIDL_RUNTIME_C__PRIMITIVE_SEQUENCE'
 ROSIDL_RUNTIME_C__PRIMITIVE_SEQUENCE(int32, int32_t)
 ^
/home/leowu/ros2/micro_ros2_arduino_test/arduino_code/micro-ros_publisher/micro-ros_publisher.ino: In function 'void setup()':
/home/leowu/Arduino/libraries/micro_ros_arduino-1.0.0/src/rosidl_runtime_c/message_type_support_struct.h:78:59: error: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32' was not declared in this scope
     rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
                                                           ^
/home/leowu/ros2/micro_ros2_arduino_test/arduino_code/micro-ros_publisher/micro-ros_publisher.ino:21:43: note: in definition of macro 'RCCHECK'
 #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                           ^
/home/leowu/ros2/micro_ros2_arduino_test/arduino_code/micro-ros_publisher/micro-ros_publisher.ino:62:5: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
     ^
exit status 1
no match for 'operator=' (operand types are 'rosidl_runtime_c__int32__Sequence' and 'int [10]')

`

pablogs9, 2021-06-01:
I recommend you to use the commit of micro-ROS for Arduino in the branches `main` or `galactic`. Once you have pull one of those versions, this code should work for you:

<details><summary>Code for main or galactic</summary>

```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <micro_ros_utilities/type_utilities.h>
#include <micro_ros_utilities/string_utilities.h>

#include <std_msgs/msg/int32_multi_array.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32MultiArray outMsg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);

  for (size_t i = 0; i < outMsg.data.capacity; i++)
  {
    outMsg.data.data[i] = i*10;
  }
  outMsg.data.size = outMsg.data.capacity;

  RCSOFTCHECK(rcl_publish(&publisher, &outMsg, NULL));
}

void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32MultiArray),
    "topic"));

  // create timer,
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(1000),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  // init message memory
  micro_ros_utilities_memory_conf_t conf = {0};
  conf.max_ros2_type_sequence_capacity = 50;
  conf.max_basic_type_sequence_capacity = 50;
  if(!micro_ros_utilities_create_message_memory(
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32MultiArray),
        &outMsg,
        conf))
  {
    error_loop();
  }
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
</details>

pablogs9, 2021-06-01:
Please close if solved.

pablogs9, 2021-06-01:
I'm closing since the provided code works. Please reopen if you have more questions or problems.

Thanks!

closed 2021-06-01 by pablogs9