micro-ROS/micro_ros_arduino #315: [old gcc version] undefined reference to `__ctype_ptr__'

wuhanstudio, 2021-06-05:
## Issue template

- Hardware description: STM32F103
- RTOS: RT-Thread
- Installation type: libmicroros.a
- Version or commit hash: foxy

#### Steps to reproduce the issue

```
$ sudo apt install gcc-arm-none-eabi
$ arm-none-eabi-nm -C --defined-only -g cortex-m3/libmicroros.a | grep __ctype_ptr__
$ # Indeed, there is no output of __ctype_ptr__
```

#### Expected behavior

#### Actual behavior

#### Additional information

I notice that the gcc version being used to generate libmicroros.a is 5.4, which is a little outdated.

This could cause errors if someone uses a newer version of gcc.

Acuadros95, 2021-06-09:
Hello,

As you can see on the [lib generation script,](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/library_generation.sh) we use the same compiler version as the Arduino IDE for each board.

Your board is not currently supported by the library generator, but you can modify the script to use your own gcc version:

1. Clone this repo and modify this line to: `export TOOLCHAIN_PREFIX=/toolchain/bin/arm-none-eabi-`
https://github.com/micro-ROS/micro_ros_arduino/blob/94cb206d03cfb19aa3f1404f0d9d9cbd6b7bac6d/extras/library_generation/library_generation.sh#L88

2. Set the local path of the desired compiler:
`export GCC_PATH=/local_path/gcc-arm-none-eabi-10-2020-q4-major` (Use your own local path)

3. Call the docker generator from the repo folder:
`docker run -it --rm -v ${GCC_PATH}:/toolchain -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m3`

3. The generated library should be on `src/cortex-m3/libmicroros.a`:
```
~/workspace/micro_ros_arduino$ strings src/cortex-m3/libmicroros.a | grep "GCC: ("
$ GCC: (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
```

wuhanstudio, 2021-06-09:
Thanks for your help. Now both ROS1 (rosserial) and ROS2 (micro_ros) works on RT-Thread (RTOS).

ROS1 (rosserial):

https://user-images.githubusercontent.com/15157070/121332945-3c1b1680-c910-11eb-80b8-5488b7d7793e.mp4

ROS2 (micro_ros):

https://user-images.githubusercontent.com/15157070/121332938-3b828000-c910-11eb-83c4-ce814bddc7a4.mp4

Acuadros95, 2021-06-09:
Great!

I am closing the issue, feel free to reopen if needed

closed 2021-06-09 by Acuadros95