micro-ROS/micro_ros_arduino #316: Running micro_ros_ardiuno on embedded seeeduino zero?

Bart-van-Ingen, 2021-06-09:
A seeeduino zero (ATSAMD21G18) is embedded in the single board computer that I am working on [(link)](https://wiki.seeedstudio.com/ODYSSEY-X86J4105/#using-the-arduino-core-atsamd21g18-in-linux-os). I have been able to compile the micro-ros_publisher example and upload it to the board, using the arduino zero profile. This was done without patching since the example in the micro_ros_arduino repo is for SAM boards, while this is a SAMD board. Is this a correct assumption?

Running the docker command to start micro-ROS Agent seems to be working.

Unfortunately I am not able to see the topics on which the board is supposed to be running when using ```ros2 topic list```.

Although the seeeduino zero is clearly based of the arduino zero, is it possible to run micro-ros on it?

pablogs9, 2021-06-09:
How are you running the micro-ROS Agent?

Bart-van-Ingen, 2021-06-09:
using the docker command:
```# Serial micro-ROS Agent
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyAM0 -v6
```

pablogs9, 2021-06-09:
Try with:

```
docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyAM0 -v6
```

and let me know if `ros2 topic list` works

Bart-van-Ingen, 2021-06-09:
running your command unfortunately still leaves ```ros2 topic list``` not showing the desired topic.

pablogs9, 2021-06-09:
Paste here the output of the agent, remember to use the `-v6` flag.

Bart-van-Ingen, 2021-06-09:
```
adi@ADI-ODYSSEY-1:~/.arduino15/packages/arduino/hardware/samd/1.8.11$ sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6

[1623232538.396008] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1623232538.396457] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

```

pablogs9, 2021-06-09:
Your micro-ROS client is not connecting to the micro-ROS Agent, probably the micro-ROS client code is not working properly or not using the proper transport.

Bart-van-Ingen, 2021-06-09:
ah ok. What am I supposed to see if it does work?
Is the fact that it is not an arduino (although it has the same chipset) zero the likely reason it is not working? Should I patch the arduino ide for SAMD even though the example seems to indicate SAM ?

pablogs9, 2021-06-09:
If the board is running the micro-ROS client correctly using the serial transport, it should use the default serial port to communicate with the agent. Every received message in the agent will be print if `-v6` is used.

Bart-van-Ingen, 2021-06-09:
I have found out that uploading through the arduino IDE and then immediately afterwards running the original docker command or the one that you suggested in this thread got the micro ros agent to output as seen below:
```
sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1623243880.345957] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1623243880.346432] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1623243880.983473] info     | Root.cpp           | create_client            | create                 | client_key: 0x7D248957, session_id: 0x81
[1623243880.983630] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2099546455, address: 0
[1623243880.983841] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1623243881.983147] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 112, data: 
0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1623243881.996236] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x7D248957, participant_id: 0x000(1)
[1623243881.996376] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1623243881.996475] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1623243881.996923] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1623243881.997958] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 148, data: 
0000: 81 80 01 00 01 05 89 00 00 0B 00 02 02 02 00 00 7B 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F
0040: 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73
0060: 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E
0080: 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00 00
[1623243881.998510] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x7D248957, topic_id: 0x000(2), participant_id: 0x000(1)
[1623243881.998573] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1623243881.998671] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1623243881.999392] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1623243881.999590] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 03 03 02 00 00 01 00 00 00 00 00 01 00
[1623243881.999768] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x7D248957, publisher_id: 0x000(3), participant_id: 0x000(1)
[1623243882.000137] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1623243882.000255] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1623243882.000634] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x7D248957, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
```
I am able to see the topic now when I use ``ros2 topic list```!
However I am still not able to see what is being published on the topic when use ```ros2 topic echo  /micro_ros_arduino_node_publisher``` nothing is printing.
Another observation that if i run the exact same docker command after already having use it once, it no longer works ...

pablogs9, 2021-06-09:
You need to reset the board to init micro-ROS client.

In order to check if something is happening this the executor, try to remove the timer in your app code an use the [`rcl_publish`](https://github.com/micro-ROS/micro_ros_arduino/blob/7436183aa7fdf03d3481a4002e326d62608b8592/examples/micro-ros_publisher/micro-ros_publisher.ino#L36) inside the [`loop()`](https://github.com/micro-ROS/micro_ros_arduino/blob/7436183aa7fdf03d3481a4002e326d62608b8592/examples/micro-ros_publisher/micro-ros_publisher.ino#L79)

pablogs9, 2021-06-09:
Also, you are using an old version of the micro-ROS library. Please pull the latest commit of the library.

Bart-van-Ingen, 2021-06-09:
I am running the foxy release of ros2 on my computer. Will the newest version of micro ros still be compatible?

pablogs9, 2021-06-09:
The newest commit of this repo in the `foxy` branch is the newest micro-ROS library for Arduino platforms.

Bart-van-Ingen, 2021-06-09:
Alright! I will check it out! You say that I have to reset the board to init micro-ROS client, is there also a timeout involved. I say this because sometimes i will have uploaded the firmware to the board and waited a little with starting the agent and then connection is not made. Only when i reset the board do I get correct connection between the two.

Bart-van-Ingen, 2021-06-09:
I have now tried with the newest commit on the foxy branch by downloading it as zip and following the steps in the readme . I am able to compile it but when using the arduino zero as target I am unable to connect using the agent. Only when i use the seeeduino as target does it work and the debug information is being print to terminal, however i still am not able to see the topics/nodes.

pablogs9, 2021-06-10:
Does it stop there or it continues printing lines with `[** <<DDS>> **]`?

Bart-van-Ingen, 2021-06-10:
it continues printing ```[** <<DDS>> **]```

pablogs9, 2021-06-10:
And if you run `ros2 topic list` in your local ROS 2 installation, it does not show your published topic?

Bart-van-Ingen, 2021-06-10:
that is right, ros2 topic list only shows ```/parameter_events``` and ```/rosout```

pablogs9, 2021-06-10:
`sudo` should not be necessary if your user has permission for the serial port.

Where are you running `ros2 topic list`, in your local machine or in a docker?

Bart-van-Ingen, 2021-06-10:
on my local machine

pablogs9, 2021-06-10:
Try to pull the latest agent docker:

```bash
docker pull microros/micro-ros-agent:foxy
```

Let me know if this solves the issue

Bart-van-Ingen, 2021-06-10:
no difference :(

Bart-van-Ingen, 2021-06-10:
an interesting development: 
I have come to the conclusion that when i start the docker image a problem is caused in the ros2 daemon. This would also explain why ```ros2 topic list``` only shows ```/parameter_events``` and ```/rosout```. After shutting down the docker image with ctrl+c and running ros nodes that I have authored I get the same result of only showing ``/parameter_events``` and ```/rosout```. Only once I run ```ros2 daemon stop``` and ```ros2 daemon start``` am i able to get a functioning ```ros2 topic list``` again

pablogs9, 2021-06-10:
Can we mark this as solved since the micro-ROS part is working and it is matter of ROS 2 daemons in dockers?

Bart-van-Ingen, 2021-06-10:
sure, if you think it is appropriate! Do you think I should raise this problem with the ROS2 docker image repo?

closed 2021-06-11 by pablogs9