micro-ROS/micro_ros_arduino #384: Adding both a timer and subscription to the executor seems to cause error_loop() to run

samiamlabs, 2021-07-20:
## Issue template

- Hardware description:  Teensy 3.2
- Installation type: Followed instructions from https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/ and modified for galactic
- Version or commit hash: galactic

#### Steps to reproduce the issue
Hi!

I just started experimenting with micro-ros today and ran into an issue with trying to add both a timer and subscriber to the executor at the same time.

I uploaded this code to a Teensy:

```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
rcl_subscription_t subscriber;

std_msgs__msg__Int32 cmd_msg;
std_msgs__msg__Int32 state_msg;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop() {
  while (1) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void * msgin)
{
  const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
  state_msg.data = msg->data;
  RCSOFTCHECK(rcl_publish(&publisher, &state_msg, NULL));
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &state_msg, NULL));
  }
}

void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "camera_servo", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "camera_servo/state_angular_pos"));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
            &subscriber,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "camera_servo/cmd_angular_pos"));

  // create timer,
  const unsigned int timer_timeout = 100;
  RCCHECK(rclc_timer_init_default(
            &timer,
            &support,
            RCL_MS_TO_NS(timer_timeout),
            timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &cmd_msg, &subscription_callback, ON_NEW_DATA));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  state_msg.data = 0;
}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
#### Expected behavior
It is very possible that I made some very basic mistake here, to be honest... but I expected this code to republish the last message it got from the subscriber at 10Hz (and publish one extra time immediately in the incoming message callback).

#### Actual behavior
Instead, it entered the error loop immediately after I ran `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`.

#### Additional information
Both the publisher and subscriber examples work fine for me, and I basically just tried to run them at the same time.
If I comment out one of the following lines:
```
RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &cmd_msg, &subscription_callback, ON_NEW_DATA));
RCCHECK(rclc_executor_add_timer(&executor, &timer));
```
Everything works as expected though... so I'm a bit confused... does the executor not support adding both a timer and a subscription?
Or am I making some type of beginner mistake here? Really cant tell :P

samiamlabs, 2021-07-20:
I get the same behaviour if trying to add 2 subscriptions to the same executor

pablogs9, 2021-07-21:
You are creating a executor with only one handle: `RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));`
Try with `RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));`

closed 2021-07-21 by pablogs9

samiamlabs, 2021-07-21:
Ah, that totally makes sense but is not something I would have expected to have to do based on my experience with other ros2 client libraries.

I think adding something like a multiple_subscribers example with a comment about multiple handles on the line that creates the executor would do a lot to help out other beginners like me :)

pablogs9, 2021-07-21:
We will take it into account, thanks!
CC: @JanStaschulat

samiamlabs, 2021-07-21:
On the subject of possible pain points for beginners, it looked to me at first as if `msg` in this macro:
`ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32)`
was referring to:
`std_msgs__msg__Int32 msg;`
I just quickly changed all occurrences of `msg` to `cmd_msg` and ended up with a pretty hard to parse build error because of it.

Other than that the experience coming into this has been pretty smooth, nice work!

samiamlabs, 2021-07-21:
Using anything else except `msg` for `std_msgs__msg__Int32` would obviously help with this, but `msg` is kind of the standard naming convention for message instances. That's a good reason to keep things like they are now also.

I don't know... just thought I would mention it anyway :)

pablogs9, 2021-07-21:
> On the subject of possible pain points for beginners, it looked to me at first as if msg in this macro:
ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32)
was referring to:
std_msgs__msg__Int32 msg;
I just quickly changed all occurrences of msg to cmd_msg and ended up with a pretty hard to parse build error because of it.

> Other than that the experience coming into this has been pretty smooth, nice work!

Can you share code for that? The macro and the variable name are not related... Can you explain this a little bit, please?

samiamlabs, 2021-07-21:
If the macro looked like:
`ROSIDL_GET_MSG_TYPE_SUPPORT("std_msgs", "msg", "Int32")`
it would have been a lot easier for me to understand what it was doing at first glance
If I see names without quotes, I tend to expect that to refer to a variable