micro-ROS/micro_ros_arduino #391: ls: cannot access '*.o': No such file or directory after building the pre-compiled library

grassjelly, 2021-07-21:
## Issue template

- Hardware description: Teensy 3.6
- RTOS: Ubuntu 20.04
- Installation type: micro_ros_setup
- Version or commit hash: galactic branch

#### Steps to reproduce the issue

    cd micro_ros_arduino 
    docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:rolling

#### Expected behavior
No errors after building the library

#### Actual behavior
```
Summary: 64 packages finished [5min 4s]
  40 packages had stderr output: action_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs geometry_msgs libyaml_vendor lifecycle_msgs lino_msgs micro_ros_msgs micro_ros_utilities microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_interface rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs stereo_msgs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
ls: cannot access '*.o': No such file or directory
```

#### Additional information
I'm trying to build my own version of this library as I want to include a custom message type. I copied the new custom_msg folder in extras/library_generation/extra_packages and ran the docker command to build. I encountered this error:

`ls: cannot access '*.o': No such file or directory
`

with 40 packages having stderr output.

pablogs9, 2021-07-22:
Despite these messages, your library should be being built, isn't it? 

The `N packages had stderr output` is the output of the [colcon build](https://github.com/micro-ROS/micro_ros_setup/blob/ca633e280eccc9c6b62420b43cca9a5dd3c6b2dc/config/generate_lib/generic/build.sh#L26). Those stderr messages can be a warning and depends on your toolchain and compiler configuration. There is no fatal error, so the colon build is ok.

The `ls: cannot access '*.o': No such file or directory` cames from [here](https://github.com/micro-ROS/micro_ros_setup/blob/ca633e280eccc9c6b62420b43cca9a5dd3c6b2dc/config/generate_lib/generic/build.sh#L49). In this point we uncompress every static library generated by the micro-ROS build process and recompress them into a single file libmicroros.a. The point is that some compilers generate .o files and other *.obj files, so we look for both of them. [This PR](https://github.com/micro-ROS/micro_ros_setup/pull/360) should hide the warning.

Let me know if this solves your questions.

grassjelly, 2021-07-22:
Got it! Thanks! Managed to build and upload my firmware with the newly created custom messages. Thanks for the clarification.

closed 2021-07-22 by grassjelly