micro-ROS/micro_ros_arduino #400: Not getting subscription callbacks anymore after I add a timer

samiamlabs, 2021-07-27:
Hi!

Is there some official forum for user help connected to this project?

I'm having a bunch of beginner issues using micro-ros on Arduino that are tricky to find answers to online.
Is this the right place maybe? https://answers.ros.org/questions/scope:all/sort:activity-desc/tags:micro-ROS

I would be willing to help with answering questions from other new users of this project if there was some reasonably clean forum for it. Ros Answers for micro-ros seems kind of all over the place in terms of what is discussed there.

Gonna ask another beginner question here even though I don't feel it is the right place.
Don't hesitate to tell me to stop if you don't want this kind of questions here :)

- Hardware description: openCR
- Version or commit hash: galactic

#### Steps to reproduce the issue
I have been making a simple node for controlling a servo and linear motor via ros-topics and am having some trouble getting timers, subscribers and reconnection to work at the same time. When I add the timer to the executor, the subscription callbacks stops working. Here is the code:

```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
rcl_subscription_t servo_subscriber;
rcl_subscription_t camera_elevator_subscriber;

std_msgs__msg__Int32 servo_cmd_msg;
std_msgs__msg__Int32 camera_elevator_cmd_msg;

std_msgs__msg__Int32 servo_state_msg;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

bool micro_ros_init_successful;

#include <Servo.h>

Servo camera_servo;

int pos = 90;    // variable to store the servo position
int vel = 0;

#define LED_PIN 13
#define SERVO_PIN 5
#define ELEVATOR_ENABLE_PIN 50
#define ELEVATOR_DIRECTION_PIN 51

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){return false;}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  (void) last_call_time;
  if (timer != NULL) {
    camera_servo.write(pos);
    rcl_publish(&publisher, &servo_state_msg, NULL);
  }
}

void servo_subscription_callback(const void * msgin)
{
  const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
  pos = msg->data;
  servo_state_msg.data = pos;
  RCSOFTCHECK(rcl_publish(&publisher, &servo_state_msg, NULL));
  camera_servo.write(pos);
}

void camera_elevator_subscription_callback(const void * msgin)
{
  const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
  servo_state_msg.data = msg->data;
  if (msg->data == 0)
  {
    // stop
    digitalWrite(ELEVATOR_ENABLE_PIN, LOW);
  } else if (msg->data > 0)
  {
    // move up
    digitalWrite(ELEVATOR_ENABLE_PIN, HIGH);
    digitalWrite(ELEVATOR_DIRECTION_PIN, HIGH);

  } else {
    // move down
    digitalWrite(ELEVATOR_ENABLE_PIN, HIGH);
    digitalWrite(ELEVATOR_DIRECTION_PIN, LOW);
  }
}

bool create_entities()
{  
  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "teensy_hardware_interface", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "camera_servo/state_angular_pos"));

  // create servo_subscriber
  RCCHECK(rclc_subscription_init_default(
            &servo_subscriber,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "camera_servo/cmd_angular_pos"));

  // create camera elevator_subscriber
  RCCHECK(rclc_subscription_init_default(
            &camera_elevator_subscriber,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "camera_elevator/cmd_vel"));

  // create timer,
  const unsigned int timer_timeout = 100;
  RCCHECK(rclc_timer_init_default(
            &timer,
            &support,
            RCL_MS_TO_NS(timer_timeout),
            timer_callback));

  // create executor
  executor = rclc_executor_get_zero_initialized_executor();
  RCCHECK(rclc_executor_init(&executor, &support.context, 3, &allocator)); // 3 handles
  RCCHECK(rclc_executor_add_subscription(&executor, &servo_subscriber, &servo_cmd_msg, &servo_subscription_callback, ON_NEW_DATA));
  RCCHECK(rclc_executor_add_subscription(&executor, &camera_elevator_subscriber, &camera_elevator_cmd_msg, &camera_elevator_subscription_callback, ON_NEW_DATA));
  
  //Having issues getting subscriptions to work at the same time as timer
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  micro_ros_init_successful = true;
}

void destroy_entities()
{
  rcl_publisher_fini(&publisher, &node);
  rcl_subscription_fini(&servo_subscriber, &node);
  rcl_subscription_fini(&camera_elevator_subscriber, &node);
  rcl_node_fini(&node);
  rcl_timer_fini(&timer);
  rclc_executor_fini(&executor);
  rclc_support_fini(&support);

  micro_ros_init_successful = false;
}

void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  pinMode(ELEVATOR_ENABLE_PIN, OUTPUT);
  pinMode(ELEVATOR_DIRECTION_PIN, OUTPUT);
  
  digitalWrite(LED_PIN, HIGH);
  digitalWrite(ELEVATOR_ENABLE_PIN, LOW);
  digitalWrite(ELEVATOR_DIRECTION_PIN, LOW);

  camera_servo.attach(SERVO_PIN); // use pin 9
 
  micro_ros_init_successful = false;
  servo_state_msg.data = 0;
}

void loop() {
  
  uint32_t delay = 100000;
  if (RMW_RET_OK == rmw_uros_ping_agent(50, 2))
  {
    delay = 100000;
    if (!micro_ros_init_successful) {
      create_entities();
    } else {
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
    }
  } 
  else if (micro_ros_init_successful)
  {
    destroy_entities();
  }

  digitalWrite(LED_PIN, !digitalRead(LED_PIN));
  delayMicroseconds(delay); 
}
```

#### Expected behavior
I did not expect that adding a 10Hz timer would affect the processing of subscription callbacks.

#### Actual behavior
The timer callback works when I add it but subscription callbacks stop working.

pablogs9, 2021-07-28:
Does it fails if you avoid the logic of checking in the loop the agent? Can you try something like:

```c
void setup(){
  create_entities();
}

void loop(){
  rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
}
...

samiamlabs, 2021-07-31:
Hi!

I could get by without the timer for the project I'm working on so took me a while to get back to troubleshooting this... sorry about that.

It does actually seem to work without the checking, as you suggested. Any idea what the problem could be?

From what I have experienced so far, the checking logic to make sure it reconnects seems pretty much necessary if you are using this as one of many things controlled by a robot.

grassjelly, 2021-08-23:
hey @samiamlabs. I think your objective is pretty much the same as mine. I managed to do it by polling the agent in a lesser rate.
```

void loop() 
{
    static unsigned long long prev_connect_test_time;

    // check if the agent got disconnected at 10Hz
    if(rolling_micros() - prev_connect_test_time > 100000)
    {
        prev_connect_test_time = rolling_micros();
        // check if the agent is connected
        if(RMW_RET_OK == rmw_uros_ping_agent(50, 2))
        {
            // reconnect if agent got disconnected or haven't at all
            if (!micro_ros_init_successful) 
            {
                createEntities();
            } 
        } 
        else if(micro_ros_init_successful)
        {
            // stop the robot when the agent got disconnected
            fullStop();
            // clean up micro-ROS components
            destroyEntities();
        }
    }
    
    if(micro_ros_init_successful)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
    }
}

```

Replacing rolling_micros() with just micros() also works.

pablogs9, 2022-01-28:
Any update on this?

Acuadros95, 2022-07-12:
Closing due to inactivity, reopen if needed.

closed 2022-07-12 by Acuadros95