micro-ROS/micro_ros_arduino #407: Example micro-ros_publisher: How to publish new topic

irsyadtc, 2021-08-02:
I'm trying to add a new Empty topic into the micro-ros_publisher example but still haven't grasp how does the code works.

Board: OpenCR
PC: Ubuntu 20.0
Arduino IDE: ver 1.8.13

Here is the sketch which I tried to modify:

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/empty.h> //

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg; //

rcl_publisher_t power_pub;
std_msgs__msg__Empty power_msg; //

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop() {
  while (1) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
    RCSOFTCHECK(rcl_publish(&power_pub, &power_msg, NULL)); //
  }
}

void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "micro_ros_arduino_node_publisher")); //(parent_folder, sub_folder, msg_type), topic_name
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, power_msg, Empty),
            "power_on")); //

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
            &timer,
            &support,
            RCL_MS_TO_NS(timer_timeout),
            timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 1;   //parameter data of Int32 msg
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
Compilation gives out this error:

```
Arduino: 1.8.13 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

/home/irsyad/Arduino/save/micro_ros_test/publisher/micro-ros_publisher_empty_210730/micro-ros_publisher_empty_210730.ino: In function 'void setup()':
/home/irsyad/Arduino/libraries/micro_ros_arduino/src/rosidl_runtime_c/message_type_support_struct.h:78:59: error: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__power_msg__Empty' was not declared in this scope
     rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
                                                           ^
/home/irsyad/Arduino/save/micro_ros_test/publisher/micro-ros_publisher_empty_210730/micro-ros_publisher_empty_210730.ino:26:43: note: in definition of macro 'RCCHECK'
 #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                           ^
/home/irsyad/Arduino/save/micro_ros_test/publisher/micro-ros_publisher_empty_210730/micro-ros_publisher_empty_210730.ino:72:13: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
             ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, power_msg, Empty),
             ^
exit status 1
Error compiling for board OpenCR Board.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Hopefully someone can help me out. Thank you

Acuadros95, 2021-08-02:
Hello @irsyadtc, you made an understandable mistake on this line:
`ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, power_msg, Empty),`

This function expects the message type path as arguments, that is: `<std_msgs/msg/empty.h> -> std_msgs, msg, Empty`. 
Try with: `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Empty),`

irsyadtc, 2021-08-02:
Thank you for pointing that out.

So far this is my best progress:

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/empty.h> //

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg; //

rcl_publisher_t power_pub;  //
std_msgs__msg__Empty power_msg; //

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

////////////////////////////////////////////////////////////////////////
void error_loop() {
  while (1) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
    RCSOFTCHECK(rcl_publish(&power_pub, &power_msg, NULL));
  }
}

/**********************************
  SETUP
**********************************/
void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "micro_ros_arduino_node_publisher")); //(parent_folder, sub_folder, msg_type), topic_name

  RCCHECK(rclc_publisher_init_default(
            &power_pub,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Empty),
            "power_on"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
            &timer,
            &support,
            RCL_MS_TO_NS(timer_timeout),
            timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 1;   //parameter data of Int32 msg

}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
As for the result, topic `/power_on` does publish but on the same frequency with `/micro_ros_arduino_node_publisher`.

So, the next question is: How can I make `/power_on` publishes at different rate than `/micro_ros_arduino_node_publisher`?

Several thing which still unclear to me is the role of these objects:
- executor
- support
- allocator

Acuadros95, 2021-08-02:
To clarify:
- Executor: Manages the execution of subscriptions and timer callbacks.
- Support: Groups initialization parameters, check [here](https://github.com/ros2/rclc/tree/foxy/rclc#rcl-convenience-functions).
- Allocator: Manages dynamic memory allocation functions (malloc, free, ...), as some boards or RTOS implements their own.

You can check this [tutorial](https://micro.ros.org//docs/tutorials/core/programming_rcl_rclc), but we are currently working to improve this section.

As to have a different publishing rate you just need to create another timer with its own callback and add it to the executor, like this:

```
// New timer
rcl_timer_t timer_power;

// Callback function:
void publish_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&power_pub, &power_msg, NULL));
  }
}

// On your setup function:
... 
// create timer
const unsigned int publish_period = 1000;  // Set publish period
RCCHECK(rclc_timer_init_default(
          &timer_power,
          &support,
          RCL_MS_TO_NS( publish_period),
          publish_callback));

// create executor
RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator)); // Increase executor handle number from 1 to 2 (Number of timers + subscriptions)
RCCHECK(rclc_executor_add_timer(&executor, &timer));
RCCHECK(rclc_executor_add_timer(&executor, &timer_power));
```

Then when `rclc_executor_spin_some` is called, the executor will call each timer callback periodically.
You can also just call `rcl_publish` in your code and manage the timing yourself, as the executor is not needed to publish.

irsyadtc, 2021-08-03:
Thank you @Acuadros95. I really appreciate your comment.

Here is my complete sketch of micro-ros_publisher with added new Empty msg:
```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/empty.h> //

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;

rcl_publisher_t power_pub;  //
std_msgs__msg__Empty power_msg; //

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

rcl_timer_t power_on_timer; //

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

////////////////////////////////////////////////////////////////////////
void error_loop() {
  while (1) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

/**********************************
  CALLBACK
**********************************/
void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

//new callback for power_on
void power_on_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&power_pub, &power_msg, NULL));
  }
}

/**********************************
  SETUP
**********************************/
void setup() {
  set_microros_transports();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
            "micro_ros_arduino_node_publisher")); //(parent_folder, sub_folder, msg_type), topic_name

  // new publisher of power_on
  RCCHECK(rclc_publisher_init_default(
            &power_pub,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Empty),
            "power_on"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
            &timer,
            &support,
            RCL_MS_TO_NS(timer_timeout),
            timer_callback));

  // new timer for power_callback
  const unsigned int power_on_timer_timeout = 500;
  RCCHECK(rclc_timer_init_default(
            &power_on_timer,
            &support,
            RCL_MS_TO_NS(power_on_timer_timeout),
            power_on_callback));

  // create executor
  // Increase executor handle number from 1 to 2 (Number of timers + subscriptions)
  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));
  
  RCCHECK(rclc_executor_add_timer(&executor, &timer));
  RCCHECK(rclc_executor_add_timer(&executor, &power_on_timer)); //new added executor for power_on_timer

  //initial value
  msg.data = 1;   //parameter data of micro_ros_arduino_node_publisher Int32 msg

}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

closed 2021-08-03 by irsyadtc