micro-ROS/micro_ros_arduino #413: Int64MultiArray subscriber not working

Sarwan1494, 2021-08-10:
## Issue template
using OpenCR to control dynamixels. The subscriber(Int64MultiArray) is not working. When I change the message type to Int32 it is working. I read the documentation and initialized the array(mentioned below). 
kindly inform me how to implement it.  

int64_t array_tic[2]= {left_tick,right_tick};
int64_t value[10] = {0,0,0,0,0,0,0,0,0,0};

goal_vel = std_msgs__msg__Int64MultiArray__create();
      goal_vel->data.capacity = 10;
      goal_vel->data.data = (int64_t*)malloc(goal_vel->data.capacity * sizeof(int64_t));
      goal_vel->data.size = sizeof(value)/sizeof(int64_t);memcpy(goal_vel->data.data, value,sizeof(value));
//            
      ticks = std_msgs__msg__Int64MultiArray__create();  
      ticks->data.capacity = 2;
      ticks->data.data = (int64_t*)malloc(sizeof(array_tic));   
      ticks->data.size = sizeof(array_tic)/sizeof(int64_t);memcpy(ticks->data.data,array_tic,sizeof(array_tic));

//////Callback function:

void subscription_callback(const void * msgin)
{  
    const std_msgs__msg__Int64MultiArray * goal_vel = (const std_msgs__msg__Int64MultiArray *)msgin;
    int64_t left_front; 
    
//    value[0] = int64_t(goal_vel->data.data[0]);    
    //left_front = int32_t(msg->data);
    dxl.setGoalVelocity(DXL_ID,value[0]);
}

pablogs9, 2021-08-11:
Avoid using `std_msgs__msg__Int64MultiArray__create` and use the instruction in [this tutorial](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/)

Sarwan1494, 2021-08-20:
I tried using without std_msgs__msg__Int64MultiArray__create. Still, it is not working.

since I want to subscribe to an array of six values, I am using this.

int64_t value[6] = {0,0,0,0,0,0};
 goal_vel->data.capacity = 6;
 goal_vel->data.data = (int64_t*)malloc(6*sizeof(int64_t));
 goal_vel->data.size = sizeof(value)/sizeof(int64_t);memcpy(goal_vel->data.data, value,sizeof(value));   //I tried assinging 0 and then assign values once the data is subscribed, but i cannot do it. so found this line in ros answers

in the callback function :
void subscription_callback(const void *msgin)
const std_msgs__msg__Int64MultiArray* goal_vel = (const std_msgs__msg__Int64MultiArray *)msgin;

goal_velocity_from_cmd[0] = goal_vel->data.data[0];
goal_velocity_from_cmd[1] = goal_vel->data.data[1];

dxl_driver.controlMotor(WHEEL_RADIUS, WHEEL_SEPARATION,  goal_velocity_from_cmd); //assigning the values to the motor 

I am new to programming. I still couldnot understand what the problem is. Kindly inform what is wrong

Thanks

pablogs9, 2021-08-23:
You are not initing the `layout` member. Check the [type](http://docs.ros.org/en/noetic/api/std_msgs/html/msg/Int64MultiArray.html)

Sarwan1494, 2021-08-23:
Thanks for the reply. I have read the documentation. From what I have understood, I should initialize the memory for layout i.e dim (labe , size , stride) and data offset

I have added these lines. still not working
goal_vel->layout.dim.data = (std_msgs__msg__MultiArrayDimension*)malloc(sizeof(std_msgs__msg__MultiArrayDimension));
  goal_vel->layout.dim.size = 2;   

Am I still missing something? 

Sorry for asking this but if you could provide me an example it would be really helpful.

Thanks

pablogs9, 2021-08-24:
That's correct, you need to init all the members of the type **recursively**:

```c
std_msgs__msg__Int64MultiArray msg;

msg.data.capacity = 100; 
msg.data.size = 0;
msg.data.data = (int64_t*) malloc(msg.data.capacity * sizeof(int64_t));

msg.layout.dim.capacity = 100;
msg.layout.dim.size = 0;
msg.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(msg.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));

for(size_t i = 0; i < msg.layout.dim.capacity; i++){
    msg.layout.dim.data[i].label.capacity = 20;
    msg.layout.dim.data[i].label.size = 0;
    msg.layout.dim.data[i].label.data = (char*) malloc(msg.layout.dim.data[i].label.capacity * sizeof(char));
}
```

Remember that if you are using micro-ROS Galactic you have available [micro_ros_utilities](https://github.com/micro-ROS/micro_ros_utilities) that can be used as explained [here](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/#micro-ros-galactic).

pablogs9, 2021-08-30:
Any update on this?

Sarwan1494, 2021-08-30:
HI,

 Thanks a lot for the example.  It is working fine.

Sorry for the late reply.

closed 2021-08-30 by pablogs9

JaYNits, 2023-09-06:
Hello, I Am using Teensy 4.1 for an Int64MultiArray Subscriber.

As mentioned above, I am recursively init  all the members, Teensy enters 
Into Error State.

code is mentioned below.

`#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/int64_multi_array.h>
#include <std_msgs/msg/multi_array_dimension.h>

rcl_subscription_t subscriber;
std_msgs__msg__Int64MultiArray * msg_test,msgin,msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(200);
  }
}

void subscription_callback(const void * msgin)
{
  std_msgs__msg__Int64MultiArray msg;

  msg.data.capacity = 2; 
  msg.data.size = 0;
  msg.data.data = (int64_t*) malloc(msg.data.capacity * sizeof(int64_t));

  msg.layout.dim.capacity = 0;
  msg.layout.dim.size = 0;
  msg.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(msg.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));

  for(size_t i = 0; i < msg.layout.dim.capacity; i++){
      msg.layout.dim.data[i].label.capacity = 0;
      msg.layout.dim.data[i].label.size = 0;
      msg.layout.dim.data[i].label.data = (char*) malloc(msg.layout.dim.data[i].label.capacity * sizeof(char));
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int64MultiArray),
    "wow"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1000)));
}
`

JaYNits, 2023-09-12:
> Hello, I Am using Teensy 4.1 for an Int64MultiArray Subscriber.
> 
> As mentioned above, I am recursively init all the members, Teensy enters Into Error State.
> 
> code is mentioned below.
> 
> ```c
> #include <micro_ros_arduino.h>
> 
> #include <stdio.h>
> #include <rcl/rcl.h>
> #include <rcl/error_handling.h>
> #include <rclc/rclc.h>
> #include <rclc/executor.h>
> 
> #include <std_msgs/msg/int32.h>
> #include <std_msgs/msg/float32_multi_array.h>
> #include <std_msgs/msg/int64_multi_array.h>
> #include <std_msgs/msg/multi_array_dimension.h>
> 
> rcl_subscription_t subscriber;
> std_msgs__msg__Int64MultiArray * msg_test,msgin,msg;
> rclc_executor_t executor;
> rclc_support_t support;
> rcl_allocator_t allocator;
> rcl_node_t node;
> rcl_timer_t timer;
> 
> #define LED_PIN 13
> 
> #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
> #define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}
> 
> 
> 
> void error_loop(){
>   while(1){
>     digitalWrite(LED_PIN, !digitalRead(LED_PIN));
>     delay(200);
>   }
> }
> 
> 
> void subscription_callback(std_msgs__msg__Int64MultiArray msg)
> {
>   msg.data.capacity = 2; 
>   msg.data.size = 0;
>   msg.data.data = (int64_t*) malloc(msg.data.capacity * sizeof(int64_t));
> 
>   msg.layout.dim.capacity = 0;
>   msg.layout.dim.size = 0;
>   msg.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(msg.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));
> 
>   for(size_t i = 0; i < msg.layout.dim.capacity; i++){
>       msg.layout.dim.data[i].label.capacity = 0;
>       msg.layout.dim.data[i].label.size = 0;
>       msg.layout.dim.data[i].label.data = (char*) malloc(msg.layout.dim.data[i].label.capacity * sizeof(char));
>   }
> }
> 
> void setup() {
>   set_microros_transports();
>   
>   pinMode(LED_PIN, OUTPUT);
>   digitalWrite(LED_PIN, HIGH);  
>   
>   delay(2000);
> 
>   allocator = rcl_get_default_allocator();
> 
>   //create init_options
>   RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
> 
>   // create node
>   RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));
> 
>   // create subscriber
>   RCCHECK(rclc_subscription_init_default(
>     &subscriber,
>     &node,
>     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int64MultiArray),
>     "wow"));
> 
>   // create executor
>   RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
>   RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
> }
> 
> void loop() {
>   delay(100);
>   RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1000)));
> }
> ```

@pablogs9 any suggestions?

pablogs9, 2023-09-12:
You are continuously allocating memory in the sub callback. It needs to be controlled (at some point it shall be freed) and more importantly, it shall be allocated **before** initing the subscription.

JaYNits, 2023-09-12:
I searched for a reference or an example code for this, even [link](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/#micro-ros-galactic)) i couldn't  make much out of this 

i was not able to implement an Int64Multiarray subscriber, if you have any Int64Multiarray subscriber code can you share it here?

pablogs9, 2023-09-12:
```c
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/int64_multi_array.h>
#include <std_msgs/msg/multi_array_dimension.h>

rcl_subscription_t subscriber;
std_msgs__msg__Int64MultiArray msg_in;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;º
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(200);
  }
}

void subscription_callback(std_msgs__msg__Int64MultiArray msg)
{
    // At this point msg is filled, just use it or print it, for example:
    for(size_t i = 0; i < msg_in.layout.dim.size; i++){
      prinf("laber at msg_in.data[%d] = %s\n", i, msg_in.layout.dim.data[i].label.data);
    }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int64MultiArray),
    "wow"));

  // Init the memory of your array in order to provide it to the executor.
  // If a message from ROS comes and it is bigger than this, it will be ignored, so ensure that capacities here are big enought.
  msg_in.data.capacity = 10; 
  msg_in.data.size = 0;
  msg_in.data.data = (int64_t*) malloc(msg_in.data.capacity * sizeof(int64_t));

  msg_in.layout.dim.capacity = 10;
  msg_in.layout.dim.size = 0;
  msg_in.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(msg_in.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));

  for(size_t i = 0; i < msg_in.layout.dim.capacity; i++){
      msg_in.layout.dim.data[i].label.capacity = 10;
      msg_in.layout.dim.data[i].label.size = 0;
      msg_in.layout.dim.data[i].label.data = (char*) malloc(msg_in.layout.dim.data[i].label.capacity * sizeof(char));
  }

  // ENSURE HERE THAT ALL THE MEMBERS ARE INIT RECURSIVELY, I HAVE JUST COPIED YOUR CODE

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg_in, &subscription_callback, ON_NEW_DATA));
}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1000)));
}
```

JaYNits, 2023-09-13:
@pablogs9 Thanks for the example. It worked

marrenryan, 2023-09-29:
This doesn't work. I get this error: `invalid conversion from 'void (*)(std_msgs__msg__Int64MultiArray)' to 'rclc_subscription_callback_t' {aka 'void (*)(const void*)'} [-fpermissive]

At the end of the day, I need this to work as a multi array float publisher. Thought I would start here

pablogs9, 2023-10-02:
use `void subscription_callback(std_msgs__msg__Int64MultiArray * msg)` as the compiler is suggesting