micro-ROS/micro_ros_arduino #415: Can't create topic and further participant while running micro_ros_agent

alienshek337, 2021-08-11:
- Hardware description: Teensy 3.2
- RTOS: from the tutorial https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/
- Version or commit hash: foxy

#### Steps to reproduce the issue
[image]

#### Expected behavior
Create and run all participant successfully

#### Actual behavior
Didn't run the create topic and other participant

#### Additional information
I didn't change any domain ID and I'm using Arduino IDE 1.8.15

pablogs9, 2021-08-11:
Which other participant? If you are using `micro-ROS-publisher` example code you only have one participant.

If you have modified the code, please provide more details.

alienshek337, 2021-08-11:
Sorry, my mistake, there is no other participants. I am just using the micro-ROS-publisher example code, but it seems like stopped after participant created.

pablogs9, 2021-08-11:
Can you copy-paste here the whole output of the agent?

alienshek337, 2021-08-11:
user@user:~$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v 6
[1628664844.055905] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1628664844.056105] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1628664844.056231] info     | Root.cpp           | create_client            | create                 | client_key: 0x50C7822F, session_id: 0x81
[1628664844.056266] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x50C7822F, address: 0
[1628664844.056285] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628664844.056366] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628664844.056402] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628664844.057869] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 112, data: 
0000: 81 80 00 00 01 05 66 00 00 0A 00 01 01 02 00 00 57 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75
0040: 69 6E 6F 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70
0060: 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1628664844.065206] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x50C7822F, participant_id: 0x000(1)
[1628664844.065293] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1628664844.065316] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628664844.065873] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628664844.065883] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628664844.065907] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628664844.065912] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628664844.065930] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628664844.065937] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628664844.065941] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628664844.069862] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 148, data: 
0000: 81 80 01 00 01 05 89 00 00 0B 00 02 02 02 00 00 7B 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F
0040: 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73
0060: 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E
0080: 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00 00
[1628664844.069881] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x50C7822F, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
user@user:~$

pablogs9, 2021-08-11:
Is your agent exiting automatically or you are closing it?
Have you tried to use a dockerized version of the agent with:

```bash
docker pull microros/micro-ros-agent:foxy
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev [YOUR BOARD PORT] -v6
```

pablogs9, 2021-08-11:
Also, it seems that you are using a old version of the micro-ROS library. Can you update the library to the latest commit?

alienshek337, 2021-08-11:
The agent exiting automatically.
I am using the micro-ROS library v1.0.0 .

Now I tried to run in docker version and the agent out put like this:
user@user:~$ sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1628665199.507207] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1628665199.507388] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1628665199.507480] info     | Root.cpp           | create_client            | create                 | client_key: 0x50C7822F, session_id: 0x81
[1628665199.507528] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x50C7822F, address: 0
[1628665199.507562] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507571] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507579] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507587] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507594] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507597] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507600] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507626] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507629] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507648] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507648] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507668] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507668] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x50C7822F, address: 0
[1628665199.507697] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507711] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507717] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507724] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507728] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628665199.507733] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x50C7822F, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

It didn't exit automatically this time but stopped before participant create.

alienshek337, 2021-08-11:
Thank you! I run it successfully after I used the latest commit from the foxy branch and run it in docker version.

closed 2021-08-11 by alienshek337

pablogs9, 2021-08-11:
Could you point if with the latest commit and you compiled agent it works?

huak95, 2021-08-19:
Hello, @pablogs9 
Hardware description: Arduino Due
Version or commit hash: foxy

I clone the foxy repo and use [this](https://micro.ros.org/docs/tutorials/core/first_application_linux/) to create micro-ROS agent I can use ping-pong example without any problem. 
Then, I put foxy repo in Arduino library and upload micro-ros_publisher example.
I run the command `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6` and got the same as other.
But when I use `ros2 topic list` it shown only /parameter_events and /rosout topic

I use docker version `sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6` but I got the same results.

Thank you.

huak95, 2021-08-19:
Here is my output from the `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6` command
```
saksorn@saksorn-ros:~/turtlebot/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6
[1629367160.838590] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1629367160.838810] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1629367160.839282] info     | Root.cpp           | create_client            | create                 | client_key: 0x07720432, session_id: 0x81
[1629367160.839326] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x07720432, address: 0
[1629367160.839367] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x07720432, address: 0
[1629367160.839398] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1629367160.839445] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1629367160.839731] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 07 72 04 32 81 00 FC 01
[1629367160.839746] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 07 72 04 32 81 00 FC 01
[1629367160.839750] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 07 72 04 32 81 00 FC 01
[1629367160.839850] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x07720432, address: 0
[1629367160.839864] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x07720432, address: 0
[1629367160.839870] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x07720432, address: 0
[1629367160.839929] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1629367160.839950] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1629367160.839955] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1629367160.855677] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 56, data: 
0000: 81 80 00 00 01 07 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 08 00 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 00 00 00 00 00 00
[1629367160.861936] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x07720432, participant_id: 0x000(1)
[1629367160.862040] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1629367160.862062] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1629367160.867828] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1629367160.879982] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 100, data: 
0000: 81 80 01 00 01 07 5A 00 00 0B 00 02 02 03 00 00 4C 00 00 00 24 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 5F 70 75 62 6C 69 73 68 65 72 00 00 01 09 00
0040: 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00
0060: 00 01 00 00
[1629367160.880133] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x07720432, topic_id: 0x000(2), participant_id: 0x000(1)
[1629367160.880210] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1629367160.880219] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1629367160.888532] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1629367160.892254] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 24, data: 
[... cut ...]
[1629367162.014627] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x07720432, len: 16, data: 
0000: 81 80 04 00 07 01 08 00 00 0E 00 05 00 00 00 00
[1629367162.014874] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data: 
0000: 00 00 00 00
[1629367162.015062] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x07720432, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
```