micro-ROS/micro_ros_arduino #429: Error: undefined reference, implementation in arduino DUE

OtaviodaCruz, 2021-08-24:
## Issue template
Sorry for my english (thanks google translation). Note: I am young in micro_ros.

#### Setup:
- I am using arduino due connected via usb to my ubuntu 20.04 computer with ros2 foxy.

#### Steps to reproduce the issue
- From link (https://github.com/micro-ROS/micro_ros_arduino/releases) downloaded v2.0.0-foxy;
- In my arduino IDE I added the .zip library;
- Ii'm using the "micro-ros_publisher" example to my arduino DUE.

#### Actual behavior
- So, I plugged my arduino DUE into my computer and tried to download the schedule. The following error appears on the screen:

```
/home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/arduino-builder -dump-prefs -logger=machine -hardware /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/hardware -hardware /home/otavio/.arduino15/packages -tools /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/tools-builder -tools /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/hardware/tools/avr -tools /home/otavio/.arduino15/packages -built-in-libraries /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/libraries -libraries /home/otavio/Arduino/libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10815 -build-path /tmp/arduino_build_260334 -warnings=none -build-cache /tmp/arduino_cache_74308 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/otavio/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/home/otavio/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /tmp/arduino_modified_sketch_640893/micro-ros_publisher.ino
/home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/arduino-builder -compile -logger=machine -hardware /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/hardware -hardware /home/otavio/.arduino15/packages -tools /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/tools-builder -tools /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/hardware/tools/avr -tools /home/otavio/.arduino15/packages -built-in-libraries /home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/libraries -libraries /home/otavio/Arduino/libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10815 -build-path /tmp/arduino_build_260334 -warnings=none -build-cache /tmp/arduino_cache_74308 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/otavio/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/home/otavio/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /tmp/arduino_modified_sketch_640893/micro-ros_publisher.ino
Using board 'arduino_due_x_dbg' from platform in folder: /home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12
Using core 'arduino' from platform in folder: /home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12
Detecting libraries used...
/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10815 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /tmp/arduino_build_260334/sketch/micro-ros_publisher.ino.cpp -o /dev/null
Alternatives for micro_ros_arduino.h: [micro_ros_arduino-2.0.0-foxy@0.0.4]
ResolveLibrary(micro_ros_arduino.h)
  -> candidates: [micro_ros_arduino-2.0.0-foxy@0.0.4]
/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10815 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src /tmp/arduino_build_260334/sketch/micro-ros_publisher.ino.cpp -o /dev/null
Using cached library dependencies for file: /home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src/default_transport.cpp
Using cached library dependencies for file: /home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src/wifi_transport.cpp
Generating function prototypes...
/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10815 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src /tmp/arduino_build_260334/sketch/micro-ros_publisher.ino.cpp -o /tmp/arduino_build_260334/preproc/ctags_target_for_gcc_minus_e.cpp
/home/otavio/Downloads/arduino-1.8.15-linux64/arduino-1.8.15/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_260334/preproc/ctags_target_for_gcc_minus_e.cpp
Compilando sketch...
/home/otavio/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10815 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/home/otavio/.arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src /tmp/arduino_build_260334/sketch/micro-ros_publisher.ino.cpp -o /tmp/arduino_build_260334/sketch/micro-ros_publisher.ino.cpp.o
[... cut ...]
/tmp/arduino_modified_sketch_640893/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Usando a biblioteca micro_ros_arduino-2.0.0-foxy na versão 0.0.4 na pasta: /home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy 
exit status 1
Erro compilando para a placa Arduino Due (Programming Port)
```
I didn't expect this behavior. Do I have to set some values ​​before running? I'm forgetting something?

Thank you very much in advance.

huak95, 2021-08-24:
I think you didn't patch SAMD
Here is the instruction from home page of Micro-ROS Arduino

Go inside your Arduino installation and replace platform.txt:
For me using Ubuntu 20.04 the Arduino installation path is /snap/arduino/61/.arduino15/packages/arduino

```
export ARDUINO_PATH=[Your Arduino path]
cd $ARDUINO_PATH/hardware/sam/1.6.12/
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/galactic/extras/patching_boards/platform_arduinocore_sam.txt > platform.txt
```

pablogs9, 2021-08-24:
Hello @OtaviodaCruz, check the solution @huak95 has written. If this does not work I will take a look.

@huak95 thanks for the answer!

OtaviodaCruz, 2021-08-27:
Hello @pablogs9. Yes, I solved the problem.

Thanks for the answers.

closed 2021-08-30 by pablogs9