micro-ROS/micro_ros_arduino #430: Error using micro-ROS on Foxy without docker

huak95, 2021-08-24:
## Issue template

Hardware description: Arduino Due
RTOS: NO
micro-ros-Agent: I was followed micro-ros [instruction](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/)
Version or commit hash: foxy 
OS: Ubuntu 20.04

#### Steps to reproduce the issue
I using Foxy pre-compile released and use micro-ros [instruction](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/) to create micro-ROS agent
Then, I put foxy repo in Arduino library and upload micro-ros_publisher example.

I use docker version `sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6` It's work fine. the topic is have when I using `sudo docker run -ti --rm --net=host ros:foxy ros2 topic list` command.

But when I build the Micro-ROS on my computer and run the command `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`
The results in the terminal is same as docker version.
But when I use `ros2 topic list` it shown only /parameter_events and /rosout topic

I try follow #196 Issue and use generate_lib options to create Micro-ROS Agent but I not working. 

I don't know what exactly the main issue of this. 
Do I need to compile the Micro-ROS Agent by my self. Or compile the Micro-ROS Arduino.

I want to use the ROS on Raspberry Pi 4 and when using docker version is very slow and complicated to use.

Thank you.

#### Expected behavior

The topic is shown.

#### Actual behavior

Topic not shown.

#### Additional information

It's my first actual robot build in ROS, I don't know that I choose wrong for ROS2 Version it's seem have a lot of problem. So I want to know from robotic professionals. If I can communicate ROS with MCU the next step for building ROS2 Robot (Ex. NAVIGATION, SLAM and AUTONOMOUS DRIVING) it's will be a lot of bug or not, If It does have a lot of bug. So, I will switch back to ROS1 that have everything. Thank you. I feel I am lost to build the ROBOT NOW. TT

huak95, 2021-08-24:
## Update

The Raspberry Pi won't run the docker command cause by ARM Architecture.

```
ubuntu@ubuntu:~/microros_wssudo docker run --init -it -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
[FATAL tini (8)] exec /bin/sh failed: Exec format error

```

pablogs9, 2021-08-24:
If I have understood correctly when you run micro-ROS agent and ROS 2 in docker it works, but if you run both of them in your local machine it does not subscribe to topics correctly? Is that correct?

huak95, 2021-08-24:
Yes, it corrected.

pablogs9, 2021-08-24:
Which version of ROS 2 you have installed locally?

huak95, 2021-08-24:
It's ROS2 foxy. From [this tutorial](https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html).

pablogs9, 2021-08-24:
If you use the docker micro-ROS agent and then you try to subscribe from your local installation, does it works?

huak95, 2021-08-24:
I have tried, It doesn't work locally.
[image]

pablogs9, 2021-08-24:
Try the same with:

```bash
docker pull microros/micro-ros-agent:foxy
sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
```

huak95, 2021-08-24:
Not work.

pablogs9, 2021-08-24:
Try the same but doing:

```bash
wget https://raw.githubusercontent.com/micro-ROS/docker/galactic/micro-ROS-Agent/disable_fastdds_shm.xml -o /tmp/disable_fastdds_shm.xml
FASTRTPS_DEFAULT_PROFILES_FILE=/tmp/disable_fastdds_shm.xml  ros2 topic list
```

huak95, 2021-08-24:
It's doesn't work.
[image]

But my arduino due led is blinking. So, It's working but it's not connect to local.

pablogs9, 2021-08-24:
Make sure that the client is connected to the agent before testing `FASTRTPS_DEFAULT_PROFILES_FILE`.

huak95, 2021-08-24:
How to know client is connected to the agent?
Run the ping-pong example?

Acuadros95, 2021-08-24:
Just to check, make sure you are on the default ROS domain id using `export ROS_DOMAIN_ID=0`

huak95, 2021-08-24:
[image]
After using 
`export ROS_DOMAIN_ID=0` command I check in the `export` having the ROS_DOMAIN_ID=0 in the ubuntu.
This is connected?

huak95, 2021-08-24:
I just wonder, Can I install everything like in Docker.
Maybe some file is not corrected.

Acuadros95, 2021-08-24:
No, that environment variable sets the ROS2 domain ID for that terminal to the default one. You need to be in the same domain id as the publisher node to view and subscribe to each topic.

To make sure that the client is connected to the agent, you can just look at the agent log.

huak95, 2021-08-24:
If I run in local machine without `-v6` 

```
~/microros_wsros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1629791849.838779] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1629791851.987378] info     | Root.cpp           | create_client            | create                 | client_key: 0x094B717E, session_id: 0x81
[1629791851.987565] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x094B717E, address: 0
[1629791853.010352] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x094B717E, participant_id: 0x000(1)
[1629791853.027559] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x094B717E, topic_id: 0x000(2), participant_id: 0x000(1)
[1629791853.039760] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x094B717E, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1629791853.052580] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x094B717E, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1629791853.069134] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x094B717E, topic_id: 0x001(2), participant_id: 0x000(1)
[1629791853.080808] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x094B717E, publisher_id: 0x000(3), participant_id: 0x000(1)
[1629791853.093304] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x094B717E, datawriter_id: 0x000(5), publisher_id: 0x000(3)

```

It shown like this I think the client can connect to the agent.

Acuadros95, 2021-08-24:
Yes, it looks like the communication with the board is fine. This looks like a problem with your ROS2 environment. How did you install ROS2?

huak95, 2021-08-24:
@Acuadros95 You are my life saver!
After I use `export ROS_DOMAIN_ID=0` before `ros2 topic list` my topic was shown.
[image]
What is does the `export ROS_DOMAIN_ID=0` do?

Acuadros95, 2021-08-24:
The `ROS_DOMAIN_ID` environment variable sets the ROS2 domain id used, which can be used to isolate nodes from each other. From the [docs](https://docs.ros.org/en/foxy/Concepts/About-Domain-ID.html):

> As explained elsewhere, the default middleware that ROS 2 uses for communication is DDS. In DDS, the primary mechanism for having different logical networks share a physical network is known as the Domain ID. ROS 2 nodes on the same domain can freely discover and send messages to each other, while ROS 2 nodes on different domains cannot. All ROS 2 nodes use domain ID 0 by default. To avoid interference between different groups of computers running ROS 2 on the same network, a different domain ID should be set for each group.

I am closing this, Glad to hear it worked!

closed 2021-08-24 by Acuadros95