micro-ROS/micro_ros_arduino #434: Teensy 4.0 Stops publishing after 71 minutes

grassjelly, 2021-08-24:
#### Setup
- Hardware description: Teensy 4.0
- OS: Ubuntu 20.04 - Foxy
- Installation type:Default installation process from the README
- Version or commit hash: Galactic Branch

#### Steps to reproduce the issue
- Uploaded [reconnection example](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/examples/micro-ros_reconnection_example) 
- Run the agent:
    `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6
`
- Subscribe to std_msgs_msg_Int32 topic

#### Expected behavior
The data should continuously be published.

#### Actual behavior
The data stops being published after 71 minutes. I tried it a couple of times just to make sure that the error is reproducible. 
Here's the output from the agent after the data has stopped streaming. It continuously prints the following but no data is received on the subscriber side:
```

[1629801405.336827] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4CDC5462, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1629801405.336981] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4CDC5462, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1629801405.895889] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4CDC5462, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1629801405.896161] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4CDC5462, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1629801406.454880] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4CDC5462, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1629801406.455133] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4CDC5462, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1629801407.013945] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4CDC5462, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1629801407.014202] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4CDC5462, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1629801407.572879] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4CDC5462, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1629801407.573176] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4CDC5462, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
```

And here's the last 7 data received from the subscriber ( `ros2 topic echo std_msgs_msg_Int32`) before it stopped receiving anything:

```
data: 4286
---
data: 4287
---
data: 4288
---
data: 4289
---
data: 4290
---
data: 4291
---
data: 4292
---

```
#### Additional information
I thought this could related to this issue https://github.com/micro-ROS/micro_ros_arduino/issues/232#issuecomment-829351791, so I've overridden clock_gettime function in my .ino file and use elapsedMicros instead.:

```
 int clock_gettime(clockid_t unused, struct timespec *tp)
 {
     (void)unused;

     static elapsedMicros m;
     tp->tv_sec = m / 1000000;
     tp->tv_nsec = (m % 1000000) * 1000;

     return 0;
 }
```
I re-uploaded the firmware after I added the function but the data still halts at 4292.

pablogs9, 2021-08-24:
Have you tried to modify the `clock_gettime` in the library as in https://github.com/micro-ROS/micro_ros_arduino/issues/232#issuecomment-829351791?

grassjelly, 2021-08-24:
Yes, I modified the default transport as well but the same results. Do I need to rebuild the library the whole micro-ros library tho?

pablogs9, 2021-08-24:
Not required to rebuild the library. Do you have the same behaviour with a simple publisher instead of the reconnection example?

grassjelly, 2021-08-24:
I first found out about this problem while testing my robot. I tested it on the reconnection example just to ensure it’s not some overseen bug on my code. To answer your question, I haven’t tested on reconnection example but I did tried on my robot code where I omitted the reconnection routines and purely pub-sub modules. Still no luck and halts at 71~minutes.

grassjelly, 2021-08-24:
More info, ran the same robot code on Teensy 3.2 and 3.5 and seems to be working fine. I’m only having this problem on Teensy 4.0

pablogs9, 2021-08-24:
We are going to check how does `elapsedMicros` differ from Teensy 3.x to Teensy 4.x.

grassjelly, 2021-08-24:
thanks @pablogs9. Btw there were no modification done on clock_gettime when running on Teensy 3.x. I just modified it on Teensy 4.0 hoping it’ll fix the issue

grassjelly, 2021-08-24:
Just an update @pablogs9 . I tried to change micros https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/src/default_transport.cpp#L23 from `uint64_t` to `unsigned long` and seemed to have solved the issue.  Current data received is

` data: 7332
`