micro-ROS/micro_ros_arduino #461: Micro ros for 2 or more nodes not working

OtaviodaCruz, 2021-09-05:
#### Hardware description
- Arduino Due and PC with ros2 foxy.

#### Steps to reproduce the issue

Step 1:

- Downloaded and loaded the micro ros foxy library in Arduino: https://github.com/micro-ROS/micro_ros_arduino/releases;
- I performed the instructions from: https://micro.ros.org/docs/tutorials/core/first_application_linux/ on my computer;
- Loading micro-ros_publisher example in arduino Due;
- Execute ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 in my terminal;
- So far everything works as expected;
-
Step 2:

- Now, I want to work with two nodes, for that I modified RMW_UXRCE_MAX_NODES in: Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src/rmw_microxrcedds_c/config.h, to accept more than 1 node;
- For my code I used it as a reference: https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/int32_multinode/main.c. With some modifications, my code is:

> #include <micro_ros_arduino.h>
> #include <stdio.h>
> #include <rcl/rcl.h>
> #include <rcl/error_handling.h>
> #include <rclc/rclc.h>
> #include <rclc/executor.h>
> #include <std_msgs/msg/int32.h>
> 
> #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
> #define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}
> #define LED_PIN 13
> 
> rcl_publisher_t publisher_1;
> rcl_subscription_t subscriber_1;
> std_msgs__msg__Int32 send_msg_1;
> std_msgs__msg__Int32 recv_msg_1;
> 
> rcl_publisher_t publisher_2;
> rcl_subscription_t subscriber_2;
> std_msgs__msg__Int32 send_msg_2;
> std_msgs__msg__Int32 recv_msg_2;
> 
> rclc_executor_t executor_2 = rclc_executor_get_zero_initialized_executor();
> rclc_executor_t executor_1 = rclc_executor_get_zero_initialized_executor();
> 
> void error_loop(){
>   while(1){
>     digitalWrite(LED_PIN, !digitalRead(LED_PIN));
>     delay(100);
>   }
> }
> 
> 
> void timer_callback_1(rcl_timer_t * timer, int64_t last_call_time)
> {
>    (void) last_call_time;
>   if (timer != NULL) {
>     RCSOFTCHECK(rcl_publish(&publisher_1, &send_msg_1, NULL));
>     send_msg_1.data++;
>   }
> }
> 
> void subscription_callback_1(const void * msgin)
> {
>   const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
> }
> 
> void timer_callback_2(rcl_timer_t * timer, int64_t last_call_time)
> {
>   (void) last_call_time;
>   if (timer != NULL) {
>     RCSOFTCHECK(rcl_publish(&publisher_2, &send_msg_2, NULL));
>     send_msg_2.data++;
>   }
> }
> 
> void subscription_callback_2(const void * msgin)
> {
>   const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
> }
> 
> 
> void setup() 
> 
> {
>   
>   set_microros_transports();
>   pinMode(LED_PIN, OUTPUT);
>   digitalWrite(LED_PIN, HIGH);  
>   delay(2000);
>   rcl_allocator_t allocator = rcl_get_default_allocator();
>   rclc_support_t support;
> 
>   // create init_options
>   RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
> 
>   // ----- NODE 1 -----
>   // create node
>   rcl_node_t node_1 = rcl_get_zero_initialized_node();
>   RCCHECK(rclc_node_init_default(&node_1, "node_1", "", &support));
> 
>   // create publisher
>   RCCHECK(rclc_publisher_init_default(
>     &publisher_1,
>     &node_1,
>     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
>     "node_1_to_2"));
> 
>   // create subscriber
>   RCCHECK(rclc_subscription_init_default(
>     &subscriber_1,
>     &node_1,
>     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
>     "node_2_to_1"));
> 
>   // create timer,
>   rcl_timer_t timer_1;
>   RCCHECK(rclc_timer_init_default(
>     &timer_1,
>     &support,
>     RCL_MS_TO_NS(1000),
>     timer_callback_1));
> 
>   // create executor
> 
>   RCCHECK(rclc_executor_init(&executor_1, &support.context, 2, &allocator));
>   RCCHECK(rclc_executor_add_timer(&executor_1, &timer_1));
>   RCCHECK(rclc_executor_add_subscription(&executor_1, &subscriber_1, &recv_msg_1, &subscription_callback_1, ON_NEW_DATA));
> 
>   // ----- NODE 2 -----
>   // create node
>   rcl_node_t node_2 = rcl_get_zero_initialized_node();
>   RCCHECK(rclc_node_init_default(&node_2, "node_2", "", &support));
> 
>   // create publisher
>   RCCHECK(rclc_publisher_init_default(
>     &publisher_2,
>     &node_2,
>     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
>     "node_2_to_1"));
> 
>   // create subscriber
>   RCCHECK(rclc_subscription_init_default(
>     &subscriber_2,
>     &node_2,
>     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
>     "node_1_to_2"));
> 
>   // create timer,
>   rcl_timer_t timer_2;
>   RCCHECK(rclc_timer_init_default(
>     &timer_2,
>     &support,
>     RCL_MS_TO_NS(1000),
>     timer_callback_2));
> 
>   // create executor
> 
>   RCCHECK(rclc_executor_init(&executor_2, &support.context, 2, &allocator));
>   RCCHECK(rclc_executor_add_timer(&executor_2, &timer_2));
>   RCCHECK(rclc_executor_add_subscription(&executor_2, &subscriber_2, &recv_msg_2, &subscription_callback_2, ON_NEW_DATA));
> 
>   // ----- MAIN LOOP -----
>   send_msg_1.data = 0;
>   send_msg_2.data = 100;
> }
> void loop()
> {
>       RCSOFTCHECK(rclc_executor_spin_some(&executor_1, RCL_MS_TO_NS(100)));
>       RCSOFTCHECK(rclc_executor_spin_some(&executor_2, RCL_MS_TO_NS(100)));
>   }
- then, when you go to execute the `RCCHECK(rclc_node_init_default(&node_2, "node_2", "", &support))` command something goes wrong and the LED starts blinking.

Thanks.

pablogs9, 2021-09-06:
Couple of things:
1. Are you [rebuilding the library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) after modifying the [max node number](https://github.com/micro-ROS/micro_ros_arduino/blob/3a2ba8720791c2b0d7d0ddd0bead29c2e30e15a8/extras/library_generation/colcon_lowmem.meta#L43)?
2. You are creating `rcl_node_t node_1` and `rcl_node_t node_2` in `setup()` stack so the may not exists in `loop()` stack, define them as global variables.

OtaviodaCruz, 2021-09-06:
1. No, I just went to the arduino libraries, found the micro ros and changed the specified value. Do I need to do anything else?
2. Ok.

pablogs9, 2021-09-07:
Yes, you need to [rebuild the library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library).  The colcon.meta file that you have modified is just the configuration file for the library build process.

OtaviodaCruz, 2021-09-07:
- I didn't exactly change this file, I changed  [this one](https://github.com/micro-ROS/micro_ros_arduino/blob/3a2ba8720791c2b0d7d0ddd0bead29c2e30e15a8/src/rmw_microxrcedds_c/config.h#:~:text=RMW_UXRCE_MAX_SESSIONS).

- I'm not using the docker. Do I need to use docker?

- Maybe I wasn't very clear, but i downloaded and loaded the library through my arduino IDE (Sketch -> Include library -> Add .ZIP Library). In this case, I believe you don't need to build the library, am I wrong?

- wreck I really need to build, the command would be: colcon build?

OtaviodaCruz, 2021-09-07:
Sorry for possible basic questions, but i'm new.

pablogs9, 2021-09-08:
> I didn't exactly change this file, I changed this one.

This header file is autogenerated, so modifying it will not change the inner memory pools

> I'm not using the docker. Do I need to use docker?

The easiest way of building the library is using Docker, you also can make it locally but it harder (check the [Dockerfile](https://github.com/micro-ROS/docker/blob/galactic/micro-ROS-static-library-builder/Dockerfile) and the [script](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/library_generation.sh) for reference)

> Maybe I wasn't very clear, but i downloaded and loaded the library through my arduino IDE (Sketch -> Include library -> Add .ZIP Library). In this case, I believe you don't need to build the library, am I wrong?

When you download this repo (using git or the .ZIP) you are getting a precompiled version of the micro-ROS library for your platform. This precompiled version has a default configuration. If you want to modify these defaults, you need to rebuild the library with you settings.

> wreck I really need to build, the command would be: colcon build?

If you have a ROS2 installation you can [generate the library](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/) but the easies way is to use the Docker provided [here](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library). Internally it will use colcon to create the library.

OtaviodaCruz, 2021-09-09:
- I dowload [foxy lib](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.0-foxy);
- Extract the file into a new folder called "test";
-  In: ~/test/micro_ros_arduino-2.0.0-foxy/extras/library_generation/colcon_lowmem.meta, I edit
DRMW_UXRCE_MAX_NODES=3;
- In: ~/test/micro_ros_arduino-2.0.0-foxy/extras/library_generation/colcon.meta, I edit
DRMW_UXRCE_MAX_NODES=3;
- Do i need to edit some more files before building?
- Continue... I used the commands:
```
sudo docker pull microros/micro_ros_static_library_builder:foxy

sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_arduino-2.0.0-foxy/extras microros/micro_ros_static_library_builder:foxy
```
- Part of the output was:

```
Summary: 60 packages finished [20.9s]
#All required rosdeps installed successfully
#All required rosdeps installed successfully
Cloning into 'geometry2'...
remote: Enumerating objects: 13382, done.
remote: Counting objects: 100% (671/671), done.
remote: Compressing objects: 100% (223/223), done.
remote: Total 13382 (delta 451), reused 589 (delta 420), pack-reused 12711
Receiving objects: 100% (13382/13382), 3.14 MiB | 5.13 MiB/s, done.
Resolving deltas: 100% (9050/9050), done.
cp: cannot stat '/project/extras/library_generation/extra_packages/*': No such file or directory
usage: vcs import [-h] [--input FILE_OR_URL] [--force] [--shallow]
                  [--recursive] [--retry N] [--skip-existing] [--debug] [-w N]
                  [--repos]
                  [path]
vcs import: error: argument --input: can't open 'extra_packages.repos': [Errno 2] No such file or directory: 'extra_packages.repos'
find: ‘/project/src/’: No such file or directory
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /project/extras/library_generation/colcon.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
--- stderr: microcdr                                                
CMake Error at /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file:
  /project/extras/library_generation/opencr_toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:53 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
---
Failed   <<< microcdr [0.07s, exited with code 1]
Aborted  <<< rosidl_typesupport_interface [0.09s]
Aborted  <<< tinydir_vendor [0.12s]
Aborted  <<< rosidl_adapter [0.11s]

Summary: 0 packages finished [0.69s]
  1 package failed: microcdr
  3 packages aborted: rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  4 packages had stderr output: microcdr rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  57 packages not processed
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /project/extras/library_generation/colcon_verylowmem.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
--- stderr: microcdr                                                
[... 158 lines cut ...]
  /project/extras/library_generation/portenta-m7_toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:53 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
---
Failed   <<< microcdr [0.08s, exited with code 1]
Aborted  <<< tinydir_vendor [0.13s]
Aborted  <<< rosidl_typesupport_interface [0.10s]
Aborted  <<< rosidl_adapter [0.11s]

Summary: 0 packages finished [0.69s]
  1 package failed: microcdr
  3 packages aborted: rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  4 packages had stderr output: microcdr rosidl_adapter rosidl_typesupport_interface tinydir_vendor
  57 packages not processed
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
```
Apparently some packages were not loaded.
Meanwhile, in my "test" folder were created
[image]
Only in the folders there is no file.
- Are these the steps to rebuild the lib?

OtaviodaCruz, 2021-09-09:
Abount that

> This header file is autogenerated, so modifying it will not change the inner memory pools

I thought the execution logic would be with the arduino reading the specified file and setting those values ​​from us. Roughly, to me, it seemed like it was just a matter of a global variable...

pablogs9, 2021-09-09:
Here you can see a screencast of how to modify your colcon meta and rebuild the library:

https://asciinema.org/a/r2ERYw4d7wRaAZP8yEzIqZ8yM

I guess that you are not running the docker in the correct folder.

As you can see in the [supported board table](https://github.com/micro-ROS/micro_ros_arduino#micro-ros-for-arduino), Arduino due use `colcon_verylowmem.meta`. So this is the file that you should modify.

> I thought the execution logic would be with the Arduino reading the specified file and setting those values ​​from us. Roughly, to me, it seemed like it was just a matter of a global variable...

Is not that simple because the micro-ROS library is written to use as less dynamic memory as possible, so all the memory configurations should be done at build-time.

OtaviodaCruz, 2021-09-11:
- Thanks for screencast, helped a lot. With that, I was able to build without the mistakes that were happening;
- Really, Arduino should be used colcon_verylowmem.meta. SO, to build, I used: `sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m0`
With *cortex_m0*... this gave me what I expected, with 3 nodes. But when I went to upload to the arduino due board, I had this output:
```
Library micro_ros_arduino-2.0.0-foxy has been declared precompiled:
Precompiled library in "/home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src/cortex-m3" not found
sketch/teste2.ino.cpp.o: In function `timer_callback_1(rcl_timer_t*, long long)':
/home/otavio/Área de Trabalho/teste2/teste2.ino:44: undefined reference to `rcl_publish'
sketch/teste2.ino.cpp.o: In function `timer_callback_2(rcl_timer_t*, long long)':
/home/otavio/Área de Trabalho/teste2/teste2.ino:58: undefined reference to `rcl_publish'
sketch/teste2.ino.cpp.o: In function `set_microros_transports':
/home/otavio/Arduino/libraries/micro_ros_arduino-2.0.0-foxy/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
sketch/teste2.ino.cpp.o: In function `setup':
/home/otavio/Área de Trabalho/teste2/teste2.ino:77: undefined reference to `rcutils_get_default_allocator'
/home/otavio/Área de Trabalho/teste2/teste2.ino:81: undefined reference to `rclc_support_init'
/home/otavio/Área de Trabalho/teste2/teste2.ino:85: undefined reference to `rcl_get_zero_initialized_node'
/home/otavio/Área de Trabalho/teste2/teste2.ino:86: undefined reference to `rclc_node_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:89: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/otavio/Área de Trabalho/teste2/teste2.ino:89: undefined reference to `rclc_publisher_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:96: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/otavio/Área de Trabalho/teste2/teste2.ino:96: undefined reference to `rclc_subscription_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:104: undefined reference to `rclc_timer_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:112: undefined reference to `rclc_executor_init'
/home/otavio/Área de Trabalho/teste2/teste2.ino:113: undefined reference to `rclc_executor_add_timer'
/home/otavio/Área de Trabalho/teste2/teste2.ino:114: undefined reference to `rclc_executor_add_subscription'
/home/otavio/Área de Trabalho/teste2/teste2.ino:119: undefined reference to `rcl_get_zero_initialized_node'
/home/otavio/Área de Trabalho/teste2/teste2.ino:121: undefined reference to `rclc_node_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:125: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/otavio/Área de Trabalho/teste2/teste2.ino:125: undefined reference to `rclc_publisher_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:132: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/otavio/Área de Trabalho/teste2/teste2.ino:132: undefined reference to `rclc_subscription_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:140: undefined reference to `rclc_timer_init_default'
/home/otavio/Área de Trabalho/teste2/teste2.ino:148: undefined reference to `rclc_executor_init'
/home/otavio/Área de Trabalho/teste2/teste2.ino:149: undefined reference to `rclc_executor_add_timer'
/home/otavio/Área de Trabalho/teste2/teste2.ino:150: undefined reference to `rclc_executor_add_subscription'
sketch/teste2.ino.cpp.o: In function `loop':
/home/otavio/Área de Trabalho/teste2/teste2.ino:158: undefined reference to `rclc_executor_spin_some'
/home/otavio/Área de Trabalho/teste2/teste2.ino:159: undefined reference to `rclc_executor_spin_some'
sketch/teste2.ino.cpp.o: In function `__static_initialization_and_destruction_0':
/home/otavio/Área de Trabalho/teste2/teste2.ino:29: undefined reference to `rclc_executor_get_zero_initialized_executor'
/home/otavio/Área de Trabalho/teste2/teste2.ino:30: undefined reference to `rclc_executor_get_zero_initialized_executor'
collect2: error: ld returned 1 exit status
exit status 1
Erro compilando para a placa Arduino Due (Programming Port)
```

I managed to solve it by removing the "cortex-m3" folder from a previous construction, with:
`sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m3`. Ie for "cortex_m3". I didn't understand why

OtaviodaCruz, 2021-09-11:
> Is not that simple because the micro-ROS library is written to use as less dynamic memory as possible, so all the memory configurations should be done at build-time.

Is this due to RTOS (I think it uses FreeRTOS)?

pablogs9, 2021-09-13:
No, this library is not using any RTOS in the Arduino port.

Regarding your library build, you need to use `-p cortex_m3` because you are using an [Arduino Due](https://store-usa.arduino.cc/products/arduino-due) with a Cortex-M3. Probably using a different flag will mess up with the other libraries.

I'm closing since this is solved.

closed 2021-09-13 by pablogs9