micro-ROS/micro_ros_arduino #474: Variable "msg" used in examples is inconsistent

Woody363, 2021-09-08:
Hardware: Teensy

#### Steps to reproduce the issue

rename msg variable to msg_1 in publisher example in all places where msg appears. 

This new sketch fails to compile. (likely true for other examples too)

Leaving the variable as just msg within the rclc_publisher_init_default() function allows the function to compile fine. 

#### Expected behavior

When renaming variables within an example it should still compile without any problems.

#### Actual behavior

the msg variable appears to be defined both within the library and separately within the examples, these are inconsistent and confusing. There are occurrences of msg that can be renamed and occurrences that cannot

These variables should be named differently in all places to ensure they are not confused

#### Additional information

pablogs9, 2021-09-08:
Can you paste the code with your modifications in order to check where is the build failing?

Woody363, 2021-09-10:
Here is the publisher example code with the cases of msg changed to msg_1, which fails to compile.

`#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg_1;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg_1, NULL));
    msg_1.data++;
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_1, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg_1.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}`

pablogs9, 2021-09-13:
You are modifying the typesupport macro line `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_1, Int32)`, this will fail becouse there is no message defined as `std_msgs/msg_1/Int32` the correct name is  `std_msgs/msg/Int32`.

Be careful when using replace all tools.

closed 2021-09-13 by pablogs9

Woody363, 2021-09-13:
Thank you for the explanation,

The issue appears to be with the definition of ROSIDL_GET_MSG_TYPE_SUPPORT that really should take a string input or at least something with a different naming convention so that variables are not so easily confused.

I would recommend that in the examples variable names such as "msg" and "Int32" etc are avoided to limit confusion with the ROSIDL_GET_MSG_TYPE_SUPPORT function