micro-ROS/micro_ros_arduino #556: rclc_executor_spin_some crashes after some execution

cedricmelancon, 2021-11-07:
## Issue template

- Hardware description: OpenCR v1.0
- RTOS: arduino
- Installation type: default setup with docker
- Version or commit hash: foxy

#### Steps to reproduce the issue
Run the following code on the OpenCR:
`#include <micro_ros_arduino.h>
#include <yaml.h>
#include <Arduino.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <geometry_msgs/msg/twist.h>

#define LED_PIN 13

rclc_executor_t executor;
rcl_allocator_t allocator = rcl_get_default_allocator();
rclc_support_t support;
rcl_node_t node;
rcl_publisher_t imuPublisher;

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void setup() 
{  
  pinMode(LED_PIN, OUTPUT);
  set_microros_transports();
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  RCCHECK(rclc_node_init_default(&node, "p3at", "p3at", &support));

  RCCHECK(rclc_publisher_init_default(
        &imuPublisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
        "imu"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 6, &allocator));
}

uint64_t getElapsed(uint64_t start)
{
    uint64_t now = micros();

    if (now > start)
    {
        return now - start;
    }
    else
    {
        uint64_t elapsed = 0xFFFFFFFF - start;
        elapsed += now;

        return elapsed;
    }
}

void loop() 
{    
    unsigned long start_loop_time = micros();

    auto msg = geometry_msgs__msg__Twist__create();
    RCSOFTCHECK(rcl_publish(&imuPublisher, msg, NULL));

    // If loop rate is less than required, wait necessary time
    while (getElapsed(start_loop_time) < 50000) 
    {
        RCSOFTCHECK(rclc_executor_spin_some(&executor, 5000));
        delayMicroseconds(10);
    }
}
`
Start the agent:
`sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

Docker output:
`
...
[1636298173.506590] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 48, data: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1636298173.506751] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BFC9351, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 80 0C 00 00 80
[1636298173.556387] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BFC9351, len: 60, data: 
0000: 81 80 80 0C 07 01 34 00 0C 8A 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1636298173.556691] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 48, data: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1636298173.556817] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BFC9351, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 81 0C 00 00 80
`
Then the LED flashes, which means that the error_loop() function is called.

#### Expected behavior
I would expect the code to run forever.

#### Actual behavior
After a while, the error_loop() is called.

#### Additional information
If I change the rate, I get the exact same result, but for a longer period of time.

pablogs9, 2021-11-10:
Both `rclc_executor_spin_some` and `rcl_publish` can return other than `RCL_RET_OK` in certain cases during a normal operation. You are using an `RCSOFTCHECK` that blocks the execution. Maybe you can check the error without stopping your run flow.

cedricmelancon, 2021-11-13:
Is there a list of "accepted" returned error code. What does it mean if I get something else than OK? Was the process run properly?

cedricmelancon, 2021-11-13:
I get the same result with the following code:
`
#include <micro_ros_arduino.h>
#include <yaml.h>
#include <Arduino.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <geometry_msgs/msg/twist.h>

#define LED_PIN 13

rclc_executor_t executor;
rcl_allocator_t allocator = rcl_get_default_allocator();
rclc_support_t support;
rcl_node_t node;
rcl_publisher_t imuPublisher;

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void setup() 
{  
  pinMode(LED_PIN, OUTPUT);
  set_microros_transports();
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  RCCHECK(rclc_node_init_default(&node, "p3at", "p3at", &support));

  RCCHECK(rclc_publisher_init_default(
        &imuPublisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
        "imu"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 6, &allocator));
}

void loop() 
{    
    unsigned long start_loop_time = micros();

    auto msg = geometry_msgs__msg__Twist__create();
    rcl_publish(&imuPublisher, msg, NULL);

    // If loop rate is less than required, wait necessary time
    rclc_executor_spin_some(&executor, 5000);
    delayMicroseconds(50000);
}
`

pablogs9, 2021-11-15:
Return codes: https://github.com/ros2/rcl/blob/9efffb1c8497cc9618446511a337829502ede227/rcl/include/rcl/types.h#L26

In this new code, you are causing a memory leak. `geometry_msgs__msg__Twist__create` will allocate memory and you are not freeing it. In any case, in micro-ROS you should not use the generated `rosidl_generator_c` API, there are other utilities instead: https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

cedricmelancon, 2021-11-17:
That was it! The memory leak was causing the issue.
Thank you very much!

closed 2021-11-17 by cedricmelancon