micro-ROS/micro_ros_arduino #572: Tutorial not working on teensy 4.1

carmiac, 2021-11-17:
- Hardware description: Teensy 4.1
- RTOS: Arduino
- Version or commit hash:Foxy 2.0.2 and 1.0.0, Arduino 1.8.15 and 1.8.13, Teensyduino 1.55 and 1.53, Ubuntu 20.04LTS

The teensy tutorial is not working. Everything builds and loads properly, but when I start the ros2 serial agent, it appears to connect to the teensy and then crash out with no error messages. When this happens the teensy jumps into the error_loop and will just blink away forever.

I'm running Arduino 1.8.13, Teensyduino 1.53 and using a Teensy 4.1 board. I've used both the v2.0.2-foxy micro_ros_arduino release as well as v1.0.0 with the same results. I've also tried it with Arduino 1.8.15 and Teensyduino 1.55 and get the same results.

Here's the output from the micro ros agent terminal:
```
milner@canterbury:~/arduino-1.8.13/hardware/teensy/avr$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1637094118.227826] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094119.231427] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094120.234235] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094121.237957] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094122.239907] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094123.242296] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094124.245006] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094125.247322] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094126.250484] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094127.251737] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094128.253653] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094129.255863] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1637094130.085380] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1637094130.085572] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1637094131.908440] info     | Root.cpp           | create_client            | create                 | client_key: 0x49E910CB, session_id: 0x81
[1637094131.908500] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x49E910CB, address: 0
milner@canterbury:~/arduino-1.8.13/hardware/teensy/avr$
```

pablogs9, 2021-11-18:
Could you try to use the latest commit of the micro_ros_arduino repo & use the agent with:

```
docker run -it --rm --net=host -v /dev/shm:/dev/shm --privileged -v /dev:/dev microros/micro-ros-agent:galactic serial --dev /dev/serial/by-id/usb-Teensyduino_USB_Serial_8180040-if00 -v6
```

I have just tested Teensy 3.2 with the repo in d67c9dbe71644f004704914f2ce2e8dce5b0b94f (foxy) building with arduino-cli v0.1.8 and seems to work, I'm going to do some more tests.

pablogs9, 2021-11-18:
Could you paste your log of the agent with the flag `-v6`?

Acuadros95, 2021-11-18:
I just tried on a fresh foxy ROS install and all examples are working fine. This could be related to your ROS installation, did you modify it in any way? What is your output for `apt list --installed | grep fast`?

Could you run the agent with the debug flag? like this:
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`

Anyway, you can try to run the dockerized agent, which should work out of the box.

carmiac, 2021-11-18:
I just tried re-checking out and building all of micro_ros and now its working.  Not sure if HEAD was broken for a few days or what, but it looks good now.  Thanks!

closed 2021-11-18 by carmiac

avielbr, 2022-09-05:
Hi, I followed [this](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk) guide for getting MicroROS up and running on a Pi Pico, and **succeeded**! I was able to run `echo` the `pico_publisher` topic... once. It ran for a while, then I CTRL+C's out of the agent, and it stopped as expected... Only now when I try and run the dockerized agent I get the same "Serial port not found..." error that @carmiac reported.

**Running Foxy on RPi4, Ubuntu 20.04 Server.**

- I thought maybe it was because I was running the docker agent with `humble` (didn't notice when I copy+pasted), so I pulled the `foxy` version and tried that, with the same result...

- When I run `apt list --installed | grep fast` as per @pablogs9 I see:

```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libfastjson4/focal,now 0.99.8-2 arm64 [installed,automatic]
ros-foxy-fastcdr/focal,now 1.0.13-1focal.20210901.042734 arm64 [installed,automatic]
ros-foxy-fastrtps-cmake-module/focal,now 1.0.4-1focal.20210901.175217 arm64 [installed,automatic]
ros-foxy-fastrtps/focal,now 2.1.1-1focal.20220209.133847 arm64 [installed,automatic]
ros-foxy-rmw-fastrtps-cpp/focal,now 1.3.0-1focal.20220209.140958 arm64 [installed,automatic]
ros-foxy-rmw-fastrtps-shared-cpp/focal,now 1.3.0-1focal.20220209.140231 arm64 [installed,automatic]
ros-foxy-rosidl-typesupport-fastrtps-c/focal,now 1.0.4-1focal.20220209.135407 arm64 [installed,automatic]
ros-foxy-rosidl-typesupport-fastrtps-cpp/focal,now 1.0.4-1focal.20220209.135238 arm64 [installed,automatic]
```

- And when I run the dockerized agent in debug mode with `-v6` I see the same thing with no additional info.

- I tried changing ports (USB3, USB2), reflashing the .uf2 file, purging modemmanager... No luck so far.

Would really appreciate some help on this! Thanks in advance.

pablogs9, 2022-09-06:
@avielbr micro-ROS does not reconnect automatically by default. Check this example for reference: https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino