micro-ROS/micro_ros_arduino #584: Passing Twist message from Ros2/foxy to Teensy3.2 using micro-ros

bob6776, 2021-11-23:
I am running Ros2/foxy under Ubuntu 20.04 and microros for Arduino on a teensy 3.2.

I seem to be able to pass simple messages like int32 or float32 across the interface, but I have tried several more complex messages like geometry_msgs::msg::Twist and I can't get it to function.  It hangs up every time I try to pass the message.  I have looked at the ROS2 created message using ros2 topic echo, and the message looks fine.  But it hangs up the teensy processor on the other end.  I have attached the code.  I use the same code for transferring a single float value and for passing the Twist message, except that I change the message type from std_msgs::msg:Float32 to geometry_msgs::msg::Twist every place it is needed.  The following is the code.  I would appreciate any help I can get resolving this.  iT hits the callback function, turn the LED off, and then hangs up.

#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <LiquidCrystal.h>
#include <geometry_msgs/msg/twist.h>

rcl_subscription_t sub_cmd;
geometry_msgs__msg__Twist* msg_cmd;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
double cmd[4];
int num;
#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

  const int rs = 19, en = 18, d4 = 17, d5 = 16, d6 = 15, d7 = 14;
  LiquidCrystal lcd(rs, en, d4, d5, d6, d7);
  
void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void * msgin)
{  
  digitalWrite(LED_PIN, LOW);  

  const geometry_msgs__msg__Twist * msg_cmd = (const geometry_msgs__msg__Twist *) msgin;

  cmd[0]=msg_cmd->angular.x;
  cmd[1]=msg_cmd->angular.y;
  cmd[2]=msg_cmd->angular.z;
  cmd[3]=msg_cmd->linear.x;

  lcd.setCursor(0,3);
  lcd.print(cmd[0];
}

void setup() {
  set_microros_transports();
  
  lcd.begin(20,4);
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("Setup");  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  

  msg_cmd=geometry_msgs__msg__Twist__create();
  
  delay(2000);

  allocator = rcl_get_default_allocator();
  lcd.setCursor(0,1);
  lcd.print("1");
  
  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
  lcd.print("2");
  
  // create node
  RCCHECK(rclc_node_init_default(&node, "Arm_node", "", &support));
  lcd.print("3");
  
  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &sub_cmd,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs,msg,Twist),
    "Arm_cmd"));
  lcd.print("4");
  
  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &sub_cmd, &msg_cmd, &subscription_callback, ON_NEW_DATA));
  lcd.print("5");  
}

void loop() {
  delay(100);
  lcd.setCursor(10,1);
  lcd.print(++num);  
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
  lcd.setCursor(10,2);
  lcd.print(num); 
  lcd.setCursor(0,3);
  lcd.print(cmd[1]); 
  
}

## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2021-11-23:
You should not be using `geometry_msgs__msg__Twist__create()` since it uses dynamic memory. In micro-ROS types should be handled with care, check this tutorial for more information: https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

bob6776, 2021-11-24:
Pablo, Thanks, but that didn't make any difference.  Actually, I am thankful it didn't work, or I would have worried a lot about the micro-ros code.  I did find the problem though.  As usual, a dumb mistake.  I guess it is true what they say.  If you have enough monkeys with enough keyboards, they will eventually reproduce the works of Shakespeare.  The error was here.

rclc_executor_add_subscription(&executor, &sub_cmd, &msg_cmd, &subscription_callback, ON_NEW_DATA)

In the code, msg_cmd was declared as a pointer, and so the line of code above was passing the address of the pointer rather than the pointer.  I removed the & in front of msg_cmd and it works fine. 

rclc_executor_add_subscription(&executor, &sub_cmd, msg_cmd, &subscription_callback, ON_NEW_DATA)

That' what I get for using exmple code and not paying enough attention to it.

Thanks again

closed 2021-11-24 by pablogs9