micro-ROS/micro_ros_arduino #607: 2 Services on a teensy

robertdijk, 2021-11-30:
## Running 2 services on a teensy

When trying to run two services on a teensy, only one shows up and it is not callable.

- Hardware description: Teensy 4.1
- Version or commit hash: foxy

#### Steps to reproduce the issue
Upload the following modified version of the addtwoints example

```cpp
#include <micro_ros_arduino.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service1;
rcl_service_t service2;
rcl_wait_set_t wait_set;

example_interfaces__srv__AddTwoInts_Response res1;
example_interfaces__srv__AddTwoInts_Request req1;
example_interfaces__srv__AddTwoInts_Response res2;
example_interfaces__srv__AddTwoInts_Request req2;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void service_callback(const void * req, void * res){
  example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
  example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

  //printf("Service request value: %d + %d.\n", (int) req_in->a, (int) req_in->b);

  res_in->sum = req_in->a + req_in->b;
}

void setup() {
  set_microros_transports();
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&service1, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints1"));
  RCCHECK(rclc_service_init_default(&service2, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints2"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &service1, &req1, &res1, service_callback));
  RCCHECK(rclc_executor_add_service(&executor, &service2, &req2, &res2, service_callback));
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
#### Expected behavior
When running `ros2 service list`, 2 services should appear.
And when running `ros2 service call /addtwoints1 example_interfaces/AddTwoInts "{a: 10, b: -3}"` it should reply with the sum.

#### Actual behavior
Running `ros2 service list` returns only
```
/addtwoints1
```
And running `ros2 service call /addtwoints1 example_interfaces/AddTwoInts "{a: 10, b: -3}"` just hangs.

#### Additional information
Does anyone have experience with running multiple services in microROS and maybe have a working example that I can test?

pablogs9, 2021-11-30:
`RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));` you are initing the executor with 1 handle and adding 2 services.

Also, by default, the library has memory for 1 service. Check [here](https://github.com/micro-ROS/micro_ros_arduino/blob/7827e3a329cec55e51ea6faafe38a7fe136d3fac/extras/library_generation/colcon.meta#L46). You have to rebuild the library with more services: https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library

closed 2021-11-30 by pablogs9

robertdijk, 2021-11-30:
Thanks! The change in `colcon.meta` did it! 😄