micro-ROS/micro_ros_arduino #646: micro_ros_arduino port disable problem

ajaykumar666, 2021-12-16:
I'm using STM32F4 boards for microros, the code gets compiled and flashes into them but the port gets disabled after 10 seconds no matter what. I even tried restarting the board and re-flashed the code but the problem still remains. I'm trying to communicate the board with foxy in docker.

pablogs9, 2021-12-16:
Please provide:
1. Instructions for reproducing the issue
2. Details on which board of the supported by `micro_ros_arduino` are you using
3. Details on how are you launching micro-ROS agent
4. Details on your transport
5. Sample code

ajaykumar666, 2021-12-17:
I'm using the stm32f411 black pill and f412 discovery board. I just uploaded the publisher example in them and was trying to pass the device port in docker to run ros2 but by the time I launch the docker the port disappears

pablogs9, 2021-12-17:
Sorry, but with those details and without a replication procedure we cannot help. I would suggest checking if using the micro-ROS Agent without a docker solves the situation.

Feel free to reopen if can provide a detailed replication procedure.

closed 2021-12-17 by pablogs9

ajaykumar666, 2021-12-17:
all I did was download the library from the repo and install it in Arduino ide then used the publisher example within the repo and flashed it in blackpill(stm32f411ce) and tried to pass the device into ros2 foxy docker, but a device error came so I was trying to check the port but could find it, I reconnected the blackpill but the port will stay up for few seconds and then disappears. so I tried to see it in arduino serial monitor, there also the same thing happened.