micro-ROS/micro_ros_arduino #658: device port disabing in micro-ros

ajaykumar666, 2021-12-22:
## Issue template

- Hardware description: stm32f4411ce blackpill
- RTOS: micro-ros on ubuntu 20 system
- Installation type: micro_ros_setup
- Version or commit hash: micro-ROS used: foxy

#### Steps to reproduce the issue
Downloaded the Arduino library from the micro_ros repository :
https://github.com/micro-ROS/micro_ros_arduino
and installed the library in Arduino ide.
From the example of micro_ros_arduino in Arduino complied and flashed the micro-ros_publisher into the Blackpill stm32f411ce. Installed micro-ros agent on ubuntu 20.
When blackpill is connected to the system the device port stays active for 10s after that eh port gets deactivated and is not found by the micro-ros agent on ubuntu
[image]
so when I reset the blackpill I get this error even when I restart the micro-ros agent
[image]
so now I have to start the blackpill and micro-ros agent all together/simultaneously then it gets connected and publishes the topic in ros2
[image]
 I don't want the hardware to deactivate the port if it doesn't find the micro-ros agent or if the wait time can be customized.
Hope I'm clear this time.

pablogs9, 2021-12-22:
You can handle the agent status from the client side, please check the client reconnection example: https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino

ajaykumar666, 2022-01-03:
thanks a lot, the example solved the error but had to change the reconnection interval parameter to 500 or above to make port available.

closed 2022-01-05 by pablogs9