micro-ROS/micro_ros_arduino #659: Teensy 4.0 rebooting with agent is canceled

rscova, 2021-12-22:
Hello, 

I am testing Teensy 4.0 with the reconnection example and a watchdog to check if the reconnection is really working. 
What I have found is that no matter the time you configure the watchdog, If you run the agent and later close the agent, the Teensy will restart.

So it seems, the code is blocked in a loop somewhere.

- Hardware description: Teensy 4.0
- RTOS: Bare metal
- Installation type: micro_ros_arduino
- Version or commit hash: version of micro-ROS used: galaxy

#### Steps to reproduce the issue
1. Connect the teensy 4.0. 
2. Open a terminal and run: `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`
3. In other terminal: `ros2 topic echo /std_msgs_msg_Int32`
You will see the count of the topic
4. Cancel the agent.
After 2.5 seconds the teensy will restart. You can see this with pin 13 (built-in led) or with the count. If you connect the agent again, the count will start in 0 again.

The code is very similar to the reconnection example, I just added a watchdog with the  [WDT_T4 Library](https://github.com/tonton81/WDT_T4).
```
#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>

#include "Watchdog_t4.h"

#define LED_PIN 13
#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){return false;}}

rclc_support_t support;
rcl_node_t node;
rcl_timer_t timer;
rclc_executor_t executor;
rcl_allocator_t allocator;
rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
bool micro_ros_init_successful;

// Watchdog
WDT_T4<WDT3> wdt;
elapsedMillis feed_wdt;

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  (void) last_call_time;
  if (timer != NULL) {
    rcl_publish(&publisher, &msg, NULL);
    msg.data++;
  }
}

// Functions create_entities and destroy_entities can take several seconds.
// In order to reduce this rebuild the library with 
// - RMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT=0
// - UCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS=3

bool create_entities()
{ 
  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "int32_publisher_rclc", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_best_effort(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "std_msgs_msg_Int32"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  executor = rclc_executor_get_zero_initialized_executor();
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  micro_ros_init_successful = true;
}

void destroy_entities()
{
  rcl_publisher_fini(&publisher, &node);
  rcl_node_fini(&node);
  rcl_timer_fini(&timer);
  rclc_executor_fini(&executor);
  rclc_support_fini(&support);

  micro_ros_init_successful = false;
}

void setup() {
  set_microros_transports(); 
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  

  micro_ros_init_successful = false;
  msg.data = 0;

  WDT_timings_t config;
  config.timeout = 2500; // config watchdog to 2.5 seconds
  wdt.begin(config);

  // Visual inspection to see if the teensy 4.0 has reboot or not.
  pinMode(13,OUTPUT);
  digitalWrite(13,HIGH);
}

void loop() {
  if (RMW_RET_OK == rmw_uros_ping_agent(50, 2))
  {
    if (!micro_ros_init_successful) {
      create_entities();
    } else {
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
    }
  } 
  else if (micro_ros_init_successful)
  {
    destroy_entities();
  }

  if (feed_wdt >= 50) { // waits 50 millis to feed the watchdog
    feed_wdt = 0; //reset count
    wdt.feed();
  }

  delay(10);
}
```

I have changed the watchdog timeout to 10 seconds and keep restarting.

#### Expected behaviour
Teensy does not restart.

#### Actual behaviour
Teensy restarts.

pablogs9, 2021-12-23:
Hello @rscova, check the `destroy_entities()` function. By default micro-ROS client tries to destroy entities reliably on the agent side. In this function, you are destructing at least 3 elements in the agent (namely publisher, node, and support or session). Those destructions takes `RMW_UXRCE_ENTITY_DESTROY_TIMEOUT` milliseconds to complete, taking into account that agent is not active, they **will** take `RMW_UXRCE_ENTITY_DESTROY_TIMEOUT` milliseconds. By default `RMW_UXRCE_ENTITY_DESTROY_TIMEOUT` is 1000 ms so at least 3 seconds will be waiting in those functions (In fact much more because a publisher has a couple of related entities in the agent).

Proposed solutions:
- As stated in the comment before `create_entities()` function, try to recompile the library for your platform using the recommended arguments in the colcon meta. Instruction on how to do that is [here](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library).
- Try to do not to call `destroy_entities()`, does the system behaves the same way?

rscova, 2021-12-23:
Hello @pablogs9, I have tried both solutions. Now the teensy does not restart but never reconnects. In the image you can see the output when I run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`, stop the agent, and run it again.
In another terminal, after that, I can't see the topics published.

[image]

rscova, 2021-12-23:
Hi, It seems that if I add a delay >= 20 ms after create_entities and destroy_entities, everything works fine.

pablogs9, 2021-12-23:
Ok, this is normal behavior:
- If you use **the second solution** you are not destructing the entities, so micro-ROS client does not have a memory to create more entities because you have not freed the space for creating them
- The **first solution** should solve the problem, but in any case, the blocking behavior is in the entity destruction is intended:
   - Try to adjust `RMW_UXRCE_ENTITY_DESTROY_TIMEOUT` and `UCLIENT_MAX_SESSION_CONNECTION_ATTEMPTS` to fit your watchdog requirements, or
   - Disable the watchdog while you are destructing entities

pablogs9, 2022-01-28:
Any update on this?

closed 2022-02-02 by pablogs9