micro-ROS/micro_ros_arduino #664: Folder structure to use  micro_ros_arduino with dependency

OtaviodaCruz, 2021-12-31:
## Setup description: 
- ubuntu facal fossa with ROS2 foxy.

##  Introduction

I'm working on a project as discussed [here](https://github.com/micro-ROS/micro_ros_arduino/issues/476). For that, I'm using micro-ros-arduino.

My idea was to have the following file/folder structure:

.scicobot_hardware
├── _include
│   ├── micro_ros_arduino project
│   └── motor_control project
│   └── (...)
├── _docs
├── _calibration
├── _(...)
├── _src
│   └── scicobot_hardware.h
│   └── scicobot_hardware.cpp

See [here](https://github.com/SciCoBot/scicobot_hardware).

However, this setup doesn't seem to work, I got the following error when trying to add scicobot_hardware.h to arduino:

> /home/otavio/Arduino/libraries/scicobot_hardware/src/micro_ros_arduino/src/micro_ros_arduino.h:16:34: fatal error: uxr/client/transport.h: No such file or directory
>  #include <uxr/client/transport.h>

As far as I understand, the compiler must be looking elsewhere for the transport.h file, because of the use of "<>". MAYBE this could be resolved by switching to something like:

` #include "uxr/client/transport.h"`

 But it seems to me a less than optimal solution, first because I didn't want to change the file and second because it might be necessary to change other include directives.

The most attractive solution I see would be to take the files from "micro_ros_arduino/src" and put them in "scicobot_hardware/src". But I don't really like this form of organization.

Considering that I adopt this last solution, I would have to find a way to update the files automatically, as soon as "micro_ros_arduino" does some push (new commit in github). As I understand it, you have a similar process, for example, "src/rclc/client.h" does not seem to be an original file from the micro_ros_arduino project, it seems to me that it comes [from here](https://github.com/ros2/rclc/blob/galactic/rclc/include/rclc/client.h).

## Questions

And finally... .
So, do you have any suggestions on how to insert micro_ros_arduino as a library in my project? How was the file architecture you thought to use the library?
 
Also, if you decide to insert the files from micro_ros_arduino/src into "scicobot_hardware/src", how could I automatically track updates from "micro_ros_arduino"? I saw about github actions (I even took a look at library_generation.yml file, but I couldn't deduce where UNBUILT libraries are updated, like src/rclc/client.h in this [pull request](https://github.com/micro-ROS/micro_ros_arduino/pull/376/files). I also looked at git submodules, but it seems to me that then I would have to use the whole project, as I am doing now.

pablogs9, 2022-01-05:
Hello @OtaviodaCruz, AFAIK in Arduino IDE you don't have control on the include folders so all the micro-ROS headers should be in the folder reachable by the compiler as an include path. It is discussed for example [here](https://github.com/arduino/arduino-cli/issues/501).

Regarding keeping track of the latest commit of the micro-ROS library, it is not straightforward in the current folder architecture. Have you thought about using [Platform.IO](https://github.com/micro-ROS/micro_ros_arduino#platformio)? I'm not an expert in it but it seems to have better library management than Arduino IDE.

OtaviodaCruz, 2022-01-05:
Hello @pablogs9, thanks for the reply. I didn't think about using Platform.IO, but I'll try it out.

As for folder architecture, do you suggest something to me? The best possibilities I think are:

- leave libraries outside the robot application (scicobot_hardware.h and scicobot_hardware.cpp). That way, I could leave in a tutorial specs to add the dependencies separately;

- put all .h and .cpp files in my scicobot_hardware/src folder.

I must admit that both of them don't seem interesting to me, but that's what I see as a possible solution.

As for tracking the last commit of the micro-ROS library (and even the others), I was thinking of making a github action that more or less followed these steps:

- access the ubuntu system;
- I clone the Arduino micro-ROS repository;
- copy the necessary files;
- create a new branch, to allocate changes;
- I create a pull request with the changes, for later analysis.

I can do this once a day maybe. Sound like a good approach? If yes, do you suggest any workflow for the action?

This is my first "big project" so these conceptual questions I imagine.

OtaviodaCruz, 2022-01-05:
(...) Perhaps these questions can help beginners who use micro_ros_arduino to create the structure of their projects, serving as a reference for use.

pablogs9, 2022-01-07:
You have an action that makes something similar here: https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/.github/workflows/library_generation.yml. It just downloads the repo, makes some modifications (in this case running a docker command) and creates a PR with the changes.

Regarding the folder structure, please keep us updated with your best choice.

OtaviodaCruz, 2022-01-08:
Okay. Thanks for the help @pablogs9.

OtaviodaCruz, 2022-01-11:
Hi, update on my decision:

- I took a quick look at the PlatformIO platform. I decided to keep it on the Arduino platform for now. This is mainly due to the time I have available to deliver the project, I would need more time to investigate the real viability of PlatformIO against Arduino.

- Another decision point is that I preferred to work with the libraries outside of my main project (scicobot_hardware). So, I use the libraries in the standard arduino folder. This decision can be very private, but I believe that this way it is better modularized and more organized and easier to change.
To optimize this process, I created a directory (https://gicobot.git/build_build_arquicobot.git) for "workbench build", I did it from some bash files, which download/update the necessary libraries in the right place ( Archive/Libraries).

I already have a part of the project in this structure and, until then, I intend to keep it there.

closed 2022-01-13 by OtaviodaCruz