micro-ROS/micro_ros_arduino #667: Compilation error for Teensy3.6

BenjaminDeHeisenberg, 2022-01-07:
## Issue template

- Hardware description: Teensy 3.6
- Installation type: micro-ros-arduino
- Version or commit hash: foxy

#### Steps to reproduce the issue
Followed the instructions mentioned on the Readme file to compile on platformIO and tried to compile the micro_ros_publisher 
code from the example folder. I have followed all the changes that needs to be done on the platform.ini file according to my folder structure.
During compilation, the error that is occurring is:
  Compiling .pio\build\teensy36\libc62\NativeEthernet\NativeMdns.cpp.o
C:\Users\sample\.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernet.cpp: In static member function 'static int EthernetClass::begin(uint8_t*, long unsigned int, long unsigned int)':
C:\Users\sample\.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernet.cpp:126:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                                    ^
C:\Users\sample\.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernet.cpp: In static member function 'static void EthernetClass::begin(uint8_t*, IPAddress, IPAddress, IPAddress, IPAddress)':
C:\Users\sample\.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernet.cpp:258:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                                    ^
*** [.pio\build\teensy36\libc62\NativeEthernet\NativeEthernet.cpp.o] Error 1
Compiling .pio\build\teensy36\libc62\NativeEthernet\Nativesocket.cpp.o

This error does not happen while I change the target board to teensy4.1 but while compiling for teensy3.6 this error is happening

#### Expected behavior
it should compile
#### Actual behavior
it isnt compiling
#### Additional information

BenjaminDeHeisenberg, 2022-01-07:
Cause of the issue is in: micro_ros_arduino/src/default_transport.cpp the default transport for teensy6.6 is yet to be fixed and built.
Currently only a few boards are supported like teensy 4.1 and a few others

It would be great if someone gets the default transport function for teensy3.6

Acuadros95, 2022-01-07:
Hi @BenjaminDeHeisenberg,

I think the problem is that plaformIO is trying to build the `NativeEthernet` library, which is intended for teensy 4.1 boards.

I'm not familiar with platformIO, so I am not sure why this is happening. `ARDUINO_TEENSY41` should not be defined, leaving the library out here: https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/src/native_ethernet_transport.cpp#L9

Maybe @DominikN could give some insight as he included the platformIO example, or @grassjelly which got an example working on a teensy 3.6 here: https://github.com/linorobot/micro_ros_platformio

Also, @carmiac may be able to help, as he included the native ethernet trasport for teensy 4.1.

closed 2022-01-12 by BenjaminDeHeisenberg

grassjelly, 2022-01-16:
Just figured out how to fix this. You can use lib_ignore on your .ini file to make PlatformIO ignore building NativeEthernet library unnecessarily. https://github.com/linorobot/linorobot2_hardware/commit/09a905787ab645cedbfea9c404ed95d281877700

BenjaminDeHeisenberg, 2022-01-16:
Thanks a lot for your support! It means a lot to me. I was able to compile
it properly after a patch for teensy3.6 board was added on the github repo,
and the code executed properly!
Regards,
Benjamin Varughese.

pablogs9, 2022-01-17:
Do we need to modify something to solve this? @grassjelly