micro-ROS/micro_ros_arduino #671: Error compiling for Teensy 3.3

NAEE09, 2022-01-10:
## Issue template

- Hardware description: Teensy 3.6
- RTOS: 
- Installation type: precompiled library Arduino IDE
                             micro ros in computer, no docker build
- Version or commit hash: foxy

#### Steps to reproduce the issue
In arduino IDE, compile the micro-ros_publisher example

Error: 

Arduino: 1.8.15 (Linux), TD: 1.54, Board: "Teensy 3.6, Serial, 180 MHz, Faster, US English"
```
WARNING: library micro_ros_arduino-2.0.2-foxy claims to run on OpenCR, Teensyduino, samd, sam, mbed architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino-2.0.2-foxy has been declared precompiled:
Precompiled library in "/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0/fpv4-sp-d16-hard" not found
Precompiled library in "/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0" not found
/tmp/arduino_build_466103/sketch/micro-ros_publisher.ino.cpp.o: In function `timer_callback(rcl_timer_t*, long long)':
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/tmp/arduino_build_466103/sketch/micro-ros_publisher.ino.cpp.o: In function `set_microros_transports':
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino_build_466103/sketch/micro-ros_publisher.ino.cpp.o: In function `setup':
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
/tmp/arduino_build_466103/sketch/micro-ros_publisher.ino.cpp.o: In function `loop':
/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.6.
```
I followed the next tutorials:

https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/
https://github.com/micro-ROS/micro_ros_arduino

pablogs9, 2022-01-11:
It seems that your Arduino IDE is looking for the library here: 

```
Precompiled library in "/home/user/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0/fpv4-sp-d16-hard" not found
```

But the library is precompiled here [`micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/`](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/src/imxrt1062/fpv5-d16-hard). There is a difference between `fpv4` and `fpv5`. 

Maybe it is related to Arduino or Teensyduino versions. We are using Arduino 1.8.13 and Teensyduino 1.53. Let me check.

pablogs9, 2022-01-11:
Seems that Teensyduino 1.54 does not give the error: https://github.com/micro-ROS/micro_ros_arduino/pull/673

Acuadros95, 2022-01-11:
It looks like teensy 3.6 support was added by the community for PlatformIO projects (https://github.com/micro-ROS/micro_ros_arduino/pull/388) and its not included in our regular Arduino IDE builds.

I have added it on this PR: https://github.com/micro-ROS/micro_ros_arduino/pull/674. Please give it a try and give feedback, as we don't have a 3.6 board to test it ourselves.

NAEE09, 2022-01-11:
Hi, thanks for the fast reply, I have tried with the new library changes and I have the same error.

NAEE09, 2022-01-11:
hey again!!!, I don't know what happened, but after I testing the new library changes, I noticed that in the zip there was no mk66fx1m0 folder, and I checked again and the folder was there, so the example compile without error.

Thanks.

closed 2022-01-11 by NAEE09